coolsnowwolf/lede #9697: 希望可以提供内网设备的IPv6的解析。

huangsijun17, 2022-07-05:
希望可以直接通过MAC或DUID来指定内网设备，并将域名AAAA解析到该设备的IPv6地址之上。
PS，我现在是通过脚本获取内网设备的地址。
`ip -6 n | grep ^2.*xx:xx:xx:xx:xx:xx | tail -n 1 | awk '{print $1}'`

OldCoding, 2022-07-06:
你为什么不直接在内网设备上开DDNS？

huangsijun17, 2022-07-07:
> 你为什么不直接在内网设备上开DDNS？

3-5个设备，每个都去独立安装、开启、维护？
再者，这一功能在爱快中也是有的，添加对于OP也并不存在负担。

huangsijun17, 2022-07-09:
> socat

socat是CPU转发，L大之前给的端口转发建议是NAT46。但内网设备本来就有IPv6地址，何必多此一举？

zopargr, 2022-07-09:
其实现在的问题是有些设备没有ipv6地址，在仅有公网ipv6地址的情况下，从外网访问内网ipv4设备是比较麻烦的，比较好的做法就是通过zerotier这种虚拟局域网来访问。

huangsijun17, 2022-07-10:
> 其实现在的问题是有些设备没有ipv6地址，在仅有公网ipv6地址的情况下，从外网访问内网ipv4设备是比较麻烦的，比较好的做法就是通过zerotier这种虚拟局域网来访问。

有ZT……我甚至连拿ZT访问公网都研究出来了。

nivalxer, 2022-07-10:
可以试试DDNS-高级设置，IP地址来源，选择脚本，然后指定脚本。

huangsijun17, 2022-07-11:
> 可以试试DDNS-高级设置，IP地址来源，选择脚本，然后指定脚本。

之前就已如是获取IP了，但希望未来可以通过DDNS内置设置实现。

liangshihu, 2022-10-05:
> 希望可以直接通过MAC或DUID来指定内网设备，并将域名AAAA解析到该设备的IPv6地址。 PS，我现在是通过脚本获取内网设备的地址。 `ip -6 n | grep ^2.*xx:xx:xx:xx:xx:xx | tail -n 1 | awk '{print $1}'`

赞同，我也想要，但有没有脚本？

huangsijun17, 2022-10-06:
> > 希望可以直接通过MAC或DUID来指定内网设备，并将域名AAAA解析到该设备的IPv6地址。 PS，我现在是通过脚本获取内网设备的地址。 `ip -6 n | grep ^2.*xx:xx:xx:xx:xx:xx | tail -n 1 | awk '{print $1}'`
> 
> 赞同，我也想要，但有没有脚本？

```
#!/bin/sh
# 2022 huangsijun17
# ip -6 n | grep ^2.*9c:5c:8e:bc:e7:13 | tail -n 1 | awk '{print $1}'

ip=$(ip -6 n | grep ^2.*9c:5c:8e:bc:e7:13 | tail -n 1 | awk '{print $1}')

if test -z $ip; then
	echo "::1"
else
	echo $ip
fi
```
我现在的脚本，仅供参考。请自行替换MAC地址。

liangshihu, 2022-10-06:
Thanks♪(･ω･)ﾉ

从 Windows 版邮件发送

发件人: huangsijun17
发送时间: 2022年10月6日 15:50
收件人: coolsnowwolf/lede
抄送: aimiya; Comment
主题: Re: [coolsnowwolf/lede] 希望可以提供内网设备的IPv6动态域名解析。 (Issue #9697)

希望可以直接通过MAC或DUID来指定内网设备，并将域名AAAA解析到该设备的IPv6地址。 PS，我现在是通过脚本获取内网设备的地址。 ip -6 n | grep ^2.*xx:xx:xx:xx:xx:xx | tail -n 1 | awk '{print $1}'
赞同，我也想要，但有没有脚本？
#!/bin/sh

# 2022 huangsijun17

# ip -6 n | grep ^2.*9c:5c:8e:bc:e7:13 | tail -n 1 | awk '{print $1}'

ip=$(ip -6 n | grep ^2.*9c:5c:8e:bc:e7:13 | tail -n 1 | awk '{print $1}')

if test -z $ip; then

	echo "::1"

else

	echo $ip

fi

我现在的脚本，仅供参考。请自行替换MAC地址。

shaojs321, 2022-12-13:
好功能我也想要，每个设备单独ddns太麻烦了，没有统一管理方便

huangsijun17, 2022-12-13:
> 好功能我也想要，每个设备单独ddns太麻烦了，没有统一管理方便

拿我写的脚本先凑活吧。
https://github.com/coolsnowwolf/lede/issues/9697#issuecomment-1269516737

coolsnowwolf, 2026-09-05:
这是为内网设备按 MAC/DUID 自动生成 IPv6 DDNS AAAA 记录的功能请求，不是当前代码错误；需要特定 DDNS 服务与地址生命周期设计，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf