coolsnowwolf/lede #9701: dns2tcp 下载报错了

wind007, 2022-07-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：dns2tcp 下载报错，源 我是不是要换一下？如果是要换成哪个？请知道的大佬给点建议

### (2) 路由器型号和固件版本

A：x86 

### (3) 详细日志

A：
install -d -m0755 /workdir/openwrt/staging_dir/hostpkg/bin
install -m0755 src/po2lmo /workdir/openwrt/staging_dir/hostpkg/bin/po2lmo
install -m0755 /workdir/openwrt/build_dir/hostpkg/LuaSrcDiet-0.12.1/bin/LuaSrcDiet.lua /workdir/openwrt/staging_dir/hostpkg/bin/LuaSrcDiet
mkdir -p /workdir/openwrt/staging_dir/hostpkg/stamp
touch /workdir/openwrt/build_dir/hostpkg/LuaSrcDiet-0.12.1/.built
touch /workdir/openwrt/staging_dir/hostpkg/stamp/.luci-base_installed
make[3]: Leaving directory '/workdir/openwrt/feeds/luci/modules/luci-base'
time: package/feeds/luci/luci-base/host-compile#1.46#0.65#1.95
make[3]: Entering directory '/workdir/openwrt/feeds/oldpackages/net/dns2tcp'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.dns2tcp-0.5.2.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "dns2tcp-0.5.2.tar.gz" "51c5dc69f5814c2936ce6832217d292d" "" "http://www.hsc.fr/ressources/outils/dns2tcp/download/"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://www.hsc.fr/ressources/outils/dns2tcp/download/dns2tcp-0.5.2.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/dns2tcp-0.5.2.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/dns2tcp-0.5.2.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/dns2tcp-0.5.2.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:47: /workdir/openwrt/dl/dns2tcp-0.5.2.tar.gz] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/oldpackages/net/dns2tcp'
time: package/feeds/oldpackages/dns2tcp/compile#0.20#0.06#2.10
    ERROR: package/feeds/oldpackages/dns2tcp failed to build.
make[2]: *** [package/Makefile:116: package/feeds/oldpackages/dns2tcp/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2022-07-06:
自己不知道到哪里找的库，隔壁helloword 去看

wind007, 2022-07-06:
> 自己不知道到哪里找的库，隔壁helloword 去看

因为我也不是 很确定是不是hello world 导致的问题，所以在这边提问看看有没有人碰到类似问题。如果大佬能确定是hello world 导致的，那我把 issue 提到那边

WYC-2020, 2022-07-06:
> > 自己不知道到哪里找的库，隔壁helloword 去看
> 
> 因为我也不是 很确定是不是hello world 导致的问题，所以在这边提问看看有没有人碰到类似问题。如果大佬能确定是hello world 导致的，那我把 issue 提到那边

你没懂我意思，我是说 你这个库哪里找的，自己去看别人helloword 的，不是说helloword的有问题，别瞎搞
[image]

wind007, 2022-07-06:
懂你意思了。谢谢大佬指点

closed 2022-07-06 by wind007

wind007, 2022-07-06:
L大 已经把 oldpackages 注释掉了， 我用云编译的脚本又把注释取消了，导致 拿到 旧的包。feeds.conf.default 中 的 oldpackages，应该是要注释的，不能取消掉。