coolsnowwolf/lede #9704: 想通过hosts屏蔽一些电视开机广告，但是似乎没有起作用，是哪里操作不对吗？

3538532, 2022-07-06:
路由器型号和固件版本
软路由，内核版本：5.15.48，OpenWrt R22.6.16

在etc/hosts里加入如下内容：
`#version: 202204170012
127.0.0.1 ad.mi.com
127.0.0.1 adv.sec.miui.com
127.0.0.1 api.e.mi.com
127.0.0.1 mivideo.g.mi.com
127.0.0.1 gallery.pandora.xiaomi.com
127.0.0.1 api.ad.xiaomi.com
127.0.0.1 cdn.ad.xiaomi.com
127.0.0.1 data.mistat.xiaomi.com
127.0.0.1 e.ad.xiaomi.com
127.0.0.1 log.ad.xiaomi.com
127.0.0.1 sdkconfig.ad.intl.xiaomi.com
127.0.0.1 sdkconfig.ad.xiaomi.com
127.0.0.1 stat.pandora.xiaomi.com
127.0.0.1 test.ad.xiaomi.com
127.0.0.1 test.e.ad.xiaomi.com
127.0.0.1 test.zeus.ad.xiaomi.com
127.0.0.1 zeus.ad.xiaomi.com
127.0.0.1 wtradv.market.xiaomi.com
127.0.0.1 bss.pandora.xiaomi.com
127.0.0.1 de.pandora.xiaomi.com
127.0.0.1 dvb.pandora.xiaomi.com
127.0.0.1 jellyfish.pandora.xiaomi.com
127.0.0.1 starfish.pandora.xiaomi.com
127.0.0.1 mishop.pandora.xiaomi.com
127.0.0.1 upgrade.mishop.pandora.xiaomi.com`

但是似乎没有起到作用，仍然会出现开机广告。
请教大佬，是这样操作不对，还是什么原因呢？
谢谢！

tedaz, 2022-07-07:
> 路由器型号和固件版本 软路由，内核版本：5.15.48，OpenWrt R22.6.16
>
> 在etc/hosts里加入如下内容： #version: 202204170012 127.0.0.1 ad.mi.com 127.0.0.1 adv.sec.miui.com 127.0.0.1 api.e.mi.com 127.0.0.1 mivideo.g.mi.com 127.0.0.1 [...]

电视机内部都有多种连接方式的，类似于翻墙工具，仅仅dns污染不能屏蔽所有广告。

zhangguanzhang, 2022-07-08:
你在op上改op的hosts没用，除非你dnsmasq的额外hosts文件设置为它

3538532, 2022-07-08:
> 额外hosts

谢谢，说的是，改为额外hosts试试。

3538532, 2022-07-12:
> > 额外hosts
> 
> 谢谢，说的是，改为额外hosts试试。

好像额外的hosts也不起作用啊，昨天又出来广告了。
哪位大佬用额外的hosts起作用的吗？

Cathgao, 2022-07-12:
host不是万能的

zhangguanzhang, 2022-07-13:
> > > 额外hosts
> > 
> > 
> > 谢谢，说的是，改为额外hosts试试。
> 
> 好像额外的hosts也不起作用啊，昨天又出来广告了。 哪位大佬用额外的hosts起作用的吗？

域名广告肯定是能hosts屏蔽的，但是有些广告是 url 广告，这类，你用浏览器的话就使用浏览器插件去屏蔽

Ho-Lay, 2022-07-13:
> 路由器型号和固件版本 软路由，内核版本：5.15.48，OpenWrt R22.6.16
>
> 在etc/hosts里加入如下内容： #version: 202204170012 127.0.0.1 ad.mi.com 127.0.0.1 adv.sec.miui.com 127.0.0.1 api.e.mi.com 127.0.0.1 mivideo.g.mi.com 127.0.0.1 [...]

你如果用啟用了ssrplus，ssrplus起作用，在ssrplus那添加屏蔽的域名

3538532, 2022-07-13:
> 你如果用啟用了ssrplus，ssrplus起作用，在ssrplus那添加屏蔽的域名

我在用ssrplus，准备试试你这个方法，谢谢！

pikachullk, 2022-07-16:
我是考虑到会不会有一种流量叫加密流量~就是让你OP都不知道这个包是要去哪的~
然后host就失去了作用...所以koolproxy才需要安装ssl根证书来解包这些流量。

drw2005, 2022-07-29:
> 你如果用啟用了ssrplus，ssrplus起作用，在ssrplus那添加屏蔽的域名

你这个方法，我用了半个月，中间出现过一次广告。

coolsnowwolf, 2026-09-05:
直接修改 `/etc/hosts` 是否生效取决于客户端是否使用路由器 DNS、dnsmasq hosts 配置及缓存；报告没有客户端 DNS/解析结果或当前配置，无法判断 LEDE 缺陷，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf