coolsnowwolf/lede #9706: ESXI 6.7 下，x64-R22.5.5  cpu 负载较高。

mailsex, 2022-07-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
ESXI 6.7 下，x64-R22.5.5  cpu 负载较高。
Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述
在软路由上，ESXI 版本为6.7.0 Update 3 (Build 15160138)，
虚拟机配置：1cpu、512 RAM，bios 启动。使用 StarWind V2V Image Converter 将openwrt-x64-R22.5.5-generic-squashfs-combined-efi.img 文件转化成 ESXI 磁盘 ，上传至esxi 平台部署。
### (1) 具体问题
同样的虚拟机配置，空负载情况下，x64-R22.5.5 cpu占用比 R21.11.11高不少，如下图
A：
CPU占用截图
[image]
[image]
网络占用截图
[image]
[image]

### (2) 路由器型号和固件版本

A：MacMini 安装了esxi 6.7，再部署的openwrt，旁路路由模式。上联另一个主路由。

### (3) 详细日志

A：x64-R22.5.5 系统日志
http://216.24.178.186:888/chfs/shared/%E7%B3%BB%E7%BB%9F%E6%97%A5%E5%BF%97.txt
B：x64-R22.5.5 内核日志
http://216.24.178.186:888/chfs/shared/%E5%86%85%E6%A0%B8%E6%97%A5%E5%BF%97.txt

WYC-2020, 2022-07-07:
这个时候不是应该看那个进程占用高了 这些截图能看出个啥

mailsex, 2022-07-07:
> 这个时候不是应该看那个进程占用高了 这些截图能看出个啥

进程上看起来差不多。。。。

GreatMichaelLee, 2022-07-07:
进来看了个寂寞，哈哈

closed 2022-07-10 by coolsnowwolf