coolsnowwolf/lede #9709: mt7621无线不能上网的问题

wuzhuang123, 2022-07-07:
极路由HC5961-mt7621，从上月底开始无线的2.4G和5G都不能传输，访问路由器主页都非常慢，有线LAN口是正常的。还有一台小米mini-mt7620则完全正常。看日志没有任何异常的地方。当同时启用ipv6时开机会同时不能获取ipv4地址，这个问题再四月底也提过。

286700234, 2022-07-07:
WiFi的这个问题好久了，也没有人修复

wuzhuang123, 2022-07-08:
> WiFi的这个问题好久了，也没有人修复

今天早上的更新无线好像复活了？正在测试

lovehackintosh, 2022-07-08:
可以尝试把/etc/rc.local改成以下内容,然后重启

# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.
# Wireless starter and Interface checker by Kenvix <i@kenvix.com>

# Start ra unconditionally
sleep 2s
ip link set rai0 up
ip link set ra0 up
echo "Wireless interface started"

# Lan Check
lanCheck=`uci get network.lan.ifname`
if [ $? -eq 0 ]; then
    echo $lanCheck | grep rai0 > /dev/null
    if [ $? -ne 0 ]; then
        uci set network.lan.ifname="$lanCheck rai0 ra0"
        uci commit
        echo "Updated wireless config of LAN Interface"
    fi
    echo "No need to update wireless config of LAN Interface"
else
    echo "wireless config of LAN Interface check failed. Interface may renamed." >&2
fi
exit 0

wuzhuang123, 2022-07-08:
> 可以尝试把/etc/rc.local改成以下内容,然后重启
> 
> ```shell
> # Put your custom commands here that should be executed once
> # the system init finished. By default this file does nothing.
> # Wireless starter and Interface checker by Kenvix <i@kenvix.com>
> 
> # Start ra unconditionally
> sleep 2s
> ip link set rai0 up
> ip link set ra0 up
> echo "Wireless interface started"
> 
> # Lan Check
> lanCheck=`uci get network.lan.ifname`
> if [ $? -eq 0 ]; then
>     echo $lanCheck | grep rai0 > /dev/null
>     if [ $? -ne 0 ]; then
>         uci set network.lan.ifname="$lanCheck rai0 ra0"
>         uci commit
>         echo "Updated wireless config of LAN Interface"
>     fi
>     echo "No need to update wireless config of LAN Interface"
> else
>     echo "wireless config of LAN Interface check failed. Interface may renamed." >&2
> fi
> exit 0 
> ```

好像用处不大。唉，继续挂二级路由

coolsnowwolf, 2022-07-08:
已修复

wuzhuang123, 2022-07-09:
> 已修复

感谢！无线已经可以访问局域网其它设备了，但是上外网仍然不行，然而开启ssr后只能上走ssr的网站，原因未知。设备已清除所有配置。

asiaon123, 2022-07-09:
> 已修复

同样测试无线没有速度，HC5962

modelsun, 2022-07-09:
> 

请问，用的是什么版本内核，开源驱动WIFI吗？

KGEW, 2022-07-09:
出现同样问题，换成闭源驱动能解决吗？

asiaon123, 2022-07-09:
> 请问，用的是什么版本内核，开源驱动WIFI吗？

5.4 开源驱动

modelsun, 2022-07-09:
> > 请问，用的是什么版本内核，开源驱动WIFI吗？
> 
> 5.4 开源驱动

谢谢

coolsnowwolf, 2022-07-10:
已修复

closed 2022-07-10 by coolsnowwolf