coolsnowwolf/lede #9713: 5.18 compile error

GreatMichaelLee, 2022-07-08:
提示缺这几个ko,好像之前升5.15也遇到过，忘了怎么处理的，要在哪个文件里加上？

mkdir -p /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-keys-trusted is missing dependencies for the following libraries:
**asn1_decoder.ko
asn1_encoder.ko
oid_registry.ko**
make[3]: *** [modules/other.mk:1148: /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-keys-trusted_5.18.10-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/package/kernel/linux'
time: package/kernel/linux/compile#2.75#0.33#3.00
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_x86$

zopargr, 2022-07-08:
检查文件完整性
`find dl -size -1024c -exec ls -l {} \;`
删除错误文件
`find dl -size -1024c -exec rm -f {} \;`
换个好点的梯子重新make download
`make download`
仅供参考
刚成功默认config编译完5.18

GreatMichaelLee, 2022-07-08:
感觉没有修复干净，我rm -rf ./tmp && make clean之后编译的，这次换了一个kernel module (cifs), 还是那几个ko依赖，麻烦大佬再看看. @coolsnowwolf 

make[3]: Entering directory '/home/builder/lede_x86/package/kernel/linux'
mkdir -p /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp

SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package **kmod-fs-cifs** is missing dependencies for the following libraries:
asn1_decoder.ko
dns_resolver.ko
oid_registry.ko
make[3]: *** [modules/fs.mk:121: /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-fs-cifs_5.18.10-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/package/kernel/linux'
time: package/kernel/linux/compile#3.03#0.32#2.94
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_x86$

WYC-2020, 2022-07-09:
我的个乖乖5.18?这么激进的哇哈哈，很多功能没移植完毕吧，包括很多patch

GreatMichaelLee, 2022-07-09:
> 我的个乖乖5.18?这么激进的哇哈哈，很多功能没移植完毕吧，包括很多patch

哈哈，玩这个不就是要鸡进+鸡动：P

WYC-2020, 2022-07-09:
> > 我的个乖乖5.18?这么激进的哇哈哈，很多功能没移植完毕吧，包括很多patch
> 
> 哈哈，玩这个不就是要鸡进+鸡动：P

躲在角落瑟瑟发抖哈哈，我还在用5.4虽然现在5.10 5.15都出来了，可以的你们先去当先锋队，我在后面捡剩下的

GreatMichaelLee, 2022-07-09:
这里这里，大佬修一下啊
[image]
[image]