coolsnowwolf/lede #9716: R2S编译失败，arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts:10:10: fatal error: rk3328-dram-nanopi2-timing.dtsi: No such file or directory

lifetyper, 2022-07-09:
最新版本代码，编译r2s时提示：
```
  GEN     arch/arm64/include/generated/asm/cpucaps.h
  CC      scripts/mod/empty.o
  DTC     arch/arm64/boot/dts/rockchip/rk3328-doornet1.dtb
  HOSTCC  scripts/mod/mk_elfconfig
  DTC     arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2c.dtb
arch/arm64/boot/dts/rockchip/rk3328-doornet1.dts:10:10: fatal error: rk3328-dram-nanopi2-timing.dtsi: No such file or directory
   10 | #include "rk3328-dram-nanopi2-timing.dtsi"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[7]: *** [scripts/Makefile.lib:347: arch/arm64/boot/dts/rockchip/rk3328-doornet1.dtb] Error 1
make[7]: *** Waiting for unfinished jobs....
  CC      scripts/mod/devicetable-offsets.s
In file included from arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2c.dts:11:
arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts:10:10: fatal error: rk3328-dram-nanopi2-timing.dtsi: No such file or directory
   10 | #include "rk3328-dram-nanopi2-timing.dtsi"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

LPDDR6-10000MHz, 2022-07-09:
具体点，切到哪个内核编译的？

lifetyper, 2022-07-09:
默认什么内核就什么内核

> 具体点，切到哪个内核编译的？

LPDDR6-10000MHz, 2022-07-09:
你把rk3328-dram-nanopi2-timing.dtsi算了，或者等下午我pr？

lifetyper, 2022-07-09:
> 你把rk3328-dram-nanopi2-timing.dtsi加上算了，或者等下午我pr？

我不会啊，大佬幸苦一下喽。感谢！

closed 2022-07-09 by coolsnowwolf