coolsnowwolf/lede #9721: x86-5.18内核

xiangfeidexiaohuo, 2022-07-09:
x86-5.18内核，kmod-fs-cifs编译失败

Package kmod-fs-cifs is missing dependencies for the following libraries:
asn1_decoder.ko
dns_resolver.ko
oid_registry.ko

ricksuzade-maker, 2022-07-09:
Packaged contents of /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-ecb into /home/zhiliang/scripts/neo3/lede/bin/targets/rockchip/armv8/packages/kmod-crypto-ecb_5.18.10-1_aarch64_generic.ipk
rstrip.sh: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-echainiv/lib/modules/5.18.10/echainiv.ko: relocatable
Packaged contents of /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-echainiv into /home/zhiliang/scripts/neo3/lede/bin/targets/rockchip/armv8/packages/kmod-crypto-echainiv_5.18.10-1_aarch64_generic.ipk
NOTICE: module '/home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.18.10/crypto/gf128mul.ko' is built-in.
Packaged contents of /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-gf128 into /home/zhiliang/scripts/neo3/lede/bin/targets/rockchip/armv8/packages/kmod-crypto-gf128_5.18.10-1_aarch64_generic.ipk
NOTICE: module '/home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.18.10/crypto/ghash-generic.ko' is built-in.
Packaged contents of /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-ghash into /home/zhiliang/scripts/neo3/lede/bin/targets/rockchip/armv8/packages/kmod-crypto-ghash_5.18.10-1_aarch64_generic.ipk
NOTICE: module '/home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.18.10/crypto/gcm.ko' is built-in.
Packaged contents of /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-gcm into /home/zhiliang/scripts/neo3/lede/bin/targets/rockchip/armv8/packages/kmod-crypto-gcm_5.18.10-1_aarch64_generic.ipk
Packaged contents of /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-kpp into /home/zhiliang/scripts/neo3/lede/bin/targets/rockchip/armv8/packages/kmod-crypto-kpp_5.18.10-1_aarch64_generic.ipk
rstrip.sh: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-lib-chacha20/lib/modules/5.18.10/libchacha.ko: relocatable
rstrip.sh: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-lib-chacha20/lib/modules/5.18.10/chacha-neon.ko: relocatable
Packaged contents of /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-lib-chacha20 into /home/zhiliang/scripts/neo3/lede/bin/targets/rockchip/armv8/packages/kmod-crypto-lib-chacha20_5.18.10-1_aarch64_generic.ipk
rstrip.sh: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-lib-poly1305/lib/modules/5.18.10/poly1305-neon.ko: relocatable
Packaged contents of /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-lib-poly1305 into /home/zhiliang/scripts/neo3/lede/bin/targets/rockchip/armv8/packages/kmod-crypto-lib-poly1305_5.18.10-1_aarch64_generic.ipk
rstrip.sh: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-lib-chacha20poly1305/lib/modules/5.18.10/libchacha20poly1305.ko: relocatable
Packaged contents of /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-lib-chacha20poly1305 into /home/zhiliang/scripts/neo3/lede/bin/targets/rockchip/armv8/packages/kmod-crypto-lib-chacha20poly1305_5.18.10-1_aarch64_generic.ipk
rstrip.sh: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-lib-curve25519/lib/modules/5.18.10/libcurve25519-generic.ko: relocatable
rstrip.sh: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-lib-curve25519/lib/modules/5.18.10/libcurve25519.ko: relocatable
Packaged contents of /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-lib-curve25519 into /home/zhiliang/scripts/neo3/lede/bin/targets/rockchip/armv8/packages/kmod-crypto-lib-curve25519_5.18.10-1_aarch64_generic.ipk
rstrip.sh: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-md4/lib/modules/5.18.10/md4.ko: relocatable
Packaged contents of /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-md4 into /home/zhiliang/scripts/neo3/lede/bin/targets/rockchip/armv8/packages/kmod-crypto-md4_5.18.10-1_aarch64_generic.ipk
rstrip.sh: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/packages/ipkg-aarch64_generic/kmod-crypto-md5/lib/modules/5.18.10/md5.ko: relocatable
[... cut ...]
time: package/kernel/linux/compile#7.59#1.05#7.31
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/zhiliang/scripts/neo3/lede'
make[1]: *** [package/Makefile:110: /home/zhiliang/scripts/neo3/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhiliang/scripts/neo3/lede'
make: *** [/home/zhiliang/scripts/neo3/lede/include/toplevel.mk:230: world] Error 2

ricksuzade-maker, 2022-07-09:
我也遇到同样问题了

ricksuzade-maker, 2022-07-09:
不过我是编译的rockchip arm64

bookwm, 2022-07-09:
我也遇到同样问题了

coolsnowwolf, 2022-07-09:
已修复

closed 2022-07-09 by coolsnowwolf