coolsnowwolf/lede #9723: x86 5.18 kernel compile error

GreatMichaelLee, 2022-07-09:
need-modorder=1
  x86_64-openwrt-linux-musl-gcc -Wp,-MMD,/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/.xt_LUA_target.o.d  -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu11 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=branch -fno-jump-tables -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -O2 -fno-caller-saves --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector -Wimplicit-fallthrough=5 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-stack-clash-protection -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-truncation -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -g -gdwarf-5 -fno-var-tracking -femit-struct-debug-baseonly -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl=target-x86_64_musl -DDEBUG -I/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/prot_buf_new -I/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/lua -I/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/lua/include  -DMODULE  -DKBUILD_BASENAME='"xt_LUA_target"' -DKBUILD_MODNAME='"xt_LUA"' -D__KBUILD_MODNAME=kmod_xt_LUA -c -o /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/xt_LUA_target.o /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/xt_LUA_target.c 
In file included from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/controller.h:24,
                 from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/xt_LUA_target.c:27:
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/lua/lua.h:12:10: fatal error: stdarg.h: No such file or directory
 #include <stdarg.h>
          ^~~~~~~~~~
compilation terminated.
make[9]: *** [scripts/Makefile.build:288: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/xt_LUA_target.o] Error 1
make[8]: *** [scripts/Makefile.build:550: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA] Error 2
make[7]: *** [Makefile:1838: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions] Error 2
make[7]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.10'
make[6]: *** [Makefile:461: modules] Error 2
make[6]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions'
make[5]: *** [Makefile:495: all-recursive] Error 1
make[5]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
make[4]: *** [Makefile:380: all] Error 2
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
make[3]: *** [Makefile:177: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#1.45#0.88#2.18
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/xtables-addons/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_x86$

minlang112, 2022-07-09:
stdarg.h 是GCC的一部分，你的编译环境有问题吧

GreatMichaelLee, 2022-07-09:
> stdarg.h 是GCC的一部分，你的编译环境有问题吧

开什么玩笑，我这build server都跑了一年了，编译5.15立马好。

GreatMichaelLee, 2022-07-10:
有人修一下这个问题吗，我确定不是环境的问题，在两台机器上都出这个

WYC-2020, 2022-07-10:
我可以说我今天编译全部通过吗哈哈

GreatMichaelLee, 2022-07-11:
> 我可以说我今天编译全部通过吗哈哈

嗯嗯，我知道，你玩的是5.4，哈哈

GreatMichaelLee, 2022-07-11:
这个对应Menu config里哪个组件，不行我取消掉算了，奇怪在5.15下面好好的，同一个.config，不知道为啥到了5.18下就不行
[image]

GreatMichaelLee, 2022-07-11:
跟kernel 5.18应该还是有关的，但不懂怎么弄。。。。

```
make[4]: Entering directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
make  all-recursive
make[5]: Entering directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
Making all in extensions
make[6]: Entering directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions'
Xtables-addons 3.21 - Linux 5.18.10
if [ -n "/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.10" ]; then make -C /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.10 M=/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions modules; fi;
make[7]: Entering directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.10'
**test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
make -f ./scripts/Makefile.build obj=/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions \**
single-build= \
need-builtin=1 need-modorder=1
make -f ./scripts/Makefile.build obj=/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA \
 \
need-builtin= \
need-modorder=1
  x86_64-openwrt-linux-musl-gcc -Wp,-MMD,/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/.xt_LUA_target.o.d  -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -Wall -Wundef -Werror=strict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -fno-PIE -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Wno-format-security -std=gnu11 -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=branch -fno-jump-tables -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -fno-jump-tables -fno-delete-null-pointer-checks -Wno-frame-address -Wno-format-truncation -Wno-format-overflow -O2 -fno-caller-saves --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector -Wimplicit-fallthrough=5 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-stack-clash-protection -Wdeclaration-after-statement -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-stringop-truncation -Wno-stringop-overflow -Wno-restrict -Wno-maybe-uninitialized -fno-strict-overflow -fno-stack-check -fconserve-stack -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wno-packed-not-aligned -g -gdwarf-5 -fno-var-tracking -femit-struct-debug-baseonly -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl=target-x86_64_musl -DDEBUG -I/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/prot_buf_new -I/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/lua -I/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/lua/include  -DMODULE  -DKBUILD_BASENAME='"xt_LUA_target"' -DKBUILD_MODNAME='"xt_LUA"' -D__KBUILD_MODNAME=kmod_xt_LUA -c -o /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/xt_LUA_target.o /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/xt_LUA_target.c 
In file included from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/controller.h:24,
                 from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/xt_LUA_target.c:27:
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/lua/lua.h:12:10: fatal error: stdarg.h: No such file or directory
 #include <stdarg.h>
          ^~~~~~~~~~
compilation terminated.
make[9]: *** [scripts/Makefile.build:288: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/xt_LUA_target.o] Error 1
make[8]: *** [scripts/Makefile.build:550: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA] Error 2
make[7]: *** [Makefile:1838: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions] Error 2
make[7]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.10'
make[6]: *** [Makefile:461: modules] Error 2
make[6]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions'
[... cut ...]
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/xtables-addons/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2
builder@Build-Server:~/lede_x86$ 
builder@Build-Server:~/lede_x86$
```

WYC-2020, 2022-07-11:
> > 我可以说我今天编译全部通过吗哈哈
> 
> 嗯嗯，我知道，你玩的是5.4，哈哈

不，我所有内核都编译过没问题，5.18 5.15 5.10 5.4 我都编译我通过没问题

WYC-2020, 2022-07-12:
> 跟kernel 5.18应该还是有关的，但不懂怎么弄。。。。
>
> ```
> make[4]: Entering directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21'
> make  all-recursive
> make[5]: Entering directory [...]

[image]
 新鲜出炉的

GreatMichaelLee, 2022-07-12:
> [image] 新鲜出炉的

哭死，我的就是不行阿，一毛一样的.config（除了一个用test kernel -5.18,一个不勾选-5.15），其他都一毛一样，5.15没任何问题，5.18就是不行阿....而且在两个不同的build server上都是如此....

WYC-2020, 2022-07-12:
你这个应该是那个库 有问题 没有适配5.18 ，我是编译的 我那套config 配置完全没问题

WYC-2020, 2022-07-12:
你这个是单线程日志吗？是的话 应该是xtables-addons-3.21这个的问题，我等下给你看下吧 @GreatMichaelLee

GreatMichaelLee, 2022-07-12:
> 你这个是单线程日志吗？是的话 应该是xtables-addons-3.21这个的问题，我等下给你看下吧 @GreatMichaelLee

是单线程日志，试过 rm -rf ./tmp && make clean 和删除dl目录，完全干净单线程编译，也是报这个错。

WYC-2020, 2022-07-12:
> > 你这个是单线程日志吗？是的话 应该是xtables-addons-3.21这个的问题，我等下给你看下吧 @GreatMichaelLee
> 
> 是单线程日志，试过 rm -rf ./tmp && make clean 和删除dl目录，完全干净单线程编译，也是报这个错。

已修好等我验证另外几个内核看有没有影响

WYC-2020, 2022-07-12:
[patch.zip](https://github.com/coolsnowwolf/lede/files/9089629/patch.zip)
[image]
自己拿去替换xtable目录下的相同patches,我这边测试所有内核都可以正常编译，不信自测

GreatMichaelLee, 2022-07-12:
喜大普奔，过了过了，求PR, 
btw: 这是为毛线阿？头文件include为啥在不同内核编译有的能过有的不能？

WYC-2020, 2022-07-12:
哪来那么多为什么，这不是很正常，猜测可能是相互包含导致，懒得去给内核打补丁没时间

GreatMichaelLee, 2022-07-13:
老板，PR一下阿

coolsnowwolf, 2022-07-13:
fixed

closed 2022-07-13 by coolsnowwolf