coolsnowwolf/lede #9725: KMS编译问题

Jejz168, 2022-07-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：只要编译kms就失败

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：make[3]: Entering directory '/workdir/openwrt/feeds/packages/net/vlmcsd'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.vlmcsd-svn1113.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "vlmcsd-svn1113.tar.gz" "62f55c48f5de1249c2348ab6b96dabbe7e38899230954b0c8774efb01d9c42cc" "" "https://codeload.github.com/Wind4/vlmcsd/tar.gz/svn1113?"    '
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113/.built
touch /workdir/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113=vlmcsd-svn1113 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113=vlmcsd-svn1113 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro " make  -C /workdir/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113=vlmcsd-svn1113 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" -C /workdir/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113 ;
make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113'
make[5]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113/src'
	CC	vlmcs.o <- vlmcs.c
	CC	kmsdata-full.o <- kmsdata-full.c
<command-line>: fatal error: X86: No such file or directory
compilation terminated.
	CC	crypto.o <- crypto.c
make[5]: *** [GNUmakefile:513: ../build/vlmcs.o] Error 1
make[5]: *** Waiting for unfinished jobs....
<command-line>: fatal error: X86: No such file or directory
compilation terminated.
make[5]: *** [GNUmakefile:513: ../build/kmsdata-full.o] Error 1
<command-line>: fatal error: X86: No such file or directory
compilation terminated.
make[5]: *** [GNUmakefile:513: ../build/crypto.o] Error 1
make[4]: *** [GNUmakefile:111: all] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113/src'
make[3]: *** [Makefile:44: /workdir/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113/.built] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/vlmcsd-svn1113'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/vlmcsd'
time: package/feeds/packages/vlmcsd/compile#0.20#0.05#0.22
    ERROR: package/feeds/packages/vlmcsd failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/vlmcsd/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

Jejz168, 2022-07-11:
有没有一样问题的，帮我看看怎么解决

WYC-2020, 2022-07-11:
问题描述都不清楚，什么内核 又不说

Jejz168, 2022-07-11:
> 问题又描述都不是什么，什么问题

5.15的

Jejz168, 2022-07-11:
> 问题又描述都不是什么，什么问题

这是和什么冲突吗？？还是哪里不对？？

WYC-2020, 2022-07-12:
> > 问题又描述都不是什么，什么问题
> 
> 5.15的。编译kms就失败。上面是错误的提示

我这边一切正常，建议万能大法后 在试下

xiaojing110, 2022-07-12:
5.15   5.18都没问题 这个我一般都是勾选的  建议万能大法

ecrasy, 2022-07-15:
fatal error: X86: No such file or directory
出现这种情况，一般两种可能：网络问题没下载完全或者磁盘空间不足生成失败

Jejz168, 2022-07-15:
> fatal error: X86: No such file or directory
> 出现这种情况，一般两种可能：网络问题没下载完全或者磁盘空间不足生成失败

还真不是。云编译。一模一样的yml我的不行。别人的可以。我重新写了份就好了。真的很奇

closed 2022-07-18 by Jejz168