coolsnowwolf/lede #9726: arm64编译mac80211时出错，5.18内核

ricksuzade-maker, 2022-07-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
编译mac80211时出错，5.18新版内核，arm64, rockchip neo3
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging/rtl8723bs/core/rtw_cmd.o
In file included from /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging/rtl8723bs/include/osdep_service_linux.h:41,
                 from /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging/rtl8723bs/include/osdep_service.h:15,
                 from /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging/rtl8723bs/include/drv_types.h:19,
                 from /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging/rtl8723bs/core/rtw_cmd.c:7:
/home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/include/net/cfg80211.h: In function 'wdev_address':
/home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/include/net/cfg80211.h:5561:22: warning: return discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   return wdev->netdev->dev_addr;
          ~~~~~~~~~~~~^~~~~~~~~~
In file included from /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging/rtl8723bs/include/osdep_service.h:15,
                 from /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging/rtl8723bs/include/drv_types.h:19,
                 from /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging/rtl8723bs/core/rtw_cmd.c:7:
/home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging/rtl8723bs/core/rtw_cmd.c: In function 'rtw_cmd_thread':
/home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging/rtl8723bs/include/osdep_service_linux.h:48:24: error: implicit declaration of function 'complete_and_exit'; did you mean 'complete_all'? [-Werror=implicit-function-declaration]
  #define thread_exit() complete_and_exit(NULL, 0)
                        ^~~~~~~~~~~~~~~~~
/home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging/rtl8723bs/core/rtw_cmd.c:527:2: note: in expansion of macro 'thread_exit'
  thread_exit();
  ^~~~~~~~~~~
/home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging/rtl8723bs/core/rtw_cmd.c:528:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: some warnings being treated as errors
make[10]: *** [scripts/Makefile.build:288: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging/rtl8723bs/core/rtw_cmd.o] Error 1
make[9]: *** [scripts/Makefile.build:550: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging/rtl8723bs] Error 2
make[8]: *** [scripts/Makefile.build:550: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/staging] Error 2
make[7]: *** [Makefile:1838: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1'
make[3]: *** [Makefile:559: /home/zhiliang/scripts/neo3/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/.built] Error 2
make[3]: Leaving directory '/home/zhiliang/scripts/neo3/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#1.43#0.32#1.65
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/zhiliang/scripts/neo3/lede'
make[1]: *** [package/Makefile:110: /home/zhiliang/scripts/neo3/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhiliang/scripts/neo3/lede'
make: *** [/home/zhiliang/scripts/neo3/lede/include/toplevel.mk:230: world] Error 2

coolsnowwolf, 2022-07-10:
已修复 [commit id:](https://github.com/coolsnowwolf/lede/commit/86ea10215695e5f8e2cabdeac90d7f6f9a620767)

closed 2022-07-11 by coolsnowwolf

aranyalau, 2022-08-12:
编译N1固件的时候也出现了