coolsnowwolf/lede #9728: 5.18 内核编译ntfs3-oot失败

HawaiiGit, 2022-07-09:
touch /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/.prepared_ea26e9fd594e15fa590bf3a0df08aa12_6664517399ebbbc92a37c5bb081b5c53_check
. /home/butcher/lede/include/shell.sh; xzcat /home/butcher/lede/dl/ntfs3-oot-2021-07-05-46d199e7.tar.xz | tar -C /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7

Applying ./patches/100-compat-mount.patch using plaintext: 
patching file super.c
touch /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/.prepared_ea26e9fd594e15fa590bf3a0df08aa12_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/.configured_*
rm -f /home/butcher/lede/staging_dir/target-x86_64_musl/stamp/.ntfs3-oot_installed
(cd /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/./; if [ -x ./configure ]; then find /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/ -name config.guess | xargs -r chmod u+w; find /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/ -name config.guess | xargs -r -n1 cp --remove-destination /home/butcher/lede/scripts/config.guess; find /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/ -name config.sub | xargs -r chmod u+w; find /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/ -name config.sub | xargs -r -n1 cp --remove-destination /home/butcher/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7=ntfs3-oot-2021-07-05-46d199e7 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7=ntfs3-oot-2021-07-05-46d199e7 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7=ntfs3-oot-2021-07-05-46d199e7 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/butcher/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/butcher/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/butcher/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/butcher/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/butcher/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/.built
touch /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/.built_check
make -C /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.10 KCFLAGS="-fmacro-prefix-map=/home/butcher/lede/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/butcher/lede/staging_dir/host/include -I/home/butcher/lede/staging_dir/hostpkg/include -I/home/butcher/lede/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Jul  9 08:28:28 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/butcher/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.18.10 M="/home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7" EXTRA_CFLAGS=""  CONFIG_NTFS3_FS=m modules
make[4]: Entering directory '/home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.10'
  CC [M]  /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/attrib.o
  CC [M]  /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/attrlist.o
  CC [M]  /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/bitfunc.o
  CC [M]  /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/bitmap.o
  CC [M]  /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/dir.o
  CC [M]  /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/fsntfs.o
In file included from ./include/linux/xarray.h:15,
                 from ./include/linux/list_lru.h:14,
                 from ./include/linux/fs.h:13,
                 from ./include/linux/highmem.h:5,
                 from ./include/linux/bvec.h:10,
                 from ./include/linux/blk_types.h:10,
                 from ./include/linux/blkdev.h:9,
                 from /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/fsntfs.c:8:
/home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/fsntfs.c: In function 'ntfs_alloc_bio':
./include/linux/gfp.h:345:18: warning: passing argument 1 of 'bio_alloc' makes pointer from integer without a cast [-Wint-conversion]
 #define GFP_NOFS (__GFP_RECLAIM | __GFP_IO)
                  ^
/home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/fsntfs.c:1485:30: note: in expansion of macro 'GFP_NOFS'
  struct bio *bio = bio_alloc(GFP_NOFS | __GFP_HIGH, nr_vecs);
                              ^~~~~~~~
In file included from ./include/linux/blkdev.h:17,
                 from /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/fsntfs.c:8:
./include/linux/bio.h:423:58: note: expected 'struct block_device *' but argument is of type 'unsigned int'
 static inline struct bio *bio_alloc(struct block_device *bdev,
                                     ~~~~~~~~~~~~~~~~~~~~~^~~~
/home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/fsntfs.c:1485:20: error: too few arguments to function 'bio_alloc'
  struct bio *bio = bio_alloc(GFP_NOFS | __GFP_HIGH, nr_vecs);
                    ^~~~~~~~~
In file included from ./include/linux/blkdev.h:17,
                 from /home/butcher/lede/build_dir/target-x86_64_musl/linux-x86_64/ntfs3-oot-2021-07-05-46d199e7/fsntfs.c:8:
./include/linux/bio.h:423:27: note: declared here
 static inline struct bio *bio_alloc(struct block_device *bdev,
                           ^~~~~~~~~
In file included from ./include/linux/xarray.h:15,
                 from ./include/linux/list_lru.h:14,
                 from ./include/linux/fs.h:13,
                 from ./include/linux/highmem.h:5,
                 from ./include/linux/bvec.h:10,
                 from ./include/linux/blk_types.h:10,
                 from ./include/linux/blkdev.h:9,
[... cut ...]
make[3]: Leaving directory '/home/butcher/lede/package/lean/ntfs3-oot'
time: package/lean/ntfs3-oot/compile#1.97#0.21#2.15
    ERROR: package/lean/ntfs3-oot failed to build.
make[2]: *** [package/Makefile:116: package/lean/ntfs3-oot/compile] Error 1
make[2]: Leaving directory '/home/butcher/lede'
make[1]: *** [package/Makefile:110: /home/butcher/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/butcher/lede'
make: *** [/home/butcher/lede/include/toplevel.mk:230：world] 错误 2

xiangfeidexiaohuo, 2022-07-09:
先跳过吧 不管ntfs

aiamadeus, 2022-07-10:
5.15 开始都内置 ntfs3 了，还要 oot 版本干啥。

closed 2022-07-10 by aiamadeus

HawaiiGit, 2022-07-10:
> 5.15 开始都内置 ntfs3 了，还要 oot 版本干啥。

5.15默认又ntfs3，5.18编译时只有ntfs和ntfs3-oot两个ntfs选项，选ntfs硬盘read-only，选ntfs3-oot无法编译，如果又ntfs3选项这个issue就不会提了

WYC-2020, 2022-07-10:
这玩意我很早就发现有问题，特别是编译了5.4然后又去编译5.15就会有问题，我这边直接统一了所有的内核都使用oot目前没任何问题，刚修复了5.18