coolsnowwolf/lede #9733: 新三最近编译的固件都是无法启动，

yuzhe1129, 2022-07-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

插件我已经去掉了很多了，都没选什么插件，固件编译才10M大小，最近很多天编译的固件都无法启动 开始我以为我加别的源码进来问题的，可这次我都没选别的插件编译刷也是启动不了，

### (2) 路由器型号和固件版本
新三 D2
A：

### (3) 详细日志

Boot and Recovery Environment for Embedded Devices
Copyright (C) 2021 HackPascal <hackpascal@gmail.com>
Build date 2021-12-15 [git-f9b74d0]
Version 1.1 (r1337)

DRAM: 512MB
Platform: MediaTek MT7621A ver 1, eco 3
Board: Newifi D2
Clocks: CPU: 880MHz, DDR: 1040MHz, Bus: 220MHz, Ref: 40MHz
Flash: Macronix MX25L25635E (32MB) on mt7621-spi.0
rt2880-eth: MAC address from EEPROM is invalid, using default settings.
rt2880-eth: Using MAC address 00:0c:43:00:00:01
eth0: MediaTek MT7530 Gigabit switch

Network started on eth0, inet addr 192.168.1.1, netmask 255.255.255.0

Press any key to interrupt autoboot ... 0

Trying to boot firmware from 0x00050000 in flash bank 0 ...
Reading data into memory ...
U-Boot firmware image header detected.
    Image Name:   MIPS OpenWrt Linux-5.4.203
    Data Size:    2543830 Bytes
    Load Address: 80001000
    Entry Point:  80001000
Flushing cache ... done.

Starting kernel at 0x80001000...

OpenWrt kernel loader for MIPS based SoC
Copyright (C) 2011 Gabor Juhos <juhosg@openwrt.org>
Decompressing kernel... done!
Starting kernel at 80001000...

[    0.000000] Linux version 5.4.203 (bbb@ubuntu) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4646-486587518)) #0 SMP Sun Jul 10 10:13:31 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Newifi-D2
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[... 150 lines cut ...]
[    3.061503] pci 0000:02:00.0: BAR 0: assigned [mem 0x60200000-0x602fffff]
[    3.068278] pci 0000:00:01.0: PCI bridge to [bus 02]
[    3.073218] pci 0000:00:01.0:   bridge window [io  0x1e161000-0x1e161fff]
[    3.079990] pci 0000:00:01.0:   bridge window [mem 0x60200000-0x602fffff]
[    3.086764] pci 0000:00:01.0:   bridge window [mem 0x60300000-0x603fffff pref]
[    3.094751] hctosys: unable to open rtc device (rtc0)
[    3.100601] /dev/root: Can't open blockdev
[    3.104722] VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
[    3.112164] Please append a correct "root=" boot option; here are the available partitions:
[    3.120488] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    3.128751] Rebooting in 1 seconds..

Boot and Recovery Environment for Embedded Devices
Copyright (C) 2021 HackPascal <hackpascal@gmail.com>
Build date 2021-12-15 [git-f9b74d0]
Version 1.1 (r1337)

DRAM: 512MB
Platform: MediaTek MT7621A ver 1, eco 3
Board: Newifi D2
Clocks: CPU: 880MHz, DDR: 1040MHz, Bus: 220MHz, Ref: 40MHz
Flash: Macronix MX25L25635E (32MB) on mt7621-spi.0
rt2880-eth: MAC address from EEPROM is invalid, using default settings.
rt2880-eth: Using MAC address 00:0c:43:00:00:01
eth0: MediaTek MT7530 Gigabit switch

coolsnowwolf, 2022-07-10:
内核找不到 spi 上面的 rootfs 分区

yuzhe1129, 2022-07-10:
> 内核找不到 spi 上面的 rootfs 分区

那么是我编译的问题还是什么？

aiamadeus, 2022-07-11:
你新三的 spi 闪存出问题了吧

uniessey, 2022-07-11:
> > 内核找不到 spi 上面的 rootfs 分区
> 
> 那么是我编译的问题还是什么？

我自己遇到过的情况而言，要么是编译的时候出了意外，要么是硬件上有问题了，试试全部删除重新拉取，如果能一次拉取成功，那么一般不会出什么大问题，如果拉取过程总是失败，那么还是有可能出问题的。

yuzhe1129, 2022-07-12:
> > > 内核找不到 spi 上面的 rootfs 分区
> > 
> > 
> > 那么是我编译的问题还是什么？
> 
> 我自己遇到过的情况而言，要么是编译的时候出了意外，要么是硬件上有问题了，试试全部删除重新拉取，如果能一次拉取成功，那么一般不会出什么大问题，如果拉取过程总是失败，那么还是有可能出问题的。

硬件百分百没问题，路由我都正在用，刷老毛子 6.25号的我编译有是可以启动的，源码已经删除重新下载过了的，

harrylwc, 2022-07-12:
https://github.com/openwrt/openwrt/issues/10232

uniessey, 2022-07-12:
> > > > 内核找不到 spi 上面的 rootfs 分区
> > > 
> > > 
> > > 那么是我编译的问题还是什么？
> > 
> > 
> > 我自己遇到过的情况而言，要么是编译的时候出了意外，要么是硬件上有问题了，试试全部删除重新拉取，如果能一次拉取成功，那么一般不会出什么大问题，如果拉取过程总是失败，那么还是有可能出问题的。
> 
> 硬件百分百没问题，路由我都正在用，刷老毛子 6.25号的我编译有是可以启动的，源码已经删除重新下载过了的，

我昨天晚上19：20左右git pull，21：00左右刷进D2，至少到今天早上没发现问题，我继续用用看。

hackpascal, 2022-07-15:
> [ 0.866369] spi-nor spi0.0: unrecognized JEDEC id bytes: c2 20 1d c2 30 1d

flash没认出来。
根据breed的启动信息，Macronix MX25L25635E对应的的jedec id是c2 20 19，而kernel输出的读取两次的id都不一样，大概率是spi总线不稳，例如频率太高。

yuzhe1129, 2022-07-24:
> > [ 0.866369] spi-nor spi0.0: unrecognized JEDEC id bytes: c2 20 1d c2 30 1d
> 
> flash没认出来。 根据breed的启动信息，Macronix MX25L25635E对应的的jedec id是c2 20 19，而kernel输出的读取两次的id都不一样，大概率是spi总线不稳，例如频率太高。

闪存正常的啊，你看我楼上的回复6.25编译的固件是正常可以启动的，然后我用了另外一个正在使用OP固件的一台刷了这个固件也是无法启动的

hackpascal, 2022-07-24:
> > > [ 0.866369] spi-nor spi0.0: unrecognized JEDEC id bytes: c2 20 1d c2 30 1d
> > 
> > 
> > flash没认出来。 根据breed的启动信息，Macronix MX25L25635E对应的的jedec id是c2 20 19，而kernel输出的读取两次的id都不一样，大概率是spi总线不稳，例如频率太高。
> 
> 闪存正常的啊，你看我楼上的回复6.25编译的固件是正常可以启动的，然后我用了另外一个正在使用OP固件的一台刷了这个固件也是无法启动的

我也没说你的闪存有问题啊。我指的是固件里面的spi通信可能有问题

aiamadeus, 2022-07-25:
> [ 0.000000] CPU Overclock PLL: 0xc2005b12
> [ 0.000000] CPU Clock: 1000MHz

现在超频已经去掉了，重新编译就完事了。

closed 2022-07-25 by yuzhe1129