coolsnowwolf/lede #9740: 最新版默认编译失败

eearphon, 2022-07-12:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：18.04.6 LTS，直接默认编译就失败，非新手，以前编译成功过的

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：
insn-output.o:(.rodata+0x4d710): undefined reference to `gen_vpdpwssd_v16si_maskz(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4d738): undefined reference to `gen_vpdpwssd_v8si_maskz(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4d760): undefined reference to `gen_vpdpwssd_v4si_maskz(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4d788): undefined reference to `gen_vpdpwssds_v16si_maskz(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4d7b0): undefined reference to `gen_vpdpwssds_v8si_maskz(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4d7d8): undefined reference to `gen_vpdpwssds_v4si_maskz(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4d800): undefined reference to `gen_sse2_lfence()'
insn-output.o:(.rodata+0x4d828): undefined reference to `gen_sse_sfence()'
insn-output.o:(.rodata+0x4d850): undefined reference to `gen_sse2_mfence()'
insn-output.o:(.rodata+0x4d878): undefined reference to `gen_mem_thread_fence(rtx_def*)'
insn-output.o:(.rodata+0x4d8a0): undefined reference to `gen_atomic_loadqi(rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4d8c8): undefined reference to `gen_atomic_loadhi(rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4d8f0): undefined reference to `gen_atomic_loadsi(rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4d918): undefined reference to `gen_atomic_loaddi(rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4d940): undefined reference to `gen_atomic_storeqi(rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4d968): undefined reference to `gen_atomic_storehi(rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4d990): undefined reference to `gen_atomic_storesi(rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4d9b8): undefined reference to `gen_atomic_storedi(rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4d9e0): undefined reference to `gen_atomic_compare_and_swapqi(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4da08): undefined reference to `gen_atomic_compare_and_swaphi(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4da30): undefined reference to `gen_atomic_compare_and_swapsi(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4da58): undefined reference to `gen_atomic_compare_and_swapdi(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4da80): undefined reference to `gen_atomic_compare_and_swapti(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4daa8): undefined reference to `gen_atomic_bit_test_and_sethi(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4dad0): undefined reference to `gen_atomic_bit_test_and_setsi(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4daf8): undefined reference to `gen_atomic_bit_test_and_setdi(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4db20): undefined reference to `gen_atomic_bit_test_and_complementhi(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4db48): undefined reference to `gen_atomic_bit_test_and_complementsi(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4db70): undefined reference to `gen_atomic_bit_test_and_complementdi(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4db98): undefined reference to `gen_atomic_bit_test_and_resethi(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4dbc0): undefined reference to `gen_atomic_bit_test_and_resetsi(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
insn-output.o:(.rodata+0x4dbe8): undefined reference to `gen_atomic_bit_test_and_resetdi(rtx_def*, rtx_def*, rtx_def*, rtx_def*, rtx_def*)'
combine.o: In function `recog_for_combine_1(rtx_def**, rtx_insn*, rtx_def**)':
/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/combine.c:11426: undefined reference to `add_clobbers(rtx_def*, int)'
modulo-sched.o: In function `ps_has_conflicts(partial_schedule*, int, int)':
/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/modulo-sched.c:3158: undefined reference to `state_reset(void*)'
/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/modulo-sched.c:3177: undefined reference to `state_dead_lock_p(void*)'
/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/modulo-sched.c:3182: undefined reference to `state_transition(void*, rtx_def*)'
modulo-sched.o: In function `advance_one_cycle':
/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/modulo-sched.c:3138: undefined reference to `state_transition(void*, rtx_def*)'
/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/modulo-sched.c:3141: undefined reference to `state_transition(void*, rtx_def*)'
/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/modulo-sched.c:3144: undefined reference to `state_transition(void*, rtx_def*)'
modulo-sched.o: In function `schedule_reg_move':
/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/modulo-sched.c:589: undefined reference to `insn_latency(rtx_insn*, rtx_insn*)'
/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/modulo-sched.c:610: undefined reference to `insn_latency(rtx_insn*, rtx_insn*)'
collect2: error: ld returned 1 exit status
/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0/gcc/c/Make-lang.in:85: recipe for target 'cc1' failed
make[6]: *** [cc1] Error 1
make[6]: Leaving directory '/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/gcc'
Makefile:4270: recipe for target 'all-gcc' failed
make[5]: *** [all-gcc] Error 2
make[5]: Leaving directory '/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final'
Makefile:905: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final'
Makefile:89: recipe for target '/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/.built' failed
make[3]: *** [/opt/openwrt/lede/build_dir/toolchain-x86_64_gcc-8.4.0_musl/gcc-8.4.0-final/.built] Error 2
make[3]: Leaving directory '/opt/openwrt/lede/toolchain/gcc/final'
time: toolchain/gcc/final/compile#7.23#2.63#26.69
    ERROR: toolchain/gcc/final failed to build.
toolchain/Makefile:95: recipe for target 'toolchain/gcc/final/compile' failed
make[2]: *** [toolchain/gcc/final/compile] Error 1
make[2]: Leaving directory '/opt/openwrt/lede22'
toolchain/Makefile:93: recipe for target '/opt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile' failed
make[1]: *** [/opt/openwrt/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/opt/openwrt/lede22'
/opt/openwrt/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

coolsnowwolf, 2022-07-12:
ubuntu 版本太老了

WYC-2020, 2022-07-12:
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
>
> * [x ] 没有类似的issue
>
> ## 2. 详细叙述
> ### (1) 具体问题
> A：18.04.6 LTS，直接默认编译就失败，非新手，以前编译成功过的
>
> ### (2) 路由器型号和固件版本
> A：x86-64
>
> ### (3) 详细日志
> A： [...]

我18.04.5  没有任何问题编译，但是gcc我是11.2

xyskwjy, 2022-07-12:
1.关于你要提交的问题
Q：是否搜索了issue (使用 "x" 选择)

[x ] 没有类似的issue
2. 详细叙述
(1) 具体问题
A：ubuntu20.04，直接默认编译就失败，非新手，以前编译成功过的

(2) 路由器型号和固件版本
A：x86-64

(3) 详细日志
 make[3] -C package/lean/igb-intel compile
    ERROR: package/lean/igb-intel failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 1
...........................................................................................
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/tools'
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14'
touch /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.built
rm -rf /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.pkgdir/libdevmapper.installed /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.pkgdir/libdevmapper
mkdir -p /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.pkgdir/libdevmapper
install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.pkgdir/libdevmapper/usr/lib
cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-install/usr/lib/libdevmapper.so.* /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.pkgdir/libdevmapper/usr/lib
touch /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.pkgdir/libdevmapper.installed
mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.pkgdir/libdevmapper/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.libdevmapper_installed
mkdir -p /workdir/openwrt/bin/targets/x86/64/packages /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper/CONTROL /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper/usr/lib
cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-install/usr/lib/libdevmapper.so.* /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper/usr/lib
find /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/workdir/openwrt/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/workdir/openwrt/scripts/strip-kmod.sh" PATCHELF="/workdir/openwrt/staging_dir/host/bin/patchelf" /workdir/openwrt/scripts/rstrip.sh /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper
rstrip.sh: /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper/usr/lib/libdevmapper.so.1.02: shared object
/workdir/openwrt/staging_dir/host/bin/sstrip: /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper/usr/lib/libdevmapper.so.1.02: Permission denied
(cd /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /workdir/openwrt/bin/packages/x86_64/packages
/workdir/openwrt/staging_dir/host/bin/fakeroot /workdir/openwrt/scripts/ipkg-build -m "" /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper /workdir/openwrt/bin/packages/x86_64/packages
Packaged contents of /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper into /workdir/openwrt/bin/packages/x86_64/packages/libdevmapper_2.03.14-1_x86_64.ipk
echo "libdevmapper" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/lvm2.normal.install
rm -rf /workdir/openwrt/tmp/stage-lvm2
mkdir -p /workdir/openwrt/tmp/stage-lvm2/host /workdir/openwrt/staging_dir/target-x86_64_musl/packages
install -d -m0755 /workdir/openwrt/tmp/stage-lvm2/usr/include
install -m0644 /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-install/usr/include/libdevmapper.h /workdir/openwrt/tmp/stage-lvm2/usr/include
install -d -m0755 /workdir/openwrt/tmp/stage-lvm2/usr/lib
cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-install/usr/lib/libdevmapper.so* /workdir/openwrt/tmp/stage-lvm2/usr/lib
install -d -m0755 /workdir/openwrt/tmp/stage-lvm2/usr/lib/pkgconfig
cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/libdm/libdevmapper.pc /workdir/openwrt/tmp/stage-lvm2/usr/lib/pkgconfig
[... cut ...]
touch /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.lvm2.normal_installed
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/lvm2'
time: package/feeds/packages/lvm2/normal/compile#54.48#6.31#59.81
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2022-07-12:
> 1.关于你要提交的问题 Q：是否搜索了issue (使用 "x" 选择)
> 
> [x ] 没有类似的issue 2. 详细叙述 (1) 具体问题 A：ubuntu20.04，直接默认编译就失败，非新手，以前编译成功过的
> 
> (2) 路由器型号和固件版本 A：x86-64
> 
> (3) 详细日志 make[3] -C package/lean/igb-intel compile ERROR: package/lean/igb-intel failed to build. make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 1 ........................................................................................... make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/tools' make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14' touch /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.built rm -rf /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.pkgdir/libdevmapper.installed /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.pkgdir/libdevmapper mkdir -p /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.pkgdir/libdevmapper install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.pkgdir/libdevmapper/usr/lib cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-install/usr/lib/libdevmapper.so.* /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.pkgdir/libdevmapper/usr/lib touch /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.pkgdir/libdevmapper.installed mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/.pkgdir/libdevmapper/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/' touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.libdevmapper_installed mkdir -p /workdir/openwrt/bin/targets/x86/64/packages /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper/CONTROL /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper/usr/lib cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-install/usr/lib/libdevmapper.so.* /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper/usr/lib find /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper -name 'CVS' -o -name '.svn' -o -name '.#_' -o -name '_~'| xargs -r rm -rf export CROSS="x86_64-openwrt-linux-musl-" NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/workdir/openwrt/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/workdir/openwrt/scripts/strip-kmod.sh" PATCHELF="/workdir/openwrt/staging_dir/host/bin/patchelf" /workdir/openwrt/scripts/rstrip.sh /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper rstrip.sh: /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper/usr/lib/libdevmapper.so.1.02: shared object /workdir/openwrt/staging_dir/host/bin/sstrip: /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper/usr/lib/libdevmapper.so.1.02: Permission denied (cd /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]_, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ "Misplaced & ${IPKG_NO_SCRIPT}" = "1" ] && exit 0"; echo "[ -s "${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". IPKGINSTROOT/lib/functions.sh";echo"defaultpostinst@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "IPKGINSTROOT/lib/functions.sh"]||exit0";echo".{IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm 0@"; ) > prerm; chmod 0755 postinst prerm; ) install -d -m0755 /workdir/openwrt/bin/packages/x86_64/packages /workdir/openwrt/staging_dir/host/bin/fakeroot /workdir/openwrt/scripts/ipkg-build -m "" /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper /workdir/openwrt/bin/packages/x86_64/packages Packaged contents of /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper into /workdir/openwrt/bin/packages/x86_64/packages/libdevmapper_2.03.14-1_x86_64.ipk echo "libdevmapper" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/lvm2.normal.install rm -rf /workdir/openwrt/tmp/stage-lvm2 mkdir -p /workdir/openwrt/tmp/stage-lvm2/host /workdir/openwrt/staging_dir/target-x86_64_musl/packages install -d -m0755 /workdir/openwrt/tmp/stage-lvm2/usr/include install -m0644 /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-install/usr/include/libdevmapper.h /workdir/openwrt/tmp/stage-lvm2/usr/include install -d -m0755 /workdir/openwrt/tmp/stage-lvm2/usr/lib cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-install/usr/lib/libdevmapper.so_ /workdir/openwrt/tmp/stage-lvm2/usr/lib install -d -m0755 /workdir/openwrt/tmp/stage-lvm2/usr/lib/pkgconfig cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/libdm/libdevmapper.pc /workdir/openwrt/tmp/stage-lvm2/usr/lib/pkgconfig find /workdir/openwrt/tmp/stage-lvm2 -name '_.la' | xargs - [... cut ...]  directory '/workdir/openwrt/feeds/packages/utils/lvm2' time: package/feeds/packages/lvm2/normal/compile#54.48#6.31#59.81 make[2]: Leaving directory '/workdir/openwrt' make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2 make[1]: Leaving directory '/workdir/openwrt' make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2 Error: Process completed with exit code 2.

非新手 都不知道贴单线程日志，只能看出package/lean/igb-intel failed to build.

xiechangan123, 2022-07-12:
跟楼上的日志一样，都是这个 package/lean/igb-intel 网卡驱动包有问题，在线编译，插件只有ssrp+，应该牵扯不到什么多线程，没clean之类的，希望尽快修复吧

make[3]: Entering directory '/workdir/openwrt/package/lean/igb-intel'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.igb-5.10.2.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "igb-5.10.2.tar.gz" "6ef2c26c18acb898d693040b56b27316" "" "http://downloads.sourceforge.net/project/e1000/igb%20stable/5.10.2/"    '
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/.built
touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/.built_check
KSRC="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.53"  make -j1 -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/src AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2=kmod-igb-intel-5.10.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" ;
make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/src'
common.mk:158: *** Linux kernel source not configured - missing autoconf.h.  Stop.
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/src'
make[3]: *** [Makefile:49: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/lean/igb-intel'
time: package/lean/igb-intel/compile#0.22#0.05#0.26
    ERROR: package/lean/igb-intel failed to build.
make[2]: *** [package/Makefile:116: package/lean/igb-intel/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

zhangguanzhang, 2022-07-12:
```
 * check_data_file_clashes: Package kmod-igb-intel wants to install file /workdir/openwrt/build_dir/target-x86_64_musl/root-x86/lib/modules/5.15.53/igb.ko
Configuring kmod-amd-xgbe.
	But that file is already provided by package  * kmod-igb
 * opkg_install_cmd: Cannot install package kmod-igb-intel.
make[2]: *** [package/Makefile:70: package/install] Error 255
```

xiangfeidexiaohuo, 2022-07-12:
> ```
>  * check_data_file_clashes: Package kmod-igb-intel wants to install file /workdir/openwrt/build_dir/target-x86_64_musl/root-x86/lib/modules/5.15.53/igb.ko
> Configuring kmod-amd-xgbe.
> 	But that [...]
现在默认用的kmod-igb-intel，但是以前默认用的kmod-igb，你们又是用的以前的配置文件，导致2个都选了，所以2个冲突，2个东西只能二选一。把kmod-igb 关掉就成。

eearphon, 2022-07-13:
> ubuntu 版本太老了

谢谢老大，换了 Ubuntu 22.04 LTS 就好了，Ubuntu 18就老了，够与时俱进！！！

eearphon, 2022-07-14:
> 我18.04.5 没有任何问题编译，但是gcc我是11.2
没注意看gcc版本，谢谢了！