coolsnowwolf/lede #9742: 最新源码编译失败

mirthlesslk, 2022-07-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

kmod-igb报错

Q：是否搜索了issue (使用 "x" 选择)

有类似的issue，内核版本不一样

## 2. 详细叙述

### (1) 具体问题

A：kmod-igb-intel报错，内核版本是5.10的，环境是ubuntu-20.04

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：make[3]: Entering directory '/workdir/openwrt/package/lean/igb-intel'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.igb-5.10.2.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "igb-5.10.2.tar.gz" "6ef2c26c18acb898d693040b56b27316" "" "http://downloads.sourceforge.net/project/e1000/igb%20stable/5.10.2/"    '
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/.built
touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/.built_check
KSRC="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.127"  make -j1 -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/src AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2=kmod-igb-intel-5.10.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" ;
make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/src'
common.mk:158: *** Linux kernel source not configured - missing autoconf.h.  Stop.
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/src'
make[3]: Leaving directory '/workdir/openwrt/package/lean/igb-intel'
time: package/lean/igb-intel/compile#0.17#0.06#0.22
make[3]: *** [Makefile:49: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/.built] Error 2
    ERROR: package/lean/igb-intel failed to build.
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [package/Makefile:116: package/lean/igb-intel/compile] Error 1
make[1]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

nbMars1980, 2022-07-12:
我也是一样的错误

xiangfeidexiaohuo, 2022-07-12:
现在默认用的kmod-igb-intel，但是以前默认用的kmod-igb，你们又是用的以前的配置文件，导致2个都选了，所以2个冲突，2个东西只能二选一。把kmod-igb 关掉就成。

ccblinda, 2022-07-12:
CONFIG_PACKAGE_kmod-igb=n
CONFIG_PACKAGE_kmod-igb-intel=y
成功！

xiangfeidexiaohuo, 2022-07-12:
> > 现在默认用的kmod-igb-intel，但是以前默认用的kmod-igb，你们又是用的以前的配置文件，导致2个都选了，所以2个冲突，2个东西只能二选一。把kmod-igb 关掉就成。
> 
> 那这应该算是bug吧？

不算bug，以前默认的kmod-igb，后面更新开了一个新的来替代它，但是很多人配置文件不是重新拉取的啊，用的老的啊，所以导致2个都是y。

vckgmh, 2022-07-12:
> > > 现在默认用的kmod-igb-intel，但是以前默认用的kmod-igb，你们又是用的以前的配置文件，导致2个都选了，所以2个冲突，2个东西只能二选一。把kmod-igb 关掉就成。
> > 
> > 
> > 那这应该算是bug吧？
> 
> 不算bug，以前默认的kmod-igb，后面更新开了一个新的来替代它，但是很多人配置文件不是重新拉取的啊，用的老的啊，所以导致2个都是y。

我用的云编译， 我输出看了一下配置文件，有 CONFIG_PACKAGE_kmod-igb-intel=y 以及 # CONFIG_PACKAGE_kmod-igb is not set ，还是报错 ERROR: package/lean/igb-intel failed to build.

mirthlesslk, 2022-07-12:
> 我用的云编译， 我输出看了一下配置文件，有 CONFIG_PACKAGE_kmod-igb-intel=y 以及 # CONFIG_PACKAGE_kmod-igb is not set ，还是报错 ERROR: package/lean/igb-intel failed to build.

前一个y，后一个带＃就是两个都勾选了，＃去掉才是取消勾选

vckgmh, 2022-07-12:
> 前一个y，后一个带＃就是两个都勾选了，＃去掉才是取消勾选

不对吧，我说的是.config文件啊

Jason6111, 2022-07-12:
> 现在默认用的kmod-igb-intel，但是以前默认用的kmod-igb，你们又是用的以前的配置文件，导致2个都选了，所以2个冲突，2个东西只能二选一。把kmod-igb 关掉就成。

5.18重新配置的也一样
# CONFIG_PACKAGE_kmod-igb is not set
CONFIG_PACKAGE_kmod-igb-intel=y

make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/src'
common.mk:158: *** Linux kernel source not configured - missing autoconf.h.  Stop.
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/src'
make[3]: *** [Makefile:49: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/lean/igb-intel'
time: package/lean/igb-intel/compile#0.16#0.07#0.23
    ERROR: package/lean/igb-intel failed to build.
make[2]: *** [package/Makefile:116: package/lean/igb-intel/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2

gc89925, 2022-07-13:
> CONFIG_PACKAGE_kmod-igb=n CONFIG_PACKAGE_kmod-igb-intel=y 成功！

请问一下，这两玩意儿在哪个文件下找到的，纯小白，请告知下路径

psonys, 2022-07-13:
简单说一遍吧，make menuconfig-Kernel modules-Network Devices-kmod-igb前面*去了，勾选kmod-igb-intel

Jason6111, 2022-07-13:
> 简单说一遍吧，make menuconfig-Kernel modules-Network Devices-kmod-igb前面*去了，勾选kmod-igb-intel

测试过本地可以过 同配置云过去不 取消igb 选用igb-intel了

xyskwjy, 2022-07-13:
make[3]: Entering directory '/workdir/openwrt/package/lean/igb-intel'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.igb-5.10.2.tar.gz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "igb-5.10.2.tar.gz" "6ef2c26c18acb898d693040b56b27316" "" "http://downloads.sourceforge.net/project/e1000/igb%20stable/5.10.2/"    '
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/.built
touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/.built_check
KSRC="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.53"  make -j1 -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/src AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2=kmod-igb-intel-5.10.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" ;
make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/src'
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/src'
common.mk:158: *** Linux kernel source not configured - missing autoconf.h.  Stop.
make[3]: *** [Makefile:49: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/lean/igb-intel'
time: package/lean/igb-intel/compile#0.17#0.04#0.22
    ERROR: package/lean/igb-intel failed to build.
make[2]: *** [package/Makefile:116: package/lean/igb-intel/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
    [CC] libdm-file.c
    [CC] libdm-report.c
    [CC] libdm-string.c
    [CC] libdm-targets.c
    [CC] libdm-timestamp.c
    [CC] pool.c
    [CC] matcher.c
    [CC] parse_rx.c
    [CC] ttree.c
    [CC] status.c
    [CC] vdo_target.c
    [AR] device_mapper/libdevice-mapper.a
/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ar: creating device_mapper/libdevice-mapper.a
make -C lib
make[5]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/lib'
    [CC] activate.c
    [CC] lvmcache.c
    [CC] writecache.c
    [CC] integrity.c
    [CC] cache.c
    [CC] toolcontext.c
    [CC] config.c
    [CC] btree.c
    [CC] str_list.c
    [CC] bcache.c
    [CC] bcache-utils.c
    [CC] dev-cache.c
    [CC] device_id.c
    [CC] dev-ext.c
    [CC] dev-io.c
    [CC] dev-md.c
    [CC] dev-mpath.c
    [CC] dev-swap.c
    [CC] dev-type.c
    [CC] dev-luks.c
    [CC] dev-dasd.c
    [CC] dev-lvm1-pool.c
    [CC] display.c
    [CC] errseg.c
    [CC] unknown.c
    [CC] filter-composite.c
    [CC] filter-persistent.c
    [CC] filter-regex.c
    [CC] filter-sysfs.c
    [CC] filter-md.c
    [CC] filter-fwraid.c
    [CC] filter-mpath.c
    [CC] filter-partitioned.c
    [CC] filter-type.c
    [CC] filter-usable.c
    [CC] filter-internal.c
    [CC] filter-signature.c
    [CC] filter-deviceid.c
    [CC] archive.c
    [CC] archiver.c
[... 289 lines cut ...]
(cd /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /workdir/openwrt/bin/packages/x86_64/packages
/workdir/openwrt/staging_dir/host/bin/fakeroot /workdir/openwrt/scripts/ipkg-build -m "" /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper /workdir/openwrt/bin/packages/x86_64/packages
Packaged contents of /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-x86_64/libdevmapper into /workdir/openwrt/bin/packages/x86_64/packages/libdevmapper_2.03.14-1_x86_64.ipk
echo "libdevmapper" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/lvm2.normal.install
rm -rf /workdir/openwrt/tmp/stage-lvm2
mkdir -p /workdir/openwrt/tmp/stage-lvm2/host /workdir/openwrt/staging_dir/target-x86_64_musl/packages
install -d -m0755 /workdir/openwrt/tmp/stage-lvm2/usr/include
install -m0644 /workdir/openwrt/build_dir/target-x86_64_musl/lvm2-normal/LVM2.2.03.14/ipkg-install/usr/include/libdevmapper.h /workdir/openwrt/tmp/stage-lvm2/usr/include
install -d -m0755 /workdir/openwrt/tmp/stage-lvm2/usr/lib
[... cut ...]
touch /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.lvm2.normal_installed
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/lvm2'
time: package/feeds/packages/lvm2/normal/compile#45.22#5.45#49.57
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

qqsir-dev, 2022-07-13:
本地编译可以，云编译仍然不行。

xiangfeidexiaohuo, 2022-07-13:
> kmod-igb

不存在，就很简单的一个东西，配置文件里，删除kmod-igb相关的，默认启用kmod-igb-intel的。云编译都跑了几天，怎么可能不行。要么你配置同步错了，搜索kmod-igb，所有的全部删掉。

qqsir-dev, 2022-07-13:
> make[3]: Leaving directory '/workdir/openwrt/package/lean/default-settings' time: package/lean/default-settings/compile#0.14#0.03#0.17 make[3]: Entering directory '/workdir/openwrt/package/lean/igb-intel' mkdir -p /workdir/openwrt/dl SHELL= flock /workdir/openwrt/tmp/.igb-5.10.2.tar.gz.flock -c ' /workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "igb-5.10.2.tar.gz" "6ef2c26c18acb898d693040b56b27316" "" "http://downloads.sourceforge.net/project/e1000/igb%20stable/5.10.2/" ' rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/.built touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/.built_check KSRC="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.11" make -j1 -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/src AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2=kmod-igb-intel-5.10.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" ; make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/src' common.mk:158: *** Linux kernel source not configured - missing autoconf.h. Stop. make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/src' make[3]: *** [Makefile:49: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/kmod-igb-intel-5.10.2/.built] Error 2 make[3]: Leaving directory '/workdir/openwrt/package/lean/igb-intel' time: package/lean/igb-intel/compile#0.19#0.04#0.23 ERROR: package/lean/igb-intel failed to build. make[2]: *** [package/Makefile:116: package/lean/igb-intel/compile] Error 1 make[2]: Leaving directory '/workdir/openwrt' make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2 make[1]: Leaving directory '/workdir/openwrt' make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2 Error: Process completed with exit code 2.
> 
> Linux kernel source not configured - missing autoconf.h. Stop. igb-5.10.2.tar.gz包里的src目录下确实是没有autoconf.h，我的理解对吗？

我也是这个错误。
make[4]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.53'
make -C /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.53 KCFLAGS="-fmacro-prefix-map=/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Jul 12 15:13:57 [202](https://github.com/qqsir-dev/AutoLede/runs/7319996011?check_suite_focus=true#step:10:202)2" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/runner/work/AutoLede/AutoLede/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.53 CC="x86_64-openwrt-linux-musl-gcc" bzImage modules
make[4]: Entering directory '/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.53'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  DESCEND objtool
<stdin>:1:10: fatal error: libelf.h: No such file or directory
compilation terminated.
  CC      /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.53/tools/objtool/arch/x86/special.o
In file included from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.53/tools/objtool/include/objtool/objtool.h:13,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.53/tools/objtool/include/objtool/arch.h:11,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.53/tools/objtool/include/objtool/check.h:11,
                 from /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.53/tools/objtool/include/objtool/special.h:10,
                 from arch/x86/special.c:4:
/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.53/tools/objtool/include/objtool/elf.h:10:10: fatal error: gelf.h: No such file or directory
   10 | #include <gelf.h>
      |          ^~~~~~~~
compilation terminated.
make[8]: *** [/home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.53/tools/build/Makefile.build:97: /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.53/tools/objtool/arch/x86/special [... cut ...]
make[3]: *** [Makefile:29: /home/runner/work/AutoLede/AutoLede/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.53/.modules] Error 2
make[2]: *** [Makefile:11: compile] Error 2
    ERROR: target/linux failed to build.
make[1]: *** [target/Makefile:25: target/linux/compile] Error 1
make[1]: Leaving directory '/home/runner/work/AutoLede/AutoLede/openwrt'
make: *** [/home/runner/work/AutoLede/AutoLede/openwrt/include/toplevel.mk:230: target/compile] Error 2
Error: Process completed with exit code 2.

Jejz168, 2022-07-13:
> > kmod-igb
> 
> 不存在，就很简单的一个东西，配置文件里，删除kmod-igb相关的，默认启用kmod-igb-intel的。云编译都跑了几天，怎么可能不行。要么你配置同步错了，搜索kmod-igb，所有的全部删掉。

我云编译怎么改都编译不过去。KMS也不行 ，igb-intel也不行。yml各种改也不行，最终用了大佬你的编译yml，直接全部过，真奇怪。我自己的改成和你的一样也不行。全部用你的完全没有问题。真不知道yml还有玄学啊，几天了，真不知道哪里问题

Jason6111, 2022-07-13:
> > > kmod-igb
> > 
> > 
> > 不存在，就很简单的一个东西，配置文件里，删除kmod-igb相关的，默认启用kmod-igb-intel的。云编译都跑了几天，怎么可能不行。要么你配置同步错了，搜索kmod-igb，所有的全部删掉。
> 
> 我云编译怎么改都编译不过去。KMS也不行 ，igb-intel也不行。yml各种改也不行，最终用了大佬你的编译yml，直接全部过，真奇怪。我自己的改成和你的一样也不行。全部用你的完全没有问题。真不知道yml还有玄学啊，几天了，真不知道哪里问题

我对比了下那位能过的大佬库 看他依赖跟P3的有区别 所以我改了试试

Jejz168, 2022-07-13:
> > > > kmod-igb
> > > 
> > > 
> > > 不存在，就很简单的一个东西，配置文件里，删除kmod-igb相关的，默认启用kmod-igb-intel的。云编译都跑了几天，怎么可能不行。要么你配置同步错了，搜索kmod-igb，所有的全部删掉。
> > 
> > 
> > 我云编译怎么改都编译不过去。KMS也不行 ，igb-intel也不行。yml各种改也不行，最终用了大佬你的编译yml，直接全部过，真奇怪。我自己的改成和你的一样也不行。全部用你的完全没有问题。真不知道yml还有玄学啊，几天了，真不知道哪里问题
> 
> 我对比了下那位能过的大佬库 看他依赖跟P3的有区别 所以我改了试试

怎么改都过不去。直接用他的完美编译。真是神奇。，。不知道哪里问题

Jejz168, 2022-07-13:
> > > > kmod-igb
> > > 
> > > 
> > > 不存在，就很简单的一个东西，配置文件里，删除kmod-igb相关的，默认启用kmod-igb-intel的。云编译都跑了几天，怎么可能不行。要么你配置同步错了，搜索kmod-igb，所有的全部删掉。
> > 
> > 
> > 我云编译怎么改都编译不过去。KMS也不行 ，igb-intel也不行。yml各种改也不行，最终用了大佬你的编译yml，直接全部过，真奇怪。我自己的改成和你的一样也不行。全部用你的完全没有问题。真不知道yml还有玄学啊，几天了，真不知道哪里问题
> 
> 我对比了下那位能过的大佬库 看他依赖跟P3的有区别 所以我改了试试

我yml依赖全部用了和他一样、步骤都是一样。也是报错

closed 2022-07-13 by coolsnowwolf

hzjnet, 2022-07-14:
确实到现在还是有问题，包括最新的全拉新源码！20220714

Jejz168, 2022-07-14:
> 确实到现在还是有问题，包括最新的全拉新源码！20220714

CONFIG_PACKAGE_samba36-server=y
CONFIG_PACKAGE_SAMBA_MAX_DEBUG_LEVEL=-1

小心这个

PR #9750 (2022-07-13): igb-intel: mark conflict with kmod-igb
Fixes: #9742