coolsnowwolf/lede #9755: 自行编译固件出现日志提示 Failed to load DMC firmware i915/kbl_dmc_ver1_04.bin

newtamax, 2022-07-14:
L大，我自己编译时出现问题，看到已有人提出issue，但没有回复，所以再提一下，其实就是如何把i915的核显驱动加入到Kernel里，找了很多地方没有得到解答，请百忙之中回复下，非常感谢。

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

(2) 路由器型号和固件版本

A：自行编译OpenWrt原版固件出现i915提示错误。

 (3) 详细日志

A：[   60.534953] i915 0000:00:02.0: Failed to load DMC firmware i915/kbl_dmc_ver1_04.bin
Wed Jan 6 06:03:13 2021 kern.notice kernel: [ 60.559426] i915 0000:00:02.0: DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915
[   60.536048] i915 0000:00:02.0: DMC firmware homepage: [https://git.kernel.org/pub/scm/l ... mware.git/tree/i915](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915)

WYC-2020, 2022-07-18:
修改makefile拷贝到那个位置即可

coolsnowwolf, 2022-07-19:
必须把 i915 写成 kmod 模块而不是集成到内核，否则内核启动的时候还没找到 rootfs 呢，自然无法挂载 DMC 的 固件

WYC-2020, 2022-07-20:
> 必须把 i915 写成 kmod 模块而不是集成到内核，否则内核启动的时候还没找到 rootfs 呢，自然无法挂载 DMC 的 固件

已经是kmod了吧，并且应该是已经读取到了这个文件才对，因为看源码 如果没读取到的话 应该会报
[image]
在看加载哪里，应该是这个判断没有通过才导致的
[image]

coolsnowwolf, 2026-09-05:
当前固件树已经提供 `i915-firmware-dmc`，并由 `i915-firmware` 元包依赖；该包会从 linux-firmware 安装所有 `*_dmc*.bin`，包括 KBL DMC。日志对应的是未安装/未选中固件包，而非内核源码缺失，关闭为“已完成”。

closed 2026-09-05 by coolsnowwolf