coolsnowwolf/lede #9758: netfs.ko缺失

vipboddy, 2022-07-14:
内核：5.18
Packaged contents of /mnt/op/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-9pnet into /mnt/op/lede/bin/targets/x86/64/packages/kmod-9pnet_5.18.11-1_x86_64.ipk
Package kmod-fs-9p is missing dependencies for the following libraries:
netfs.ko
make[2]: *** [modules/fs.mk:27: /mnt/op/lede/bin/targets/x86/64/packages/kmod-fs-9p_5.18.11-1_x86_64.ipk] Error 1
make[2]: Leaving directory '/mnt/op/lede/package/kernel/linux'
time: package/kernel/linux/compile#10.48#1.88#10.63
    ERROR: package/kernel/linux failed to build.
make[1]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[1]: Leaving directory '/mnt/op/lede'
make: *** [/mnt/op/lede/include/toplevel.mk:230: package/compile] Error 2

WYC-2020, 2022-07-15:
@vipboddy  
[image]

umarizulkifli, 2022-08-19:
```
directory '/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/workspace/openwrt/package/kernel/linux'
2022-08-19T13:36:24.6965900Z Package kmod-fs-9p is missing dependencies for the following libraries:
2022-08-19T13:36:24.6972570Z netfs.ko
2022-08-19T13:36:24.6976910Z make[2]: *** [modules/fs.mk:27: /home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/workspace/openwrt/bin/targets/x86/64/packages/kmod-fs-9p_5.19.2-1_x86_64.ipk] Error 1
2022-08-19T13:36:24.6995331Z make[2]: Leaving directory '/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/workspace/openwrt/package/kernel/linux'
2022-08-19T13:36:24.6996358Z time: package/kernel/linux/compile#3.03#0.33#3.36
2022-08-19T13:36:24.6998560Z     ERROR: package/kernel/linux failed to build.
2022-08-19T13:36:24.7012249Z make[1]: Leaving directory '/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/workspace/openwrt'
2022-08-19T13:36:24.7014810Z make[1]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
2022-08-19T13:36:24.7019874Z make: *** [/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/workspace/openwrt/include/toplevel.mk:230: package/compile] Error 2
2022-08-19T13:36:24.7105790Z ##[error]Process completed with exit code 2.
```

closed 2023-04-26 by aiamadeus

wxk6b1203, 2023-05-14:
各位好，想请教下这个问题是如何解决的？我的环境是Ubuntu 20.04。

wxk6b1203, 2023-05-18:
> 各位好，想请教下这个问题是如何解决的？我的环境是Ubuntu 20.04。

Update：勾选了netfs的内核模块但是又没有对应依赖，取消掉就好了。