coolsnowwolf/lede #9760: cryptography编译出错

quanjindeng, 2022-07-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：cryptography编译失败

### (2) 路由器型号和固件版本

A：N1（aarch64_cortex-a53），最新源码

### (3) 详细日志

A：
cd "/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/pypi/cryptography-3.4.8/" && CC="aarch64-openwrt-linux-musl-gcc" CCSHARED="aarch64-openwrt-linux-musl-gcc -DPIC -fPIC" CXX="aarch64-openwrt-linux-musl-g++" LD="aarch64-openwrt-linux-musl-gcc" LDSHARED="aarch64-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -march=armv8-a+crypto -mtune=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/pypi/cryptography-3.4.8=cryptography-3.4.8 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/include -I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/python3.9" LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib -znow -zrelro -lpython3.9" _PYTHON_HOST_PLATFORM="linux-aarch64" __PYVENV_LAUNCHER__="/usr/bin/python3.9" PYTHONPATH="/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/python3.9:/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl//usr/lib/python3.9/site-packages:/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/pypi/cryptography-3.4.8/ipkg-install//usr/lib/python3.9/site-packages" PYTHONDONTWRITEBYTECODE=1 PYTHONOPTIMIZE="" _python_sysroot="/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl" _python_prefix="/usr" _python_exec_prefix="/usr"     /workdir/openwrt/staging_dir/hostpkg/bin/python3.9  setup.py   install --prefix="/usr" --root="/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/pypi/cryptography-3.4.8/ipkg-install" --single-version-externally-managed
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.
Traceback (most recent call last):
    
=============================DEBUG ASSISTANCE=============================
  File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.9/site-packages/setuptools/dist.py", line 453, in __init__
    _Distribution.__init__(
  File "/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/python3.9/distutils/dist.py", line 292, in __init__
    self.finalize_options()
  File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.9/site-packages/setuptools/dist.py", line 831, in finalize_options
    ep(self)
  File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.9/site-packages/setuptools/dist.py", line 852, in _finalize_setup_keywords
    ep.load()(self, ep.name, value)
  File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2450, in load
    return self.resolve()
  File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.9/site-packages/pkg_resources/__init__.py", line 2456, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/pypi/cryptography-3.4.8/.eggs/setuptools_rust-1.4.1-py3.9.egg/setuptools_rust/__init__.py", line 1, in <module>
    from .build import build_rust
  File "/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/pypi/cryptography-3.4.8/.eggs/setuptools_rust-1.4.1-py3.9.egg/setuptools_rust/build.py", line 20, in <module>
    from setuptools.command.build import build as CommandBuild  # type: ignore[import]
ModuleNotFoundError: No module named 'setuptools.command.build'
make[3]: *** [Makefile:57: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/pypi/cryptography-3.4.8/.built] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/python/python-cryptography'
time: package/feeds/packages/python-cryptography/compile#1.56#0.24#2.11
    ERROR: package/feeds/packages/python-cryptography failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/python-cryptography/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

quanjindeng, 2022-07-15:
还是直接取消勾选算了，关闭issue了
# CONFIG_PACKAGE_python3-cryptography is not set
# CONFIG_PACKAGE_python3-docker is not set
# CONFIG_PACKAGE_python3-paramiko is not set

closed 2022-07-15 by quanjindeng

WYC-2020, 2022-07-15:
没aarch64_cortex-a53 环境，刚用x86 编译了一下python3-cryptography 没有问题