coolsnowwolf/lede #9767: x64 5.18编SHORTCUT_FE:kmod-fast-classifier不过帮忙看下

Jason6111, 2022-07-15:
make[3]: Entering directory '/workdir/openwrt/package/lean/shortcut-fe/shortcut-fe'
[2802](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2803)
mkdir -p /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
[2803](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2804)
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.pkgdir/kmod-shortcut-fe/. /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/'
[2804](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2805)
touch /workdir/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-shortcut-fe_installed
[2805](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2806)
echo "kmod-shortcut-fe" >> /workdir/openwrt/staging_dir/target-x86_64_musl/pkginfo/shortcut-fe.default.install
[2806](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2807)
make[3]: Leaving directory '/workdir/openwrt/package/lean/shortcut-fe/shortcut-fe'
[2807](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2808)
time: package/lean/shortcut-fe/shortcut-fe/compile#0.14#0.05#0.18
[2808](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2809)
make[3]: Entering directory '/workdir/openwrt/package/lean/shortcut-fe/fast-classifier'
[2809](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2810)
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/fast-classifier/.built
[2810](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2811)
touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/fast-classifier/.built_check
[2811](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2812)
make -j1 -C "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.11" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Jul 15 02:05:30 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8125.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/kmod-igb-intel.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8152.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/fullconenat.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/button-hotplug.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8168.symvers /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/symvers/shortcut-fe.symvers" KERNELRELEASE=5.18.11  M="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/fast-classifier" CONFIG_FAST_CLASSIFIER=m EXTRA_CFLAGS+="-DSFE_SUPPORT_IPV6"  modules
[2812](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2813)
make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.11'
[2813](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2814)
  CC [M]  /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/fast-classifier/fast-classifier.o
[2814](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2815)
/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/fast-classifier/fast-classifier.c: In function 'fast_classifier_update_protocol':
[2815](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2816)
/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/fast-classifier/fast-classifier.c:473:6: error: 'nf_ct_tcp_no_window_check' undeclared (first use in this function); did you mean 'nf_ct_expect_find_get'?
[2816](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2817)
  if (nf_ct_tcp_no_window_check
[2817](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2818)
      ^~~~~~~~~~~~~~~~~~~~~~~~~
[2818](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2819)
      nf_ct_expect_find_get
[2819](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2820)
/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/fast-classifier/fast-classifier.c:473:6: note: each undeclared identifier is reported only once for each function it appears in
[2820](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2821)
make[5]: *** [scripts/Makefile.build:288: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/fast-classifier/fast-classifier.o] Error 1
[2821](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2822)
make[4]: *** [Makefile:1838: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/fast-classifier] Error 2
[2822](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2823)
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.11'
[2823](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2824)
make[3]: *** [Makefile:110: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/fast-classifier/.built] Error 2
[2824](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2825)
make[3]: Leaving directory '/workdir/openwrt/package/lean/shortcut-fe/fast-classifier'
[2825](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2826)
time: package/lean/shortcut-fe/fast-classifier/compile#0.97#0.19#1.15
[2826](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2827)
    ERROR: package/lean/shortcut-fe/fast-classifier failed to build.
[2827](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2828)
make[2]: *** [package/Makefile:116: package/lean/shortcut-fe/fast-classifier/compile] Error 1
[2828](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2829)
make[2]: Leaving directory '/workdir/openwrt'
[2829](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2830)
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
[2830](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2831)
make[1]: Leaving directory '/workdir/openwrt'
[2831](https://github.com/Jason6111/OpenWrt_Personal/runs/7357934117?check_suite_focus=true#step:12:2832)
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2

luhe222, 2022-07-16:
5.18的r2s和r4s编译出现同样情况

WYC-2020, 2022-07-16:
没修而已，改下宏

luhe222, 2022-07-16:
> 没修而已，改下宏

请教一下，怎么修改宏

WYC-2020, 2022-07-16:
> > 没修而已，改下宏
> 
> 请教一下，怎么修改宏

等我修了再说

WYC-2020, 2022-07-16:
[image]
修好了，我是一直没搞明白为啥要闭环，原来的修复有问题， @luhe222 @Jason6111

closed 2022-07-16 by Jason6111

luhe222, 2022-07-16:
> [image] 修好了，我是一直没搞明白为啥要闭环，原来的修复有问题， @luhe222 @Jason6111
你好，编译5.18内核的r2s和r4s内裤仍然提示fast-classifier无法编译

Jason6111, 2022-07-17:
> [image] 修好了，我是一直没搞明白为啥要闭环，原来的修复有问题， @luhe222 @Jason6111
> 你好，编译5.18内核的r2s和r4s内裤仍然提示fast-classifier无法编译

检查下原配置有没有残留

PR #9777 (2022-07-17): Update fast-classifier.c fix kmod-fast-classifier compile error
Fix #9767

Copy from @WYC-2020