coolsnowwolf/lede #9768: 最近mt7621的机器运行几分钟后奔溃

izma669, 2022-07-15:
### 反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
机器除了默认的插件自己又整合了adbyby plus+ ，SAMBA,DLNA,uhttpd,
固件刷写时间是7月15日18点16分，宕机时间是21点40分左右
### (1) 具体问题
A：第一次刷完使用3个小时候左右正常，再访问samba后就机器奔溃了，这时候把机器重启最多只能坚持几分钟后然后就继续宕机

### (2) 路由器型号和固件版本

A：ZBT-WG3526 32M,固件版本：5.4.203 -R22.7.7

### (3) 详细日志

A：内核日志

[    0.000000] Linux version 5.4.203 (izma@izma-virtual-machine) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4700-e66a94a8d)) #0 SMP Thu Jul 14 13:46:49 2022
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Zbtlink ZBT-WG3526 (32M)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 14 pages/cpu s27184 r8192 d21968 u57344
[    0.000000] pcpu-alloc: s27184 r8192 d21968 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=000312d0
[    0.000000] Readback ErrCtl register=000312d0
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 509224K/524288K available (6304K kernel code, 271K rwdata, 1312K rodata, 1248K init, 236K bss, 15064K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007910] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.074073] pid_max: default: 32768 minimum: 301
[    0.078889] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.086091] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.097165] rcu: Hierarchical SRCU implementation.
[    0.102714] smp: Bringing up secondary CPUs ...
[    1.662824] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[... 555 lines cut ...]
Fri Jul 15 21:37:19 2022 daemon.notice procd: /etc/rc.d/S90turboacc: sh: you need to specify whom to kill
Fri Jul 15 21:37:19 2022 daemon.notice procd: /etc/rc.d/S90turboacc: Stop DNS Caching
Fri Jul 15 21:37:19 2022 daemon.notice procd: /etc/rc.d/S90turboacc: DNSMASQ change
Fri Jul 15 21:37:20 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
Fri Jul 15 21:37:20 2022 daemon.info hostapd: wlan0: STA 28:6c:07:16:8c:ab IEEE 802.11: authenticated
Fri Jul 15 21:37:20 2022 daemon.info hostapd: wlan0: STA 28:6c:07:16:8c:ab IEEE 802.11: associated (aid 1)
Fri Jul 15 21:37:21 2022 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 28:6c:07:16:8c:ab
Fri Jul 15 21:37:21 2022 daemon.info hostapd: wlan0: STA 28:6c:07:16:8c:ab RADIUS: starting accounting session 71735E62B0636475
Fri Jul 15 21:37:21 2022 daemon.info hostapd: wlan0: STA 28:6c:07:16:8c:ab WPA: pairwise key handshake completed (RSN)
Fri Jul 15 21:37:21 2022 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 28:6c:07:16:8c:ab
Fri Jul 15 21:37:21 2022 user.notice firewall: Reloading firewall due to ifup of wan (wan)
Fri Jul 15 21:37:22 2022 daemon.notice netifd: 4G (3745): SIM not initialized
Fri Jul 15 21:37:22 2022 daemon.notice netifd: 4G (5965): Stopping network 4G
Fri Jul 15 21:37:22 2022 daemon.notice netifd: 4G (5965): Command failed: Permission denied
Fri Jul 15 21:37:22 2022 daemon.notice netifd: Interface '4G' is now down
Fri Jul 15 21:37:24 2022 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (wan)
Fri Jul 15 21:37:54 2022 daemon.info hostapd: wlan1: STA f4:96:34:6c:87:70 IEEE 802.11: authenticated
Fri Jul 15 21:37:54 2022 daemon.info hostapd: wlan1: STA f4:96:34:6c:87:70 IEEE 802.11: associated (aid 1)
Fri Jul 15 21:37:54 2022 daemon.notice hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED f4:96:34:6c:87:70 off
Fri Jul 15 21:37:54 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED f4:96:34:6c:87:70
Fri Jul 15 21:37:54 2022 daemon.info hostapd: wlan1: STA f4:96:34:6c:87:70 RADIUS: starting accounting session 877214E6B4A99588
Fri Jul 15 21:37:54 2022 daemon.info hostapd: wlan1: STA f4:96:34:6c:87:70 WPA: pairwise key handshake completed (RSN)
Fri Jul 15 21:37:54 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED f4:96:34:6c:87:70
Fri Jul 15 21:38:00 2022 user.notice transmission: Starting with 255236000 virt mem
Fri Jul 15 21:38:01 2022 daemon.info procd: - init complete -

lili8086, 2022-07-16:
我的也有问题，出租屋有个💩米rm2100，编译时几乎就是在原有配置上增加了helloworld，症状如下
1.重启没有无线，无论开不开helloworld都可能出现。
2.开启helloworld xray vless绕过大陆ip，跑几个小时后挂掉（开启lan分流似乎更容易出现），有时能访问大陆ip，web登录非常非常慢或者登不上去。
鄙人也是嵌入式linux工作但是未接触过路由器相关，需要其他诊断信息也指导我获取。

**重启无wifi日志**：
**系统日志：**
Sat Jul 16 11:10:14 2022 kern.notice kernel: [    0.000000] Linux version 5.4.203 (lee@ubuntu) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4649-a6893cf1b)) #0 SMP Tue Jul 12 15:13:57 2022
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] MIPS: machine is Xiaomi Redmi Router AC2100
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Sat Jul 16 11:10:14 2022 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sat Jul 16 11:10:14 2022 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sat Jul 16 11:10:14 2022 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] Zone ranges:
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000]   HighMem  empty
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] Movable zone start for each node
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] Early memory node ranges
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Sat Jul 16 11:10:14 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Sat Jul 16 11:10:14 2022 kern.debug kernel: [    0.000000]   Normal zone: 288 pages used for memmap
Sat Jul 16 11:10:14 2022 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sat Jul 16 11:10:14 2022 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s27152 r8192 d22000 u57344
Sat Jul 16 11:10:14 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s27152 r8192 d22000 u57344 alloc=14*4096
Sat Jul 16 11:10:14 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Sat Jul 16 11:10:14 2022 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] Writing ErrCtl register=000118c8
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] Readback ErrCtl register=000118c8
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] Memory: 120264K/131072K available (6164K kernel code, 274K rwdata, 1296K rodata, 1276K init, 235K bss, 10808K reserved, 0K cma-reserved, 0K highmem)
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] NR_IRQS: 256
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.007874] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.074042] pid_max: default: 32768 minimum: 301
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.078831] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.086031] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.096784] rcu: Hierarchical SRCU implementation.
Sat Jul 16 11:10:14 2022 kern.info kernel: [    0.102256] smp: Bringing up secondary CPUs ...
Sat Jul 16 11:10:14 2022 kern.warn kernel: [    2.413798] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[... cut ...]
[   34.643641] device eth0 entered promiscuous mode
[   34.652701] br-lan: port 1(eth0) entered blocking state
[   34.658068] br-lan: port 1(eth0) entered forwarding state
[   34.697123] mtk_soc_eth 1e100000.ethernet eth1: configuring for fixed/rgmii link mode
[   34.705659] mtk_soc_eth 1e100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[   34.714810] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   35.643717] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

aiamadeus, 2022-07-16:
去掉超频补丁重新编译即可

izma669, 2022-07-18:
> 没有这个补丁

closed 2022-07-19 by coolsnowwolf