coolsnowwolf/lede #9783: 最新提交的/realtek-r8125-dkms-9.009.02-1驱动错误

moonerick, 2022-07-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
SHELL= flock /home/sp/lede/tmp/.r8125-9.009.02-1.tar.gz.flock -c '      /home/sp/lede/scripts/download.pl "/home/sp/lede/dl" "r8125-9.009.02-1.tar.gz" "e3f0a0a02a4ba847cb962ea7e12c89e3f4ab732944f04ebeb76d4d9711fa8ca6" "" "https://codeload.github.com/awesometic/realtek-r8125-dkms/tar.gz/9.009.02-1?"    '
touch /home/sp/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8125-dkms-9.009.02-1/.prepared_7b8791520da2434585120eaa5def8ece_6664517399ebbbc92a37c5bb081b5c53_check
. /home/sp/lede/include/shell.sh; gzip -dc /home/sp/lede/dl/r8125-9.009.02-1.tar.gz | tar -C /home/sp/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8125-dkms-9.009.02-1/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/sp/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8125-dkms-9.009.02-1
Applying ./patches/010-config.patch using plaintext:
patching file src/Makefile

Applying ./patches/020-5.18-support.patch using plaintext:
patching file src/r8125_n.c
Hunk #1 FAILED at 11952.
1 out of 1 hunk FAILED -- saving rejects to file src/r8125_n.c.rej
Patch failed!  Please fix ./patches/020-5.18-support.patch!
make[3]: *** [Makefile:46: /home/sp/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8125-dkms-9.009.02-1/.prepared_7b8791520da2434585120eaa5def8ece_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/sp/lede/package/lean/r8125'
time: package/lean/r8125/compile#0.11#0.01#0.14
    ERROR: package/lean/r8125 failed to build.

WYC-2020, 2022-07-18:
删掉这个补丁

moonerick, 2022-07-18:
> 

删除可行， 谢谢

closed 2022-07-18 by moonerick

luhe222, 2022-07-18:
> 删掉这个补丁

你好，请教一下，如何删除这个补丁，编译x86同样出错

luhe222, 2022-07-18:
> > 
> 
> 删除可行， 谢谢

你好，请教一下，如何删除这个补丁，编译x86同样出错

moonerick, 2022-07-18:
> > > 
> > 
> > 
> > 删除可行， 谢谢
> 
> 你好，请教一下，如何删除这个补丁，编译x86同样出错

我使用的是intel的网卡，我在make menuconfig里面将Kernel modules > Network Devices > kmod-r8125直接取消了，可以通过编译

我看lean大已经重新提交了修改，移除了，你重新pull一下，应该就可以了