coolsnowwolf/lede #9784: X86 5.15.55 最新更新编译失败

baodp, 2022-07-18:
X86 5.15.55 最新更新编译失败。
从新默认配置 添加勾选了 IPV6支持选项。也是这样：

Applying ./patches/020-5.18-support.patch using plaintext: 
patching file src/r8125_n.c
Hunk #1 FAILED at 11952.
touch /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libgcc_installed
1 out of 1 hunk FAILED -- saving rejects to file src/r8125_n.c.rej
Patch failed!  Please fix ./patches/020-5.18-support.patch!
make[3]: *** [Makefile:46: /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8125-dkms-9.009.02-1/.prepared_46d36ddcef87af762b5b6c202024aaf5_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/bdp/lede/package/lean/r8125'
time: package/lean/r8125/compile#0.25#0.16#0.39
echo "libgcc" >> /home/bdp/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
    ERROR: package/lean/r8125 failed to build.
make[2]: *** [package/Makefile:116: package/lean/r8125/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
mkdir -p /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/stamp
make[3]: Leaving directory '/home/bdp/lede/package/libs/libjson-c'
time: package/libs/libjson-c/host-compile#0.30#0.10#0.41
echo '6' | cmp -s - /home/bdp/lede/staging_dir/target-x86_64_musl/pkginfo/libstdcpp.version || { echo '6' > /home/bdp/lede/staging_dir/target-x86_64_musl/pkginfo/libstdcpp.version;  } 
SHELL= flock /home/bdp/lede/tmp/.root-copy.flock -c 'cp -fpR /home/bdp/lede/build_dir/target-x86_64_musl/toolchain/.pkgdir/libstdcpp/. /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/'
mkdir -p /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/stamp
touch /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libstdcpp_installed
SHELL= flock /home/bdp/lede/tmp/.root-copy.flock -c 'cp -fpR /home/bdp/lede/build_dir/target-x86_64_musl/linux-x86_64/r8168-8.050.03/.pkgdir/kmod-r8168/. /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/'
echo "libstdcpp" >> /home/bdp/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
make[3]: Leaving directory '/home/bdp/lede/package/system/fwtool'
time: package/system/fwtool/host-compile#0.27#0.16#0.42
mkdir -p /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/stamp
mkdir -p /home/bdp/lede/dl
make[3]: Leaving directory '/home/bdp/lede/package/system/usign'
time: package/system/usign/host-compile#0.29#0.14#0.43
make[3]: Leaving directory '/home/bdp/lede/package/utils/lua'
SHELL= flock /home/bdp/lede/tmp/.root-copy.flock -c 'cp -fpR /home/bdp/lede/build_dir/target-x86_64_musl/toolchain/.pkgdir/libpthread/. /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/'
time: package/utils/lua/host-compile#0.29#0.17#0.43
SHELL= flock /home/bdp/lede/tmp/.ncurses-6.3.tar.gz.flock -c '  	/home/bdp/lede/scripts/download.pl "/home/bdp/lede/dl" "ncurses-6.3.tar.gz" "97fc51ac2b085d4cde31ef4d2c3122c21abc217e9090a43a30fc5ec21684e059" "" "@GNU/ncurses"    '
touch /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-r8168_installed
touch /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.libpthread_installed
echo "libpthread" >> /home/bdp/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
mkdir -p /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/bdp/lede/tmp/.root-copy.flock -c 'cp -fpR /home/bdp/lede/build_dir/target-x86_64_musl/toolchain/.pkgdir/librt/. /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/'
echo "kmod-r8168" >> /home/bdp/lede/staging_dir/target-x86_64_musl/pkginfo/r8168.default.install
make[3]: Leaving directory '/home/bdp/lede/package/lean/r8168'
touch /home/bdp/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.librt_installed
time: package/lean/r8168/compile#0.31#0.14#0.45
echo "librt" >> /home/bdp/lede/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
make[3]: Leaving directory '/home/bdp/lede/package/libs/toolchain'
time: package/libs/toolchain/compile#0.29#0.16#0.47
mkdir -p /home/bdp/lede/dl
make[3]: Leaving directory '/home/bdp/lede/package/boot/grub2'
time: package/boot/grub2/host-compile#0.35#0.16#0.49
SHELL= flock /home/bdp/lede/tmp/.go1.18.3.src.tar.gz.flock -c '  	/home/bdp/lede/scripts/download.pl "/home/bdp/lede/dl" "go1.18.3.src.tar.gz" "0012386ddcbb5f3350e407c679923811dbd283fcdc421724931614a842ecbc2d" "" "https://dl.google.com/go/" "https://mirrors.ustc.edu.cn/golang/" "https://mirrors.nju.edu.cn/golang/"    '
make[3]: Leaving directory '/home/bdp/lede/package/utils/e2fsprogs'
time: package/utils/e2fsprogs/host-compile#0.40#0.12#0.49
make[3]: Leaving directory '/home/bdp/lede/package/libs/ncurses'
time: package/libs/ncurses/host-compile#0.39#0.16#0.50
make[3]: Leaving directory '/home/bdp/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#0.48#0.29#0.65
 make[4] scripts/config/conf
mkdir -p /home/bdp/lede/dl
SHELL= flock /home/bdp/lede/tmp/.LuaSrcDiet-0.12.1.tar.bz2.flock -c '  	/home/bdp/lede/scripts/download.pl "/home/bdp/lede/dl" "LuaSrcDiet-0.12.1.tar.bz2" "ed7680f2896269ae8633756e7edcf09050812f78c8f49e280e63c30d14f35aea" "" "https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/luasrcdiet"    '
. /home/bdp/lede/include/shell.sh; bzcat /home/bdp/lede/dl/LuaSrcDiet-0.12.1.tar.bz2 | tar -C /home/bdp/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/bdp/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1
touch /home/bdp/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.prepared3badc7915ff87ae7f4bb10391a3e30c2_6664517399ebbbc92a37c5bb081b5c53
touch /home/bdp/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.configured
make -C src/ clean po2lmo
make[4]: Entering directory '/home/bdp/lede/feeds/luci/modules/luci-base/src'
rm -f po2lmo parser.so version.lua *.o
cc    -c -o po2lmo.o po2lmo.c
cc    -c -o template_lmo.o template_lmo.c
cc  -o po2lmo po2lmo.o template_lmo.o
make[4]: Leaving directory '/home/bdp/lede/feeds/luci/modules/luci-base/src'
make -C /home/bdp/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1 bin/LuaSrcDiet.lua
make[4]: Entering directory '/home/bdp/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1'
make[4]: Nothing to be done for 'bin/LuaSrcDiet.lua'.
make[4]: Leaving directory '/home/bdp/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1'
touch /home/bdp/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.built
install -d -m0755 /home/bdp/lede/staging_dir/hostpkg/bin
install -m0755 src/po2lmo /home/bdp/lede/staging_dir/hostpkg/bin/po2lmo
install -m0755 /home/bdp/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/bin/LuaSrcDiet.lua /home/bdp/lede/staging_dir/hostpkg/bin/LuaSrcDiet
mkdir -p /home/bdp/lede/staging_dir/hostpkg/stamp
touch /home/bdp/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.built
touch /home/bdp/lede/staging_dir/hostpkg/stamp/.luci-base_installed
make[3]: Leaving directory '/home/bdp/lede/feeds/luci/modules/luci-base'
time: package/feeds/luci/luci-base/host-compile#1.54#0.51#1.88
make[2]: Leaving directory '/home/bdp/lede'
make[1]: *** [package/Makefile:110: /home/bdp/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/bdp/lede'
make[1]: Leaving directory '/home/bdp/lede'
make: *** [/home/bdp/lede/include/toplevel.mk:230：world] 错误 2

mitsukileung, 2022-07-18:
5.15 没试过，不过5.18也是同样错误
你可以试试把 package/lean/r8125/patches/020-5.18-support.patch 删掉
好像新版驱动不需这个patch了

baodp, 2022-07-18:
5.18 也是过不了。

mitsukileung, 2022-07-18:
> 5.18 也是过不了。

新版的驱动文件应该不用打这个PATCH
我删了后，可以编译过