coolsnowwolf/lede #9785: 7月17日的版本无法成功编译package/lean/r8125

lzy0702, 2022-07-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述
7月17日更新的版本编译失败
### (1) 具体问题
编译到package/lean/r8125时失败

### (2) 路由器型号和固件版本
x86_64
内核版本5.15.55

### (3) 详细日志
$ make package/lean/r8125/compile V=s
make[1]: Entering directory '/home/lzy0702/lede-working/lede'
make[2]: Entering directory '/home/lzy0702/lede-working/lede/package/lean/r8125'
mkdir -p /home/lzy0702/lede-working/lede/dl
SHELL= flock /home/lzy0702/lede-working/lede/tmp/.r8125-9.009.02-1.tar.gz.flock -c '  	/home/lzy0702/lede-working/lede/scripts/download.pl "/home/lzy0702/lede-working/lede/dl" "r8125-9.009.02-1.tar.gz" "e3f0a0a02a4ba847cb962ea7e12c89e3f4ab732944f04ebeb76d4d9711fa8ca6" "" "https://codeload.github.com/awesometic/realtek-r8125-dkms/tar.gz/9.009.02-1?"    '
touch /home/lzy0702/lede-working/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8125-dkms-9.009.02-1/.prepared_b794c67fc7fc9dc8db3277fe10c01847_6664517399ebbbc92a37c5bb081b5c53_check
. /home/lzy0702/lede-working/lede/include/shell.sh; gzip -dc /home/lzy0702/lede-working/lede/dl/r8125-9.009.02-1.tar.gz | tar -C /home/lzy0702/lede-working/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8125-dkms-9.009.02-1/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/lzy0702/lede-working/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8125-dkms-9.009.02-1

Applying ./patches/010-config.patch using plaintext: 
patching file src/Makefile

Applying ./patches/020-5.18-support.patch using plaintext: 
patching file src/r8125_n.c
Hunk #1 FAILED at 11952.
1 out of 1 hunk FAILED -- saving rejects to file src/r8125_n.c.rej
Patch failed!  Please fix ./patches/020-5.18-support.patch!
make[2]: *** [Makefile:46: /home/lzy0702/lede-working/lede/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8125-dkms-9.009.02-1/.prepared_b794c67fc7fc9dc8db3277fe10c01847_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[2]: Leaving directory '/home/lzy0702/lede-working/lede/package/lean/r8125'
time: package/lean/r8125/compile#0.24#0.20#0.38
    ERROR: package/lean/r8125 failed to build.
make[1]: *** [package/Makefile:116: package/lean/r8125/compile] Error 1
make[1]: Leaving directory '/home/lzy0702/lede-working/lede'
make: *** [/home/lzy0702/lede-working/lede/include/toplevel.mk:230：package/lean/r8125/compile] 错误 2

lzy0702, 2022-07-18:
如果把第二个patch删掉就可以通过编译，但我没有8125网卡，不知道这么改了会不会影响正常运行

mitsukileung, 2022-07-18:
大雕更新了，应该不会有影响

lzy0702, 2022-07-18:
> 大雕更新了，应该不会有影响

不知道啊

closed 2022-07-19 by aiamadeus

TankCN, 2022-07-19:
你好问一下怎么删掉第二个patch啊我是小白，编译到这一步不知道怎么弄

TankCN, 2022-07-19:
> 如果把第二个patch删掉就可以通过编译，但我没有8125网卡，不知道这么改了会不会影响正常运行

你好问一下怎么删掉第二个patch啊我是小白，编译到这一步不知道怎么弄

hcym, 2022-07-19:
默认设置没问题，8125