coolsnowwolf/lede #9786: R22.7.16 友善 r5s ShadowSocksR Plus+问题

zzx, 2022-07-18:
每次设备重启， 客服端 显示 ShadowSocksR Plus+运行中，实际网络无法访问。只能重新启动服务，才可正常使用。
2022-07-18 11:39:45: -----------end------------
2022-07-18 11:39:42: Main node:Xray 1.5.8 (Xray, Penetrates Everything.) OpenWrt (go1.18.3 linux/arm64) Started!
2022-07-18 11:39:41: ----------start------------
2022-07-18 11:39:15: -----------end------------
2022-07-18 11:39:15: 禁止连接的域名加载完毕。
2022-07-18 11:39:15: 未启动主节点，禁止连接的域名正在加载。
2022-07-18 11:39:15: ----------start------------
2022-07-18 11:38:04: -----------end------------
2022-07-18 11:38:00: Main node:Xray 1.5.8 (Xray, Penetrates Everything.) OpenWrt (go1.18.3 linux/arm64) Started!
2022-07-18 11:37:59: ----------start------------
2022-07-18 11:37:59: boot！

WYC-2020, 2022-07-18:
隔壁

097, 2022-07-18:
r5s可以编译了？选哪个profile？

coolsnowwolf, 2022-07-18:
重新下载img 再刷入，refresh 了

closed 2022-07-19 by aiamadeus