coolsnowwolf/lede #9800: 编译失败，安装时提示冲突，云编译

achinese888, 2022-07-19:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
编译失败，提示冲突，云编译

A：编译失败，提示冲突，云编译  luci-app-turboacc kmod-fast-classifier

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：Configuring shadowsocks-libev-ss-local.
Collected errors:
Configuring kmod-alx.
 * check_conflicts_for: The following packages conflict with kmod-fast-classifier:
Configuring kmod-8139too.
 * check_conflicts_for: 	kmod-shortcut-fe-cm * 
Configuring simple-obfs-client.
 * opkg_install_cmd: Cannot install package kmod-fast-classifier.
Configuring kmod-pcnet32.
 * check_conflicts_for: The following packages conflict with kmod-fast-classifier:
Configuring shadowsocksr-libev-ssr-local.
 * check_conflicts_for: 	kmod-shortcut-fe-cm * 
Configuring libsensors5.
 * opkg_install_cmd: Cannot install package luci-app-turboacc.
Configuring luci-theme-argon-18.06.
Configuring luci-i18n-ddns-zh-cn.
Configuring kmod-i40e.
Configuring kmod-nf-nathelper-extra.
make[2]: *** [package/Makefile:70: package/install] Error 255
Configuring lm-sensors.
Configuring kmod-e1000.
Configuring libcares.
Configuring kmod-crypto-crc32.
Configuring kmod-fs-f2fs.
Configuring luci-app-samba4.
Configuring kmod-usb-net-rtl8150.
Configuring kmod-tulip.
Configuring luci-app-upnp.
Configuring shadowsocks-libev-ss-redir.
Configuring iptables-mod-conntrack-extra.
Configuring default-settings.
Configuring luci-i18n-dockerman-zh-cn.
Configuring luci-i18n-samba4-zh-cn.
Configuring luci-i18n-ttyd-zh-cn.
Configuring coreutils-nohup.
Configuring tcping.
Configuring brook.
Configuring shadowsocks-libev-ss-server.
Configuring shadowsocksr-libev-ssr-redir.
Configuring v2ray-core.
Configuring xray-core.
Configuring luci-app-passwall.
Configuring luci-i18n-passwall-zh-cn.
Configuring htop.
Configuring kmod-iavf.
Configuring kmod-igbvf.
Configuring bnx2x-firmware.
Configuring kmod-bnx2x.
Configuring kmod-forcedeth.
Configuring luci-i18n-base-zh-cn.
Configuring kmod-amd-xgbe.
Configuring bc.
Configuring kmod-usb-uhci.
Configuring kmod-drm-i915.
Configuring kmod-amazon-ena.
Configuring kmod-e1000e.
Configuring autocore-x86.
Configuring luci-i18n-upnp-zh-cn.
Configuring kmod-button-hotplug.
Configuring kmod-dummy.
Configuring iperf3.
Configuring kmod-igc.
Configuring kmod-bnx2.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring dnsmasq-full.
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:111: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2022-07-19:
只能选一个

achinese888, 2022-07-20:
ok,解决了，多谢

closed 2022-07-20 by aiamadeus

zhouzikk, 2023-10-05:
我云编译超过6小时。。。超时了