coolsnowwolf/lede #9803: rockchip 编译到 mac80211 失败

zhangguanzhang, 2022-07-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

```
 make[3] -C package/kernel/mac80211 compile
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
```

### (2) 路由器型号和固件版本

r2s 
```
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_rockchip_armv8_DEVICE_friendlyarm_nanopi-r2s=y
```

今天下午 17.00 拉取的源码编译的

### (3) 详细日志

```
make[3]: Entering directory '/workdir/openwrt/package/kernel/mac80211'
mkdir -p /workdir/openwrt/dl
SHELL= flock /workdir/openwrt/tmp/.backports-5.15.33-1.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "backports-5.15.33-1.tar.xz" "1b6b3bded4c81814ebebe2d194c2f8966d2399005b85ebb0557285b6e73f5422" "" "@KERNEL/linux/kernel/projects/backports/stable/v5.15.33/"    '
rm -f /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/.built
touch /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/.built_check
. /workdir/openwrt/include/shell.sh; var2file "V_mac80211_config" /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/.config
make -C "/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Jul 19 06:51:16 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/cryptodev-linux.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/fast-classifier.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/gpio-button-hotplug.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/fullconenat.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/ksmbd.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/shortcut-fe.symvers" KERNELRELEASE=5.18.12 EXTRA_CFLAGS="-I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/include -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1=backports-5.15.33-1 " KLIB_BUILD="/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.18.12" MODPROBE=true KLIB=/lib/modules/5.18.12 KERNEL_SUBLEVEL=18 KBUILD_LDFLAGS_MODULE_PREREQ= allnoconfig
make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1'
make[5]: 'Kconfig.versions' is up to date.
make[6]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1'
rm -rf /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/modules
make  -C "/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Jul 19 06:51:16 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/cryptodev-linux.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/fast-classifier.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/gpio-button-hotplug.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/fullconenat.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/ksmbd.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/shortcut-fe.symvers" KERNELRELEASE=5.18.12 EXTRA_CFLAGS="-I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/include -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1=backports-5.15.33-1 " KLIB_BUILD="/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.18.12" MODPROBE=true KLIB=/lib/modules/5.18.12 KERNEL_SUBLEVEL=18 KBUILD_LDFLAGS_MODULE_PREREQ= modules
make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1'
make[5]: 'Kconfig.versions' is up to date.
make[7]: 'Kconfig.versions' is up to date.
make[8]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/ath/ath6kl/cfg80211.o
In file included from /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/ath/ath6kl/core.h:26,
                 from /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/ath/ath6kl/cfg80211.c:25:
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/include/net/cfg80211.h: In function 'wdev_address':
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/include/net/cfg80211.h:5561:22: warning: return discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   return wdev->netdev->dev_addr;
          ~~~~~~~~~~~~^~~~~~~~~~
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/ath/ath6kl/cfg80211.c: In function 'ath6kl_interface_add':
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/ath/ath6kl/cfg80211.c:3806:13: warning: passing argument 1 of 'memcpy' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  memcpy(ndev->dev_addr, ar->mac_addr, ETH_ALEN);
         ~~~~^~~~~~~~~~
In file included from ./include/linux/string.h:20,
                 from /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/backport-include/linux/string.h:3,
[... 6 lines cut ...]
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/ath/ath6kl/cfg80211.c:3808:21: error: assignment of read-only location '*ndev->dev_addr'
   ndev->dev_addr[0] = (ndev->dev_addr[0] ^ (1 << fw_vif_idx)) |
                     ^
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/ath/ath6kl/cfg80211.c:3812:22: error: assignment of read-only location '*(ndev->dev_addr + 4)'
    ndev->dev_addr[4] ^= 0x80;
                      ^~
make[11]: *** [scripts/Makefile.build:288: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/drivers/net/wireless/ath/ath6kl/cfg80211.o] Error 1
[... cut ...]
time: package/kernel/mac80211/regular/compile#1.20#0.53#2.20
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
```

WYC-2020, 2022-07-20:
我修好了稍等，我在测下另外的内核看有没影响

WYC-2020, 2022-07-20:
[image]
[image]
[010-fix-Remove-netif_rx_any_context-and-netif_rx_ni.patch.patch.txt](https://github.com/coolsnowwolf/lede/files/9147140/010-fix-Remove-netif_rx_any_context-and-netif_rx_ni.patch.patch.txt)
[010-fix-Remove-netif_rx_any_context-and-netif_rx_ni.patch.txt](https://github.com/coolsnowwolf/lede/files/9147141/010-fix-Remove-netif_rx_any_context-and-netif_rx_ni.patch.txt)
 自己对照的图片 放到相应的位置，Makefile自己修改 @zhangguanzhang

zhangguanzhang, 2022-07-26:
凌晨源码编译的，还是存在该问题

zhangguanzhang, 2022-07-29:
@WYC-2020 大佬，你给的patch有个failed了：
```
Applying ./patches/ath6kl/010-fix-Remove-netif_rx_any_context-and-netif_rx_ni.patch.patch using plaintext: 
can't find file to patch at input line 14
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From bd416f7d4216dc177712772f94eb18b67d05208f Mon Sep 17 00:00:00 2001
|From: W_Y_CPP <383152993@qq.com>
|Date: Wed, 20 Jul 2022 13:42:35 +0800
|Subject: [PATCH] fix Remove netif_rx_any_context and netif_rx_ni
|
|---
| include/linux/netdevice.h | 10 ++++++++++
| 1 file changed, 10 insertions(+)
|
make[2]: *** [Makefile:566: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/.prepared_59f7cf577d190729ac90e53c2edc7[728](https://github.com/zhangguanzhang/Actions-OpenWrt/runs/7571385044?check_suite_focus=true#step:18:729)_666451[739](https://github.com/zhangguanzhang/Actions-OpenWrt/runs/7571385044?check_suite_focus=true#step:18:740)9ebbbc92a37c5bb081b5c53] Error 1
|diff --git a/include/linux/netdevice.h b/include/linux/netdevice.h
|index bfa27972f..a59dd2d5c 100644
|--- a/include/linux/netdevice.h
|+++ b/include/linux/netdevice.h
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch failed!  Please fix ./patches/ath6kl/010-fix-Remove-netif_rx_any_context-and-netif_rx_ni.patch.patch!
make[2]: Leaving directory '/workdir/openwrt/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#1.18#0.40#1.50
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
```

zhangguanzhang, 2022-07-31:
切到 5.4 内核编译过了这个包

zhangguanzhang, 2022-08-12:
最新源码不切内核版本编译还是失败
```
make -C "/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Aug 12 01:06:39 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/cryptodev-linux.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/fast-classifier.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/gpio-button-hotplug.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/fullconenat.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/ksmbd.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/shortcut-fe.symvers" KERNELRELEASE=5.19.1 EXTRA_CFLAGS="-I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/include -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1=backports-5.15.33-1 " KLIB_BUILD="/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.19.1" MODPROBE=true KLIB=/lib/modules/5.19.1 KERNEL_SUBLEVEL=19 KBUILD_LDFLAGS_MODULE_PREREQ= allnoconfig
make[3]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1'
make[4]: 'Kconfig.versions' is up to date.
make[5]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
make[3]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1'
rm -rf /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/modules
make  -C "/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-aarch64_generic_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Fri Aug 12 01:06:39 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/cryptodev-linux.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/fast-classifier.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/gpio-button-hotplug.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/fullconenat.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/ksmbd.symvers /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/shortcut-fe.symvers" KERNELRELEASE=5.19.1 EXTRA_CFLAGS="-I/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/include -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1=backports-5.15.33-1 " KLIB_BUILD="/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.19.1" MODPROBE=true KLIB=/lib/modules/5.19.1 KERNEL_SUBLEVEL=19 KBUILD_LDFLAGS_MODULE_PREREQ= modules
make[3]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1'
make[4]: 'Kconfig.versions' is up to date.
make[6]: 'Kconfig.versions' is up to date.
make[7]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/net/mac80211/iface.o
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/net/mac80211/iface.c: In function 'ieee80211_do_open':
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/net/mac80211/iface.c:1282:19: error: 'ndev' undeclared (first use in this function); did you mean 'dev'?
   eth_hw_addr_set(ndev, params->macaddr);
                   ^~~~
                   dev
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/net/mac80211/iface.c:1282:19: note: each undeclared identifier is reported only once for each function it appears in
[... cut ...]
make[4]: *** [Makefile.real:93: modules] Error 2
make[3]: *** [Makefile:121: modules] Error 2
make[2]: *** [Makefile:559: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/backports-5.15.33-1/.built] Error 2
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[1]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make: *** [/workdir/openwrt/include/toplevel.mk:230: package/kernel/mac80211/compile] Error 2
```

zhangguanzhang, 2022-08-12:
@asushugo 大佬能帮忙看看吗

aranyalau, 2022-08-12:
> 切到 5.4 内核编译过了这个包

怎么切换呀 我也是编译N1的时候出现了这个错误

zhangguanzhang, 2022-08-12:
> > 切到 5.4 内核编译过了这个包
> 
> 怎么切换呀 我也是编译N1的时候出现了这个错误

你的日志发下

freedyYang, 2022-08-13:
N1换成KERNEL_TESTING_PATCHVER:=5.4 还是报错
cc1: some warnings being treated as errors
make[12]: *** [scripts/Makefile.build:286: /home/freedy/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/backports-5.15.33-1/drivers/net/wireless/broadcom/brcm80211/brcmfmac/sdio.o] Error 1
make[11]: *** [scripts/Makefile.build:503: /home/freedy/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/backports-5.15.33-1/drivers/net/wireless/broadcom/brcm80211/brcmfmac] Error 2
make[10]: *** [scripts/Makefile.build:503: /home/freedy/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/backports-5.15.33-1/drivers/net/wireless/broadcom/brcm80211] Error 2
make[9]: *** [scripts/Makefile.build:503: /home/freedy/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/backports-5.15.33-1/drivers/net/wireless/broadcom] Error 2
make[8]: *** [scripts/Makefile.build:503: /home/freedy/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/backports-5.15.33-1/drivers/net/wireless] Error 2
make[8]: *** Waiting for unfinished jobs....
make[7]: *** [Makefile:1831: /home/freedy/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/backports-5.15.33-1] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/home/freedy/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/backports-5.15.33-1'
make[3]: *** [Makefile:559: /home/freedy/lede/build_dir/target-aarch64_cortex-a53_musl/linux-armvirt_64/backports-5.15.33-1/.built] Error 2
make[3]: Leaving directory '/home/freedy/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#4.31#1.19#30.36
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: *** Waiting for unfinished jobs....

zhangguanzhang, 2022-08-13:
你再试试，brcm这个最新代码修复了的

freedyYang, 2022-08-13:
刚刚拉了最新的代码，还是一样

zhangguanzhang, 2022-08-13:
> 刚刚拉了最新的代码，还是一样

你看看make defconfig后是不是有brcm的网卡开了造成的

closed 2022-09-06 by aiamadeus