coolsnowwolf/lede #9804: 5.18.12内核驱动drivers/net/phy/swconfig编译失败

lzy0702, 2022-07-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

5.18.12内核驱动drivers/net/phy/swconfig编译失败

### (2) 路由器型号和固件版本

平台x86_64
内核版本5.18.12
工具链GCC版本11.2

### (3) 详细日志

```
2022-07-19 18:45:57.741196   CC      drivers/net/phy/swconfig.o
In file included from ./include/linux/spinlock.h:303,
                 from ./include/linux/kref.h:16,
                 from ./include/linux/mm_types.h:8,
                 from ./include/linux/buildid.h:5,
                 from ./include/linux/module.h:14,
                 from drivers/net/phy/swconfig.c:18:
drivers/net/phy/swconfig_leds.c: In function 'swconfig_trig_update_port_mask':
drivers/net/phy/swconfig_leds.c:88:19: error: passing argument 1 of '_raw_read_lock' from incompatible pointer type [-Werror=incompatible-pointer-types]
   88 |         read_lock(&trigger->leddev_list_lock);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                   |
      |                   spinlock_t * {aka struct spinlock *}
./include/linux/rwlock.h:56:48: note: in definition of macro 'read_lock'
   56 | #define read_lock(lock)         _raw_read_lock(lock)
      |                                                ^~~~
In file included from ./include/linux/spinlock_api_smp.h:183,
                 from ./include/linux/spinlock.h:310,
                 from ./include/linux/kref.h:16,
                 from ./include/linux/mm_types.h:8,
                 from ./include/linux/buildid.h:5,
                 from ./include/linux/module.h:14,
                 from drivers/net/phy/swconfig.c:18:
./include/linux/rwlock_api_smp.h:18:42: note: expected 'rwlock_t *' but argument is of type 'spinlock_t *' {aka 'struct spinlock *'}
   18 | void __lockfunc _raw_read_lock(rwlock_t *lock)          __acquires(lock);
      |                                ~~~~~~~~~~^~~~
In file included from ./include/linux/spinlock.h:303,
                 from ./include/linux/kref.h:16,
                 from ./include/linux/mm_types.h:8,
                 from ./include/linux/buildid.h:5,
                 from ./include/linux/module.h:14,
                 from drivers/net/phy/swconfig.c:18:
drivers/net/phy/swconfig_leds.c:101:21: error: passing argument 1 of '_raw_read_unlock' from incompatible pointer type [-Werror=incompatible-pointer-types]
  101 |         read_unlock(&trigger->leddev_list_lock);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                     |
      |                     spinlock_t * {aka struct spinlock *}
./include/linux/rwlock.h:96:58: note: in definition of macro 'read_unlock'
   96 | #define read_unlock(lock)               _raw_read_unlock(lock)
      |                                                          ^~~~
In file included from ./include/linux/spinlock_api_smp.h:183,
                 from ./include/linux/spinlock.h:310,
                 from ./include/linux/kref.h:16,
                 from ./include/linux/mm_types.h:8,
                 from ./include/linux/buildid.h:5,
                 from ./include/linux/module.h:14,
                 from drivers/net/phy/swconfig.c:18:
./include/linux/rwlock_api_smp.h:31:44: note: expected 'rwlock_t *' but argument is of type 'spinlock_t *' {aka 'struct spinlock *'}
   31 | void __lockfunc _raw_read_unlock(rwlock_t *lock)        __releases(lock);
      |                                  ~~~~~~~~~~^~~~
In file included from ./include/linux/spinlock.h:303,
                 from ./include/linux/kref.h:16,
                 from ./include/linux/mm_types.h:8,
                 from ./include/linux/buildid.h:5,
                 from ./include/linux/module.h:14,
                 from drivers/net/phy/swconfig.c:18:
drivers/net/phy/swconfig_leds.c: In function 'swconfig_trig_update_leds':
drivers/net/phy/swconfig_leds.c:421:19: error: passing argument 1 of '_raw_read_lock' from incompatible pointer type [-Werror=incompatible-pointer-types]
  421 |         read_lock(&trigger->leddev_list_lock);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                   |
      |                   spinlock_t * {aka struct spinlock *}
./include/linux/rwlock.h:56:48: note: in definition of macro 'read_lock'
   56 | #define read_lock(lock)         _raw_read_lock(lock)
      |                                                ^~~~
In file included from ./include/linux/spinlock_api_smp.h:183,
                 from ./include/linux/spinlock.h:310,
                 from ./include/linux/kref.h:16,
                 from ./include/linux/mm_types.h:8,
                 from ./include/linux/buildid.h:5,
                 from ./include/linux/module.h:14,
                 from drivers/net/phy/swconfig.c:18:
./include/linux/rwlock_api_smp.h:18:42: note: expected 'rwlock_t *' but argument is of type 'spinlock_t *' {aka 'struct spinlock *'}
   18 | void __lockfunc _raw_read_lock(rwlock_t *lock)          __acquires(lock);
      |                                ~~~~~~~~~~^~~~
In file included from ./include/linux/spinlock.h:303,
                 from ./include/linux/kref.h:16,
                 from ./include/linux/mm_types.h:8,
                 from ./include/linux/buildid.h:5,
                 from ./include/linux/module.h:14,
                 from drivers/net/phy/swconfig.c:18:
drivers/net/phy/swconfig_leds.c:428:21: error: passing argument 1 of '_raw_read_unlock' from incompatible pointer type [-Werror=incompatible-pointer-types]
  428 |         read_unlock(&trigger->leddev_list_lock);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                     |
      |                     spinlock_t * {aka struct spinlock *}
./include/linux/rwlock.h:96:58: note: in definition of macro 'read_unlock'
   96 | #define read_unlock(lock)               _raw_read_unlock(lock)
      |                                                          ^~~~
In file included from ./include/linux/spinlock_api_smp.h:183,
                 from ./include/linux/spinlock.h:310,
                 from ./include/linux/kref.h:16,
                 from ./include/linux/mm_types.h:8,
                 from ./include/linux/buildid.h:5,
                 from ./include/linux/module.h:14,
                 from drivers/net/phy/swconfig.c:18:
./include/linux/rwlock_api_smp.h:31:44: note: expected 'rwlock_t *' but argument is of type 'spinlock_t *' {aka 'struct spinlock *'}
   31 | void __lockfunc _raw_read_unlock(rwlock_t *lock)        __releases(lock);
      |                                  ~~~~~~~~~~^~~~
cc1: some warnings being treated as errors
make[8]: *** [scripts/Makefile.build:288: drivers/net/phy/swconfig.o] Error 1
make[7]: *** [scripts/Makefile.build:550: drivers/net/phy] Error 2
make[6]: *** [scripts/Makefile.build:550: drivers/net] Error 2
make[5]: *** [Makefile:1838: drivers] Error 2
2022-07-19 18:45:59.206314 make[5]: Leaving directory '/home/lzy0702/lede-working/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.12'
make[4]: *** [Makefile:29: /home/lzy0702/lede-working/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18.12/.modules] Error 2
2022-07-19 18:45:59.206528 make[4]: Leaving directory '/home/lzy0702/lede-working/openwrt/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
2022-07-19 18:45:59.206706 make[3]: Leaving directory '/home/lzy0702/lede-working/openwrt/target/linux'
2022-07-19 18:45:59.207207 time: target/linux/compile#10.17#4.50#14.93
2022-07-19 18:45:59.208456     ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
2022-07-19 18:45:59.209511 make[2]: Leaving directory '/home/lzy0702/lede-working/openwrt'
make[1]: *** [target/Makefile:18: /home/lzy0702/lede-working/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
2022-07-19 18:45:59.211966 make[1]: Leaving directory '/home/lzy0702/lede-working/openwrt'
make: *** [/home/lzy0702/lede-working/openwrt/include/toplevel.mk:230：world] 错误 2
```

Zaphkito, 2022-07-20:
Ubuntu 22.04 LTS勾选swconfig编译最新代码没问题

lzy0702, 2022-07-20:
> Ubuntu 22.04 LTS勾选swconfig编译最新代码没问题
我也是UBuntu 22.04 LTS 
kernel_menuconfig里这部分配置如下
```
#
# Switch configuration API + drivers
#
CONFIG_SWCONFIG=y
CONFIG_SWCONFIG_LEDS=y
CONFIG_ADM6996_PHY=y
CONFIG_AR8216_PHY=y
CONFIG_AR8216_PHY_LEDS=y
# CONFIG_SWCONFIG_B53 is not set
CONFIG_IP17XX_PHY=y
CONFIG_PSB6970_PHY=y
CONFIG_RTL8306_PHY=y
CONFIG_RTL8366_SMI=y
# CONFIG_RTL8366_SMI_DEBUG_FS is not set
CONFIG_RTL8366S_PHY=y
CONFIG_RTL8366RB_PHY=y
CONFIG_RTL8367_PHY=y
CONFIG_RTL8367B_PHY=y

#
# MII PHY device drivers
#
CONFIG_AMD_PHY=y
CONFIG_MESON_GXL_PHY=y
CONFIG_ADIN_PHY=y
CONFIG_AQUANTIA_PHY=y
CONFIG_AX88796B_PHY=y
CONFIG_BROADCOM_PHY=m
CONFIG_BCM54140_PHY=m
CONFIG_BCM63XX_PHY=m
CONFIG_BCM7XXX_PHY=y
CONFIG_BCM84881_PHY=m
CONFIG_BCM87XX_PHY=m
CONFIG_BCM_NET_PHYLIB=y
CONFIG_CICADA_PHY=y
CONFIG_CORTINA_PHY=y
CONFIG_DAVICOM_PHY=y
CONFIG_ICPLUS_PHY=y
CONFIG_LXT_PHY=y
CONFIG_INTEL_XWAY_PHY=y
CONFIG_LSI_ET1011C_PHY=y
CONFIG_MARVELL_PHY=y
CONFIG_MARVELL_10G_PHY=y
CONFIG_MARVELL_88X2222_PHY=y
CONFIG_MAXLINEAR_GPHY=y
CONFIG_MEDIATEK_GE_PHY=y
CONFIG_MICREL_PHY=y
CONFIG_MICROCHIP_PHY=y
CONFIG_MICROCHIP_T1_PHY=y
CONFIG_MICROSEMI_PHY=y
CONFIG_MOTORCOMM_PHY=m
CONFIG_NATIONAL_PHY=y
CONFIG_NXP_C45_TJA11XX_PHY=y
CONFIG_NXP_TJA11XX_PHY=m
CONFIG_AT803X_PHY=y
CONFIG_QSEMI_PHY=y
CONFIG_REALTEK_PHY=y
CONFIG_RENESAS_PHY=y
CONFIG_ROCKCHIP_PHY=y
CONFIG_SMSC_PHY=y
CONFIG_STE10XP=y
CONFIG_TERANETICS_PHY=y
CONFIG_DP83822_PHY=y
CONFIG_DP83TC811_PHY=y
CONFIG_DP83848_PHY=y
CONFIG_DP83867_PHY=y
CONFIG_DP83869_PHY=y
CONFIG_VITESSE_PHY=y
CONFIG_XILINX_GMII2RGMII=y
# CONFIG_MICREL_KS8995MA is not set
CONFIG_MDIO_DEVICE=y
CONFIG_MDIO_BUS=y
CONFIG_FWNODE_MDIO=y
CONFIG_OF_MDIO=y
CONFIG_ACPI_MDIO=y
CONFIG_MDIO_DEVRES=y
# CONFIG_MDIO_SUN4I is not set
CONFIG_MDIO_XGENE=y
# CONFIG_MDIO_ASPEED is not set
CONFIG_MDIO_BITBANG=y
# CONFIG_MDIO_BCM_IPROC is not set
CONFIG_MDIO_BCM_UNIMAC=y
CONFIG_MDIO_CAVIUM=y
CONFIG_MDIO_GPIO=m
# CONFIG_MDIO_HISI_FEMAC is not set
CONFIG_MDIO_MVUSB=m
CONFIG_MDIO_MSCC_MIIM=y
CONFIG_MDIO_MOXART=m
# CONFIG_MDIO_OCTEON is not set
# CONFIG_MDIO_IPQ4019 is not set
# CONFIG_MDIO_IPQ8064 is not set
CONFIG_MDIO_THUNDER=y
```

WYC-2020, 2022-07-20:
万能大法

Zaphkito, 2022-07-20:
先老老实实用5.15不香吗，一个路由器系统别那么着急追新

lzy0702, 2022-07-20:
> 先老老实实用5.15不香吗，一个路由器系统别那么着急追新

反正编译一遍也花不了多久，两个版本同时编译

coolsnowwolf, 2022-07-20:
x86 没有switch ，别乱选

WYC-2020, 2022-07-20:
[image]
这个玩意在x86 下面是不会定义的 你怕是自己去修改了内核的默认配置，不建议去修改，除非明确知道这样可行，l大已经给你说了 x86不支持这个玩意，如果x86 你选中了kmod-swconfig 模块，默认生成应该是这样才对，虽然x86不支持没有任何意义，但是编译也不会出错
[image]
[image]
@lzy0702

lzy0702, 2022-07-20:
我知道这玩意平时遇不到，编译拿来用的内核也不会开启它，只是觉得不应该以这种方式出错。

WYC-2020, 2022-07-20:
> 我知道这玩意平时遇不到，编译拿来用的内核也不会开启它，只是觉得不应该以这种方式出错。

为啥不会 这玩意又不是内核自带的，没匹配好这不是常有的事

WYC-2020, 2022-07-20:
> > 我知道这玩意平时遇不到，编译拿来用的内核也不会开启它，只是觉得不应该以这种方式出错。
> 
> 为啥不会 这玩意又不是内核自带的，没匹配好这不是常有的事

我看下能不能修复一下

lzy0702, 2022-07-20:
会不会和5.18从以前的C89转换到了C11标准有关？

WYC-2020, 2022-07-20:
> 会不会和5.18从以前的C89转换到了C11标准有关？

怎么可能 ，锁类型变了而已，改成自旋锁了以前是读写锁，改下就ok 简单

WYC-2020, 2022-07-20:
@lzy0702  把你的config 发给我我测试一下

lzy0702, 2022-07-20:
> > [config.zip](https://github.com/coolsnowwolf/lede/files/9148134/config.zip) .config是openwrt/.config config-5.18是openwrt/target/linux/x86/config-5.18 kernel5.18.config是openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.18*/.config
> 
> 没对 你给的这个是编译5.15内核

.config就改了个选用测试版内核
这个附件里的已经又改过别的选项了
[config.zip](https://github.com/coolsnowwolf/lede/files/9148298/config.zip)

lzy0702, 2022-07-20:
我这时间已经半夜了，先睡遁了

WYC-2020, 2022-07-20:
> 我这时间已经半夜了，先睡遁了

已修好，你还真是无意间发现问题哈哈

closed 2022-07-22 by aiamadeus

totalatot, 2022-07-26:
Will get similar errors compiling with 5.15 testing kernel.

totalatot, 2022-07-26:
CC      drivers/net/phy/swconfig.o
In file included from drivers/net/phy/swconfig.c:32:
drivers/net/phy/swconfig_leds.c: In function 'swconfig_trig_update_port_mask':
drivers/net/phy/swconfig_leds.c:89:12: error: passing argument 1 of 'spin_lock' from incompatible pointer type [-Werror=incompatible-pointer-types]
  spin_lock(&trigger->leddev_list_lock);
            ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./include/linux/mm_types.h:9,
                 from ./include/linux/buildid.h:5,
                 from ./include/linux/module.h:14,
                 from drivers/net/phy/swconfig.c:18:
./include/linux/spinlock.h:361:51: note: expected 'spinlock_t *' {aka 'struct spinlock *'} but argument is of type 'rwlock_t *' {aka 'struct <anonymous> *'}
 static __always_inline void spin_lock(spinlock_t *lock)
                                       ~~~~~~~~~~~~^~~~
In file included from drivers/net/phy/swconfig.c:32:
drivers/net/phy/swconfig_leds.c:106:14: error: passing argument 1 of 'spin_unlock' from incompatible pointer type [-Werror=incompatible-pointer-types]
  spin_unlock(&trigger->leddev_list_lock);
              ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./include/linux/mm_types.h:9,
                 from ./include/linux/buildid.h:5,
                 from ./include/linux/module.h:14,
                 from drivers/net/phy/swconfig.c:18:
./include/linux/spinlock.h:401:53: note: expected 'spinlock_t *' {aka 'struct spinlock *'} but argument is of type 'rwlock_t *' {aka 'struct <anonymous> *'}
 static __always_inline void spin_unlock(spinlock_t *lock)
                                         ~~~~~~~~~~~~^~~~
In file included from drivers/net/phy/swconfig.c:32:
drivers/net/phy/swconfig_leds.c: In function 'swconfig_trig_update_leds':
drivers/net/phy/swconfig_leds.c:429:12: error: passing argument 1 of 'spin_lock' from incompatible pointer type [-Werror=incompatible-pointer-types]
  spin_lock(&trigger->leddev_list_lock);
            ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./include/linux/mm_types.h:9,
                 from ./include/linux/buildid.h:5,
                 from ./include/linux/module.h:14,
                 from drivers/net/phy/swconfig.c:18:
./include/linux/spinlock.h:361:51: note: expected 'spinlock_t *' {aka 'struct spinlock *'} but argument is of type 'rwlock_t *' {aka 'struct <anonymous> *'}
 static __always_inline void spin_lock(spinlock_t *lock)
                                       ~~~~~~~~~~~~^~~~
In file included from drivers/net/phy/swconfig.c:32:
drivers/net/phy/swconfig_leds.c:440:14: error: passing argument 1 of 'spin_unlock' from incompatible pointer type [-Werror=incompatible-pointer-types]
  spin_unlock(&trigger->leddev_list_lock);
              ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ./include/linux/mm_types.h:9,
                 from ./include/linux/buildid.h:5,
                 from ./include/linux/module.h:14,
                 from drivers/net/phy/swconfig.c:18:
./include/linux/spinlock.h:401:53: note: expected 'spinlock_t *' {aka 'struct spinlock *'} but argument is of type 'rwlock_t *' {aka 'struct <anonymous> *'}
 static __always_inline void spin_unlock(spinlock_t *lock)
                                         ~~~~~~~~~~~~^~~~
cc1: some warnings being treated as errors
make[8]: *** [scripts/Makefile.build:289: drivers/net/phy/swconfig.o] Error 1
make[7]: *** [scripts/Makefile.build:552: drivers/net/phy] Error 2
make[6]: *** [scripts/Makefile.build:552: drivers/net] Error 2
make[5]: *** [Makefile:1893: drivers] Error 2
make[5]: Leaving directory '/home/total/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.15.57'
make[4]: *** [Makefile:22: /home/total/lede/build_dir/target-mips_24kc_musl/linux-ath79_nand/linux-5.15.57/.modules] Error 2
make[4]: Leaving directory '/home/total/lede/target/linux/ath79'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/total/lede/target/linux'
time: target/linux/compile#261.25#31.93#284.71
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/total/lede'
make[1]: *** [target/Makefile:18: /home/total/lede/staging_dir/target-mips_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/total/lede'
make: *** [/home/total/lede/include/toplevel.mk:230: world] Error 2

totalatot, 2022-07-26:
Ubuntu 22.04
[generic: fix swconfig_leds.c in 5.18](https://github.com/coolsnowwolf/lede/commit/ebfa1b93b76d5a2faacecbc8ed237ed44f5a7018)