coolsnowwolf/lede #9810: 5.18内核编译turboacc"Installing packages"部分报错

Zaphkito, 2022-07-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：5.18内核中，turboacc会存在在installing中失败的情况，相同设置5.15可以编译成功

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：
[image]
[image]

WYC-2020, 2022-07-21:
[image]
 二选一不能同时选

WYC-2020, 2022-07-21:
我搞了一个修改版 可以同时兼容，
[image]
需要自己去看，我去除了dns那部分哈 ，想要的 自己参考我这个改

xiangfeidexiaohuo, 2022-07-21:
> 我搞了一个修改版 可以同时兼容， [image] 需要自己去看，我去除了dns那部分哈 ，想要的 自己参考我这个改

这个有点刷，同时内置几种，有点意思。

Zaphkito, 2022-07-21:
[image]
[image]

只勾选一个也会出现编译失败的情况，只勾选fcm的话可以完成编译

WYC-2020, 2022-07-22:
> [image] [image]
> 
> 只勾选一个也会出现编译失败的情况，只勾选fcm的话可以完成编译

去除不干净，还有这个的依赖也要去掉，在kernel module哪里，

Zaphkito, 2022-07-22:
> > > [image] [image]
> > > 只勾选一个也会出现编译失败的情况，只勾选fcm的话可以完成编译
> > 
> > 
> > 去除不干净，还有这个的依赖也要去掉，在kernel module哪里，
> 
> 但是我是重新编的config，我再试试好了（

[image]
[image]
只勾选下面的以后这个选项会被强行勾选，取消勾选sfe后才能取消依赖
虽然不清楚是bug还是啥，我直接不勾选sfe得了

WYC-2020, 2022-07-22:
[image]
这个二选一

[image]
[image]
这两个二选一
@Zaphkito

closed 2022-07-22 by Zaphkito