coolsnowwolf/lede #9814: lean大，这个功能可以加上嘛，挺需要的

AngeloASX, 2022-07-22:
R68s的固件可以加上这个功能嘛，刚需
[image]

WYC-2020, 2022-07-22:
这那个固件的嘛，直接这部分逻辑抠出来就行

AngeloASX, 2022-07-22:
看看这个源码能用嘛，https://github.com/Twinzo1/openwrt/tree/master/luci-app-mac

zhangguanzhang, 2022-07-22:
这个自己加进去就行，可以放到 `/package/openwrt-packages`

WYC-2020, 2022-07-22:
> 看看这个源码能用嘛，https://github.com/Twinzo1/openwrt/tree/master/luci-app-mac

你这个是第三方的 ，不会体现在接口哪里的 ，刚看了也就是一句话的事 luci.tools.proto.opt_macaddr(section, ifc, translate("Clone MAC address"))

shu0802, 2022-07-25:
> > 看看这个源码能用嘛，https://github.com/Twinzo1/openwrt/tree/master/luci-app-mac
> 
> 你这个是第三方的 ，不会体现在接口哪里的 ，刚看了也就是一句话的事 luci.tools.proto.opt_macaddr(section, ifc, translate("Clone MAC address"))

能详说一下，这句加到哪个里吗。谢谢！

coolsnowwolf, 2026-09-05:
该条只有图片和“希望增加功能”的描述，没有说明功能名称、目标硬件和验收方式，无法形成可验证的源码修改，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf