coolsnowwolf/lede #9820: R5S无法扩容/overlay

hashmaparraylist, 2022-07-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

参考eSir的/overlay扩容方法

通过命令将`/dev/mmcblk0p7`挂载到`/mnt/mmcblk0p7` 以后
从`/overlay`下复制所有文件以及文件夹到`/mnt/mmcblk0p7`

然后在`系统` > `挂载点`功能 将 `/dev/mmcblk0p7`挂到`/overlay`上以后，重启路由器。
`/overlay`空间没有扩大，同时在终端使用lsblk命令显示`mmcblk0p7`没有挂载点。

```shell
root@OpenWrtR5S:~# lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
loop0          7:0    0 933.1M  0 loop /overlay
mmcblk0      179:0    0  14.6G  0 disk
├─mmcblk0p1  179:1    0     4M  0 part
├─mmcblk0p2  179:2    0     4M  0 part
├─mmcblk0p3  179:3    0    32M  0 part
├─mmcblk0p4  179:4    0    32M  0 part
├─mmcblk0p5  179:5    0    32M  0 part /mnt/mmcblk0p5
├─mmcblk0p6  179:6    0     1G  0 part /rom
└─mmcblk0p7  179:7    0  13.5G  0 part
mmcblk0boot0 179:32   0     4M  1 disk
mmcblk0boot1 179:64   0     4M  1 disk
```

### (2) 路由器型号和固件版本

- 型号: `FriendlyElec NanoPi R5S`
- 固件版本: `OpenWrt R22.7.19 / LuCI Master (git-22.200.61437-0c3c82f)`
- 内核版本: `4.19.237`

hashmaparraylist, 2022-07-22:
### (3) 详细日志

- 系统日志

```
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.844764] rockchip-thermal fe710000.tsadc: tsadc is probed successfully!
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.847123] cpu cpu0: leakage=41
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.847501] cpu cpu0: pvtm = 93160, from nvmem
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.848207] cpu cpu0: pvtm-volt-sel=2
Fri Jul 22 22:53:42 2022 kern.debug kernel: [    0.849235] cpu cpu0: Looking up cpu-supply from device tree
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.849489] cpu cpu0: Linked as a consumer to regulator.9
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.850099] cpu cpu0: Dropping the link to regulator.9
Fri Jul 22 22:53:42 2022 kern.debug kernel: [    0.851140] cpu cpu0: Looking up cpu-supply from device tree
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.851367] cpu cpu0: Linked as a consumer to regulator.9
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.853139] cpu cpu0: avs=0
Fri Jul 22 22:53:42 2022 kern.warn kernel: [    0.853480] energy_model: pd0: hertz/watts ratio non-monotonically decreasing: em_cap_state 2 >= em_cap_state1
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.854881] cpu cpu0: l=0 h=2147483647 hyst=5000 l_limit=0 h_limit=0 h_table=0
Fri Jul 22 22:53:42 2022 kern.err kernel: [    0.858547] cpu cpu0: failed to find power_model node
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.859826] sdhci: Secure Digital Host Controller Interface driver
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.860424] sdhci: Copyright(c) Pierre Ossman
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.861351] Synopsys Designware Multimedia Card Interface Driver
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.862357] sdhci-pltfm: SDHCI platform and OF driver helper
Fri Jul 22 22:53:42 2022 kern.debug kernel: [    0.864107] sdhci-dwcmshc fe310000.sdhci: Looking up vmmc-supply from device tree
Fri Jul 22 22:53:42 2022 kern.debug kernel: [    0.864115] sdhci-dwcmshc fe310000.sdhci: Looking up vmmc-supply property in node /sdhci@fe310000 failed
Fri Jul 22 22:53:42 2022 kern.debug kernel: [    0.864138] sdhci-dwcmshc fe310000.sdhci: Looking up vqmmc-supply from device tree
Fri Jul 22 22:53:42 2022 kern.debug kernel: [    0.864142] sdhci-dwcmshc fe310000.sdhci: Looking up vqmmc-supply property in node /sdhci@fe310000 failed
Fri Jul 22 22:53:42 2022 kern.err kernel: [    0.864154] mmc0: Unknown controller version (5). You may experience problems.
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.895941] mmc0: SDHCI controller on fe310000.sdhci [fe310000.sdhci] using ADMA
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.897560] hidraw: raw HID events driver (C) Jiri Kosina
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.898211] usbcore: registered new interface driver usbhid
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.899062] usbhid: USB HID core driver
Fri Jul 22 22:53:42 2022 kern.debug kernel: [    0.900137] rockchip,bus bus-npu: Looking up bus-supply from device tree
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.900190] rockchip,bus bus-npu: Linked as a consumer to regulator.10
Fri Jul 22 22:53:42 2022 kern.err kernel: [    0.900895] rockchip,bus bus-npu: Failed to get leakage
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.901985] rockchip,bus bus-npu: pvtm = 93160, from nvmem
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.902903] rockchip,bus bus-npu: pvtm-volt-sel=2
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.903976] rockchip,bus bus-npu: avs=0
Fri Jul 22 22:53:42 2022 kern.debug kernel: [    0.905625] rockchip-saradc fe720000.saradc: Looking up vref-supply from device tree
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.905686] rockchip-saradc fe720000.saradc: Linked as a consumer to regulator.21
Fri Jul 22 22:53:42 2022 kern.err kernel: [    0.906863] rksfc_base v1.1 2016-01-08
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.908039] rksfc fe300000.sfc: rksfc_probe clk rate = 99000000
Fri Jul 22 22:53:42 2022 kern.err kernel: [    0.908654] rkflash_dev_init enter
Fri Jul 22 22:53:42 2022 kern.warn kernel: [    0.909437] sfc nor id: 0 0 0
Fri Jul 22 22:53:42 2022 kern.err kernel: [    0.909712] rkflash[1] is invalid
Fri Jul 22 22:53:42 2022 kern.err kernel: [    0.909714] rkflash_dev_init enter
Fri Jul 22 22:53:42 2022 kern.warn kernel: [    0.910895] sfc_nand id: 0 0 0
Fri Jul 22 22:53:42 2022 kern.err kernel: [    0.911185] rkflash[2] is invalid
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.911672] Initializing XFRM netlink socket
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.913417] NET: Registered protocol family 10
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.914383] Segment Routing with IPv6
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.914786] NET: Registered protocol family 17
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.915257] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.916457] 8021q: 802.1Q VLAN Support v1.8
Fri Jul 22 22:53:42 2022 kern.warn kernel: [    0.916861] Mass Storage accelerater start.
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.919532] flash vendor_init_thread!
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.919876] flash vendor storage:20170308 ret = -1
Fri Jul 22 22:53:42 2022 kern.info kernel: [    0.921508] rga2: Driver loaded successfully ver:3.2.63318
[... cut ...]
Fri Jul 22 22:54:12 2022 daemon.notice ttyd[7128]: LWS: 4.1.6-20220505-273-g58120dbe3, loglevel 7
Fri Jul 22 22:54:12 2022 daemon.notice ttyd[7128]: NET CLI SRV H1 H2 WS IPV6-off
Fri Jul 22 22:54:12 2022 daemon.notice ttyd[7128]:    /usr/lib/libwebsockets-evlib_uv.so
Fri Jul 22 22:54:12 2022 daemon.notice ttyd[7128]:  Using foreign event loop...
Fri Jul 22 22:54:12 2022 daemon.notice ttyd[7128]:  Listening on port: 7681
Fri Jul 22 22:54:17 2022 daemon.notice procd: /etc/rc.d/S99zerotier: disabled in config
Fri Jul 22 22:54:17 2022 daemon.info procd: - init complete -
```

hashmaparraylist, 2022-07-22:
> 内核日志

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 4.19.237-android (lean@lean-esxi) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4779-6fd6edc2a), GNU ld (GNU Binutils) 2.34) #0 SMP Tue Jul 19 16:23:37 2022
[    0.000000] Machine model: FriendlyElec NanoPi R5S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] bootconsole [uart8250] enabled
[    0.000000] OF: fdt: Reserved memory: failed to reserve memory for node 'drm-cubic-lut@00000000': base 0x0000000000000000, size 0 MiB
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000010000000, size 8 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1043968
[    0.000000]   DMA32 zone: 15288 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 978432 pages, LIFO batch:63
[    0.000000]   Normal zone: 1024 pages used for memmap
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 22 pages/cpu s51176 r8192 d30744 u90112
[    0.000000] pcpu-alloc: s51176 r8192 d30744 u90112 alloc=22*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Speculative Store Bypassing Safe (SSBS)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1027656
[    0.000000] Kernel command line: storagemedia=emmc androidboot.storagemedia=emmc androidboot.mode=normal  androidboot.verifiedbootstate=orange androidboot.serialno=71b9bc8b93cb9f68  			 rw rootwait console=ttyS2,1500000 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=614e0000-0000 rootfstype=squashfs,ext4
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0xe9f00000-0xedf00000] (64MB)
[    0.000000] Memory: 3995924K/4175872K available (9662K kernel code, 782K rwdata, 3420K rodata, 832K init, 474K bss, 171756K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] GICv3: no VLPI support, no direct LPI support
[    0.000000] ITS [mem 0xfd440000-0xfd45ffff]
[    0.000000] ITS@0x00000000fd440000: allocated 8192 Devices @210000 (indirect, esz 8, psz 64K, shr 0)
[    0.000000] ITS@0x00000000fd440000: allocated 32768 Interrupt Collections @220000 (flat, esz 2, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] GIC: using LPI property table @0x0000000000230000
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] CPU0: using LPI pending table @0x0000000000240000
[    0.000000] GIC: using cache flushing for LPI property table
[    0.000000] random: random: get_random_bytes called from start_kernel+0x344/0x434 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000005] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001856] Console: colour dummy device 80x25
[    0.002332] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=24000)
[    0.003322] pid_max: default: 32768 minimum: 301
[    0.003922] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes)
[    0.004590] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes)
[    0.006927] ASID allocator initialised with 32768 entries
[    0.007565] rcu: Hierarchical SRCU implementation.
[    0.008448] Platform MSI: interrupt-controller@fd440000 domain created
[    0.009564] PCI/MSI: /interrupt-controller@fd400000/interrupt-controller@fd440000 domain created
[    0.011063] smp: Bringing up secondary CPUs ...
[    0.012839] Detected VIPT I-cache on CPU1
[    0.012877] GICv3: CPU1: found redistributor 100 region 0:0x00000000fd480000
[    0.012934] CPU1: using LPI pending table @0x0000000000250000
[    0.012987] CPU1: Booted secondary processor 0x0000000100 [0x412fd050]
[... 769 lines cut ...]
[   11.497684] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   11.507313] eth2: 0xffffff800e000000, 7c:b5:9b:6d:96:02, IRQ 114
[   11.554389] br-lan: port 2(eth2) entered blocking state
[   11.554897] br-lan: port 2(eth2) entered disabled state
[   11.556077] device eth2 entered promiscuous mode
[... cut ...]
[   14.330616] br-lan: port 1(eth1) entered blocking state
[   14.331416] br-lan: port 1(eth1) entered forwarding state
[   14.332136] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   20.466799] overlayfs: filesystem on '/opt/docker/check-overlayfs-support1278602272/upper' not supported as upperdir
[   20.482587] overlayfs: filesystem on '/opt/docker/check-overlayfs-support1946515303/upper' not supported as upperdir
[   20.726675] IPv6: ADDRCONF(NETDEV_UP): docker0: link is not ready
[   31.153144] vcc3v3_pcie: disabling
```

ecrasy, 2022-07-23:
可以考虑ext4固件，烧写到存储设备之后，用虚拟机启动GParted手动给存储设备的分区永久性扩容。

zhangguanzhang, 2022-07-23:
等后面我做 r5s 固件 😎，squashfs 也能扩容，不会创建第三个分区

closed 2022-07-25 by aiamadeus

shen-lan, 2022-11-29:
> 等后面我做 r5s 固件 😎，squashfs 也能扩容，不会创建第三个分区

馆长能帮忙看下 R5s eth2无法正常识别的问题吗 ？Q群里问了也没人回，哎
https://github.com/coolsnowwolf/lede/issues/10130#issuecomment-1330933592

zhangguanzhang, 2022-11-30:
> > 等后面我做 r5s 固件 😎，squashfs 也能扩容，不会创建第三个分区
> 
> 馆长能帮忙看下 R5s eth2无法正常识别的问题吗 ？Q群里问了也没人回，哎 [#10130 (comment)](https://github.com/coolsnowwolf/lede/issues/10130#issuecomment-1330933592)

这个问题我记得是只有友善固件没这问题，骷髅头的固件貌似也解决了(他好像是吧pice还是啥功率调高了解决的)