coolsnowwolf/lede #9821: R4S 最低频率调节

ZZMHA227, 2022-07-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
已经把R4S的小核调到了408MHz了，但好像没生效，还是600MHz在跑。
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
[image]
[image]
[image]

### (2) 路由器型号和固件版本

A：R4S

### (3) 详细日志

A：[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.18.12 (runner@fv-az121-875) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 8.4.0 r4737-7af074fd1) 8.4.0, GNU ld (GNU Binutils) 2.34) #0 SMP PREEMPT Thu Jul 21 08:58:40 2022
[    0.000000] Machine model: FriendlyElec NanoPi R4S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff1a0000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 8 MiB at 0x00000000f3000000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 18 pages/cpu s34664 r8192 d30872 u73728
[    0.000000] pcpu-alloc: s34664 r8192 d30872 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 999432
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff1a0000 root=PARTUUID=52fdaf93-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 3956748K/4061184K available (9536K kernel code, 2454K rwdata, 2764K rodata, 2048K init, 797K bss, 96244K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=6.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=6
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 256 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] Root IRQ handler: gi?hƸq
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fef00000
[... 346 lines cut ...]
[    8.897285] usbcore: registered new interface driver ums-sddr55
[    8.898321] usbcore: registered new interface driver ums-usbat
[    8.910383] xt_time: kernel timezone is -0000
[    8.913172] PPP generic driver version 2.4.2
[    8.913996] PPP MPPE Compression module registered
[    8.914821] NET: Registered PF_PPPOX protocol family
[    8.916222] kmodloader: done loading kernel modules from /etc/modules.d/*
[   11.028269] EXT4-fs (mmcblk1p1): mounted filesystem without journal. Quota mode: none.
[   11.035769] EXT4-fs (mmcblk1p3): mounted filesystem with ordered data mode. Quota mode: none.
[   11.118056] eth1: 0xffff800009187000, 82:34:28:34:a4:d6, IRQ 89
[   11.178027] br-lan: port 1(eth1) entered blocking state
[   11.178525] br-lan: port 1(eth1) entered disabled state
[   11.179213] device eth1 entered promiscuous mode
[   11.227140] rk_gmac-dwmac fe300000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet] (irq=87)
[   11.230031] rk_gmac-dwmac fe300000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   12.750001] rk_gmac-dwmac fe300000.ethernet eth0: No Safety Features support found
[   12.750686] rk_gmac-dwmac fe300000.ethernet eth0: PTP not supported by HW
[   12.751834] rk_gmac-dwmac fe300000.ethernet eth0: configuring for phy/rgmii link mode
[   15.286177] r8168: eth1: link up
[   15.286723] br-lan: port 1(eth1) entered blocking state
[   15.287198] br-lan: port 1(eth1) entered forwarding state
[   15.288035] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   31.717585] vbus_typec: disabling
[ 9428.473208] rd 0-5: Disabling EAS, schedutil is mandatory
[84072.655757] rd 0-5: Disabling EAS, schedutil is mandatory

closed 2022-07-25 by aiamadeus

keeno1982, 2022-08-05:
@ZZMHA227 为什么你编译出来的插件有大小核选项？是另外安装的吗

1989car, 2022-08-14:
> @ZZMHA227 为什么你编译出来的插件有大小核选项？是另外安装的吗

似乎是用了ImmortalWrt 源码里的cpufreq