coolsnowwolf/lede #9822: r4s 开启 kmod-gpu-lima 编译 linux 报错缺少依赖 drm_shmem_helper.ko

zhangguanzhang, 2022-07-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

r4s 开启 `kmod-gpu-lima` 编译失败
```
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_rockchip_armv8_DEVICE_friendlyarm_nanopi-r4s=y

CONFIG_TARGET_OPTIONS=y

CONFIG_TARGET_OPTIMIZATION="-O3 -pipe -march=armv8-a+crypto+crc -mcpu=cortex-a72.cortex-a53+crypto+crc -mtune=cortex-a72.cortex-a53"

CONFIG_PACKAGE_kmod-gpu-lima=y
```

### (2) 路由器型号和固件版本

A：r4s 

### (3) 详细日志

```
make[2]: Leaving directory '/workdir/openwrt/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#1.56#0.89#2.21
make[2]: Entering directory '/workdir/openwrt/package/kernel/linux'
Package kmod-gpu-lima is missing dependencies for the following libraries:
drm_shmem_helper.ko
make[2]: *** [/workdir/openwrt/target/linux/rockchip/modules.mk:68: /workdir/openwrt/bin/targets/rockchip/armv8/packages/kmod-gpu-lima_5.18.13-1_aarch64_generic.ipk] Error 1
make[2]: Leaving directory '/workdir/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#2.68#0.32#3.11
    ERROR: package/kernel/linux failed to build.
make[1]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: package/compile] Error 2
```

xiangfeidexiaohuo, 2022-07-24:
5.18的 drm相关 目前还有问题呢，非要这个，暂时回5.15吧

zhangguanzhang, 2022-07-24:
> 5.18的 drm相关 目前还有问题呢，非要这个，暂时回5.15吧

我试试

closed 2022-07-26 by zhangguanzhang