coolsnowwolf/lede #9825: 缺少wmi.ko

vipboddy, 2022-07-23:
Package kmod-hwmon-nct6775 is missing dependencies for the following libraries:
wmi.ko
make[2]: *** [modules/hwmon.mk:333: /mnt/op/lede/bin/targets/x86/64/packages/kmod-hwmon-nct6775_5.18.14-1_x86_64.ipk] Error 1
make[2]: Leaving directory '/mnt/op/lede/package/kernel/linux'
time: package/kernel/linux/compile#14.90#2.90#15.20
    ERROR: package/kernel/linux failed to build.
make[1]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[1]: Leaving directory '/mnt/op/lede'

WYC-2020, 2022-07-24:
我这边一切正常
[image]

vipboddy, 2022-07-24:
> 我这边一切正常 [image]

大佬忘记说明是5.18，正常吗这个编译

WYC-2020, 2022-07-24:
> > 我这边一切正常 [image]
> 
> 大佬忘记说明是5.18，正常吗这个编译

我就是5.18

closed 2023-04-26 by aiamadeus