coolsnowwolf/lede #9827: 帮忙看看系统日志。有什么问题吗？谢谢。

baodp, 2022-07-24:
X86 N5095 最新内核5.15.18
基本是默认配置。
系统日志：

Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] Mountpoint-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] CPU0: Thermal monitoring enabled (TM1)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] x86/cpu: User Mode Instruction Prevention (UMIP) activated
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] process: using mwait in idle threads
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] Last level iTLB entries: 4KB 0, 2MB 0, 4MB 0
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] Last level dTLB entries: 4KB 0, 2MB 0, 4MB 0, 1GB 0
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] Spectre V1 : Mitigation: usercopy/swapgs barriers and __user pointer sanitization
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] Spectre V2 : Mitigation: Enhanced IBRS
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] Spectre V2 : Spectre v2 / SpectreRSB mitigation: Filling RSB on context switch
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] Spectre V2 : mitigation: Enabling conditional Indirect Branch Prediction Barrier
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] Speculative Store Bypass: Mitigation: Speculative Store Bypass disabled via prctl
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] MMIO Stale Data: Vulnerable: Clear CPU buffers attempted, no microcode
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] SRBDS: Vulnerable: No microcode
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] Freeing SMP alternatives memory: 44K
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] smpboot: Estimated ratio of average max frequency by base frequency (times 1024): 1433
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] smpboot: CPU0: Intel(R) Celeron(R) N5095 @ 2.00GHz (family: 0x6, model: 0x9c, stepping: 0x0)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268546] cblist_init_generic: Setting adjustable number of callback queues.
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.268549] cblist_init_generic: Setting shift to 2 and lim to 1.
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.274664] cblist_init_generic: Setting shift to 2 and lim to 1.
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.278565] Performance Events: PEBS fmt4+-baseline, PEBS-via-PT,  AnyThread deprecated, Tremont events, 32-deep LBR, full-width counters, Intel PMU driver.
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.288548] ... version:                5
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.292557] ... bit width:              48
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.296652] ... generic registers:      4
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.298547] ... value mask:             0000ffffffffffff
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.303858] ... max period:             00007fffffffffff
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.308547] ... fixed-purpose events:   3
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.312554] ... event mask:             000000070000000f
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.317944] rcu: Hierarchical SRCU implementation.
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.318588] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.326308] smp: Bringing up secondary CPUs ...
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.328643] x86: Booting SMP configuration:
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.332831] .... node  #0, CPUs:      #1 #2 #3
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.334450] smp: Brought up 1 node, 4 CPUs
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.342655] smpboot: Max logical packages: 1
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.346920] smpboot: Total of 4 processors activated (15974.40 BogoMIPS)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.349359] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.358550] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.365448] pinctrl core: initialized pinctrl subsystem
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.368984] NET: Registered PF_NETLINK/PF_ROUTE protocol family
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.375040] thermal_sys: Registered thermal governor 'step_wise'
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.375041] thermal_sys: Registered thermal governor 'user_space'
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.378581] cpuidle: using governor ladder
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.388804] ACPI FADT declares the system doesn't support PCIe ASPM, so disable it
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.388804] acpiphp: ACPI Hot Plug PCI Controller Driver version: 0.5
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.395025] PCI: Using configuration type 1 for base access
Sun Jul 24 08:57:51 2022 kern.warn kernel: [    1.408580] ENERGY_PERF_BIAS: Set to 'normal', was 'performance'
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.416308] cryptd: max_cpu_qlen set to 1000
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.418658] ACPI: Added _OSI(Module Device)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.422846] ACPI: Added _OSI(Processor Device)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.427287] ACPI: Added _OSI(3.0 _SCP Extensions)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.428548] ACPI: Added _OSI(Processor Aggregator Device)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.433948] ACPI: Added _OSI(Linux-Dell-Video)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.438396] ACPI: Added _OSI(Linux-Lenovo-NV-HDMI-Audio)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.438547] ACPI: Added _OSI(Linux-HPI-Hybrid-Graphics)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.516029] ACPI: 10 ACPI AML tables successfully acquired and loaded
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.524608] ACPI: Dynamic OEM Table Load:
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.524608] ACPI: SSDT 0xFFFF8881001C2600 0001B8 (v02 PmRef  Cpu0Psd  00003000 INTL 20191018)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.529347] ACPI: \_SB_.PR00: _OSC native thermal LVT Acked
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.537017] ACPI: Dynamic OEM Table Load:
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.548553] ACPI: SSDT 0xFFFF888100347000 000394 (v02 PmRef  Cpu0Cst  00003001 INTL 20191018)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.558002] ACPI: Dynamic OEM Table Load:
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.558551] ACPI: SSDT 0xFFFF8881002EA800 000479 (v02 PmRef  Cpu0Ist  00003000 INTL 20191018)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.568299] ACPI: Dynamic OEM Table Load:
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.568552] ACPI: SSDT 0xFFFF8881002E1000 0008E7 (v02 PmRef  ApIst    00003000 INTL 20191018)
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.578128] ACPI: Dynamic OEM Table Load:
Sun Jul 24 08:57:51 2022 kern.info kernel: [    1.588553] ACPI: SSDT 0xFFFF8881002EB800 00048A (v02 PmRef  ApHwp    00003000 INTL 20191018)
[... 71 lines cut ...]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.078547] pci 0000:00:1e.3: reg 0x10: [mem 0x00000000-0x00000fff 64bit]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.088897] pci 0000:00:1f.0: [8086:4d87] type 00 class 0x060100
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.095198] pci 0000:00:1f.3: [8086:4dc8] type 00 class 0x040300
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.098582] pci 0000:00:1f.3: reg 0x10: [mem 0x6001120000-0x6001123fff 64bit]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.105757] pci 0000:00:1f.3: reg 0x20: [mem 0x6001000000-0x60010fffff 64bit]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.108637] pci 0000:00:1f.3: PME# supported from D3hot D3cold
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.118547] pci 0000:00:1f.4: [8086:4da3] type 00 class 0x0c0500
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.118575] pci 0000:00:1f.4: reg 0x10: [mem 0x6001126000-0x60011260ff 64bit]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.128547] pci 0000:00:1f.4: reg 0x20: [io  0xefa0-0xefbf]
[... cut ...]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.205394] pci 0000:02:00.0: [8086:1521] type 00 class 0x020000
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.208566] pci 0000:02:00.0: reg 0x10: [mem 0x7fd00000-0x7fdfffff]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.218547] pci 0000:02:00.0: reg 0x18: [io  0x3020-0x303f]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.218555] pci 0000:02:00.0: reg 0x1c: [mem 0x7fe84000-0x7fe87fff]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.228547] pci 0000:02:00.0: reg 0x30: [mem 0x7fe00000-0x7fe7ffff pref]

baodp, 2022-07-24:
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.228611] pci 0000:02:00.0: PME# supported from D0 D3hot D3cold
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.238547] pci 0000:02:00.1: [8086:1521] type 00 class 0x020000
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.248564] pci 0000:02:00.1: reg 0x10: [mem 0x7fc00000-0x7fcfffff]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.254845] pci 0000:02:00.1: reg 0x18: [io  0x3000-0x301f]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.258555] pci 0000:02:00.1: reg 0x1c: [mem 0x7fe80000-0x7fe83fff]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.268547] pci 0000:02:00.1: PME# supported from D0 D3hot D3cold
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.268602] pci 0000:00:1c.4: PCI bridge to [bus 02]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.278547] pci 0000:00:1c.4:   bridge window [io  0x3000-0x3fff]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.278549] pci 0000:00:1c.4:   bridge window [mem 0x7fc00000-0x7fefffff]
Sun Jul 24 08:57:51 2022 kern.debug kernel: [    2.288547] pci_bus 0000:00: on NUMA node 0
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.288547] ACPI: PCI: Interrupt link LNKA configured for IRQ 0
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.298638] ACPI: PCI: Interrupt link LNKB configured for IRQ 1
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.304639] ACPI: PCI: Interrupt link LNKC configured for IRQ 0
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.308636] ACPI: PCI: Interrupt link LNKD configured for IRQ 0
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.314641] ACPI: PCI: Interrupt link LNKE configured for IRQ 0
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.318635] ACPI: PCI: Interrupt link LNKF configured for IRQ 0
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.328547] ACPI: PCI: Interrupt link LNKG configured for IRQ 0
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.328637] ACPI: PCI: Interrupt link LNKH configured for IRQ 0
Sun Jul 24 08:57:51 2022 kern.notice kernel: [    2.348547] SCSI subsystem initialized
Sun Jul 24 08:57:51 2022 kern.debug kernel: [    2.348592] libata version 3.00 loaded.
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.348596] ACPI: bus type USB registered
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.352621] usbcore: registered new interface driver usbfs
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.358116] usbcore: registered new interface driver hub
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.368553] usbcore: registered new device driver usb
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.373653] Registered efivars operations
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.378576] PCI: Using ACPI for IRQ routing
Sun Jul 24 08:57:51 2022 kern.debug kernel: [    2.378576] PCI: pci_cache_line_size set to 64 bytes
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.378695] pci 0000:00:1f.5: can't claim BAR 0 [mem 0xfe010000-0xfe010fff]: no compatible bridge window
Sun Jul 24 08:57:51 2022 kern.debug kernel: [    2.388547] e820: reserve RAM buffer [mem 0x0009f000-0x0009ffff]
Sun Jul 24 08:57:51 2022 kern.debug kernel: [    2.388547] e820: reserve RAM buffer [mem 0x70f04000-0x73ffffff]
Sun Jul 24 08:57:51 2022 kern.debug kernel: [    2.388547] e820: reserve RAM buffer [mem 0x710df000-0x73ffffff]
Sun Jul 24 08:57:51 2022 kern.debug kernel: [    2.388547] e820: reserve RAM buffer [mem 0x71fb2000-0x73ffffff]
Sun Jul 24 08:57:51 2022 kern.debug kernel: [    2.388547] e820: reserve RAM buffer [mem 0x74f00000-0x77ffffff]
Sun Jul 24 08:57:51 2022 kern.debug kernel: [    2.388547] e820: reserve RAM buffer [mem 0x480400000-0x483ffffff]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.394305] clocksource: Switched to clocksource tsc-early
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.394306] pnp: PnP ACPI init
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.397378] system 00:00: [io  0x0a00-0x0a0f] has been reserved
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.403301] system 00:00: [io  0x0a10-0x0a1f] has been reserved
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.409220] system 00:00: [io  0x0a20-0x0a2f] has been reserved
Sun Jul 24 08:57:51 2022 kern.debug kernel: [    2.415747] pnp 00:03: [dma 0 disabled]
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.415894] system 00:04: [io  0x0680-0x069f] has been reserved
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.421814] system 00:04: [io  0x164e-0x164f] has been reserved
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.427840] system 00:05: [io  0x1854-0x1857] has been reserved
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.434467] system 00:06: [mem 0xfea80000-0xfeafffff] has been reserved
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.441089] system 00:06: [mem 0xfeda0000-0xfeda0fff] has been reserved
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.447701] system 00:06: [mem 0xfeda1000-0xfeda1fff] has been reserved
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.454312] system 00:06: [mem 0xc0000000-0xcfffffff] has been reserved
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.460926] system 00:06: [mem 0xfed20000-0xfed7ffff] could not be reserved
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.467883] system 00:06: [mem 0xfed90000-0xfed93fff] has been reserved
Sun Jul 24 08:57:51 2022 kern.info kernel: [    2.474495] system 00:06: [mem 0xfed45000-0xfed8ffff] could not be reserved
[... cut ...]
Sun Jul 24 10:59:54 2022 daemon.err uhttpd[2553]: sh: out of range
Sun Jul 24 10:59:59 2022 daemon.err uhttpd[2553]: sh: out of range
Sun Jul 24 11:00:04 2022 daemon.err uhttpd[2553]: sh: out of range
Sun Jul 24 11:00:09 2022 daemon.err uhttpd[2553]: sh: out of range
Sun Jul 24 11:00:14 2022 daemon.err uhttpd[2553]: sh: out of range
Sun Jul 24 11:00:19 2022 daemon.err uhttpd[2553]: sh: out of range
Sun Jul 24 11:00:24 2022 daemon.err uhttpd[2553]: sh: out of range
Sun Jul 24 11:00:29 2022 daemon.err uhttpd[2553]: sh: out of range

zhangguanzhang, 2022-07-24:
web打不开吗

baodp, 2022-07-24:
> 网络打不开吗

能打开。

WYC-2020, 2022-07-24:
这问题问的真是，哎，看日志你的先说你什么问题，你这是让别人猜吗？大雕会给你看才有鬼

closed 2022-07-25 by baodp