coolsnowwolf/lede #9829: 5.18 x86/64固件虚拟机问题

Sir-Speed-a-Lot, 2022-07-24:
x86/64 efi编译好之后用esxi, vm workstation跑起来完全正常。
但在hyper-v上开机1秒后无限重启，勉勉强强看清楚写了 Kernel panic - not syncing: Fatal exception 虚拟机硬件配置都一样。
仅5.18内核有问题，5.15正常。

重启闪的太快了，不知道如何搞到日志

shu0802, 2022-07-25:
录屏然后再慢放，不过麻烦，但可以行

> x86/64 efi编译好之后用esxi, vm workstation跑起来完全正常。 但在hyper-v上开机1秒后无限重启，勉勉强强看清楚写了 Kernel panic - not syncing: Fatal exception 虚拟机硬件配置都一样。 仅5.18内核有问题，5.15正常。
> 
> 重启闪的太快了，不知道如何搞到日志

录屏然后再慢放，不过麻烦，但可以行

WYC-2020, 2022-07-25:
刚测试没任何问题5.18，记得用第一代，用第二代是引导不起来的，这应该和uefi有关系

WYC-2020, 2022-07-25:
如果非要装下要用uefi 就把 安全启动关闭就可以了，我这边试过 都没问题，你可能的排查下自己原因了 @Sir-Speed-a-Lot

Sir-Speed-a-Lot, 2022-07-25:
> 如果非要装下要用uefi 就把 安全启动关闭就可以了，我这边试过 都没问题，你可能的排查下自己原因了 @Sir-Speed-a-Lot

谢谢回复！我也两代都试了，用第一代能勉强看见错误提示 就是上面说的fatal exception ，第二代（跑固件我都是不开secureboot的）是只有hyper v字的开机界面，然后就重启了，什么都看不见。第一代大概2秒重启，第二代1秒没到。

但是上述情况都是我是用的efi的镜像转出vhdx跑的第一代第二代，一会我试试看不efi的吧。

还有我环境是windows server 2022

WYC-2020, 2022-07-25:
这个只能你自己排查了 我这边复现不了，应该是你那边自己哪里有问题慢慢看吧

Wilson-coder, 2022-07-28:
用着5.18.13没有任何问题，唯一遇到的异常是利用编译环境直接生成的VHDX文件建立的虚拟机，后来在openwrt系统里升级新编译的固件之后，虚拟机自带硬盘从/dev/sda变成了/dev/sdb。而直通给Openwrt的一个固态硬盘变成了/dev/sda，不过磁盘管理里面仍然挂载的是/mnt/sdb，不影响各个插件的设置。

Sir-Speed-a-Lot, 2022-07-28:
> 用着5.18.13没有任何问题，唯一遇到的异常是利用编译环境直接生成的VHDX文件建立的虚拟机，后来在openwrt系统里升级新编译的固件之后，虚拟机自带硬盘从/dev/sda变成了/dev/sdb。而直通给Openwrt的一个固态硬盘变成了/dev/sda，不过磁盘管理里面仍然挂载的是/mnt/sdb，不影响各个插件的设置。

既然提到，那顺便问下大神们，编译直接生成的VHDX，和img用starwind转换器转出来的vhdx，有啥区别？文件大小我记得差很大对吧。平时我都用的转换器，从未试过直接生成的。

Sir-Speed-a-Lot, 2022-08-08:
试了试，截到了几个屏，请问哪位大神懂哪里的问题吗？

环境是Windows server 2022，第一代虚拟机，非EFI版固件，5.19内核。

依旧是5.15都能正常使用，但编译时只有勾选了test kernel就挂了，无论是5.18还是5.19，无论是EFI（EFI版没任何提示就不停重启）还是非EFI，无论是编译时候的vhdx还是用镜像传出来的vhdx都不行。

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Hellrabb, 2022-08-09:
我用esxi也这样。。

> 试了试，截到了几个屏，请问哪位大神懂哪里的问题吗？
> 
> 环境是Windows server 2022，第一代虚拟机，非EFI版固件，5.19内核。
> 
> 依旧是5.15都能正常使用，但编译时只有勾选了test kernel就挂了，无论是5.18还是5.19，无论是EFI（EFI版没任何提示就不停重启）还是非EFI，无论是编译时候的vhdx还是用镜像传出来的vhdx都不行。
> 
> [image] [image] [image] [image] [image] [image] [image] [image]

Sir-Speed-a-Lot, 2022-08-09:
> 

请问你用的什么CPU？我家两个电脑 小主机用的i5-11320H就有问题 笔记本的R5-4600H又没问题。

后来我自己是搞定了 打开了hyper-v虚拟机设置 CPU里面有个兼容性 迁移处理器的不同版本勾上就好了

所以建议你找下esxi里面cpu设置吧 挨个勾起来试试看

Sir-Speed-a-Lot, 2022-08-09:
> 我用esxi也这样。。
> 
> > 试了试，截到了几个屏，请问哪位大神懂哪里的问题吗？
> > 环境是Windows server 2022，第一代虚拟机，非EFI版固件，5.19内核。
> > 依旧是5.15都能正常使用，但编译时只有勾选了test kernel就挂了，无论是5.18还是5.19，无论是EFI（EFI版没任何提示就不停重启）还是非EFI，无论是编译时候的vhdx还是用镜像传出来的vhdx都不行。
> > [image] [image] [image] [image] [image] [image] [image] [image]

请问你用的什么CPU？我家两个电脑 小主机用的i5-11320H就有问题 笔记本的R5-4600H又没问题。

后来我自己是搞定了 打开了hyper-v虚拟机设置 CPU里面有个兼容性 迁移处理器的不同版本勾上就好了

所以建议你找下esxi里面cpu设置吧 挨个勾起来试试看

Hellrabb, 2022-08-10:
> 请问你用的什么CPU？我家两个电脑 小主机用的i5-11320H就有问题 笔记本的R5-4600H又没问题。
>
> 后来我自己是搞定了 打开了hyper-v虚拟机设置 CPU里面有个兼容性 迁移处理器的不同版本勾上就好了
>
> 所以建议你找下esxi里面cpu设置吧 挨个勾起来试试看

我用的是N5105的软路由，也是11代
所以可能是核心和11代有点兼容性问题，esxi我再看看在哪找

closed 2022-09-14 by aiamadeus