coolsnowwolf/lede #9831: 编译出错；

whfweb, 2022-07-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
以前可以正常编译的.config文件，在更新源码后云端编译不能通过，在电脑搭建环境还是出错。

### (1) 具体问题

A：编译出错；

### (2) 路由器型号和固件版本

A：wrt1900acv2

### (3) 详细日志

A：/home/whfweb/lede/staging_dir/host/bin/install -c -m 644 './libltdl/lt_error.c' '/home/whfweb/lede/staging_dir/host/share/libtool/libltdl/lt_error.c'
 /home/whfweb/lede/staging_dir/host/bin/install -c -m 644 './libltdl/ltdl.c' '/home/whfweb/lede/staging_dir/host/share/libtool/libltdl/ltdl.c'
 /home/whfweb/lede/staging_dir/host/bin/install -c -m 644 './libltdl/ltdl.h' '/home/whfweb/lede/staging_dir/host/share/libtool/libltdl/ltdl.h'
 /home/whfweb/lede/staging_dir/host/bin/install -c -m 644 './libltdl/slist.c' '/home/whfweb/lede/staging_dir/host/share/libtool/libltdl/slist.c'
chmod a+x /home/whfweb/lede/staging_dir/host/share/libtool/libltdl/configure
 /usr/bin/mkdir -p '/home/whfweb/lede/staging_dir/host/include'
 /home/whfweb/lede/staging_dir/host/bin/install -c -m 644 libltdl/ltdl.h '/home/whfweb/lede/staging_dir/host/include'
 /usr/bin/mkdir -p '/home/whfweb/lede/staging_dir/host/share/info'
 /home/whfweb/lede/staging_dir/host/bin/install -c -m 644 ./doc/libtool.info ./doc/libtool.info-1 ./doc/libtool.info-2 '/home/whfweb/lede/staging_dir/host/share/info'
 install-info --info-dir='/home/whfweb/lede/staging_dir/host/share/info' '/home/whfweb/lede/staging_dir/host/share/info/libtool.info'
 /usr/bin/mkdir -p '/home/whfweb/lede/staging_dir/host/include/libltdl'
 /home/whfweb/lede/staging_dir/host/bin/install -c -m 644 libltdl/libltdl/lt_system.h libltdl/libltdl/lt_error.h libltdl/libltdl/lt_dlloader.h '/home/whfweb/lede/staging_dir/host/include/libltdl'
 /usr/bin/mkdir -p '/home/whfweb/lede/staging_dir/host/share/man/man1'
 /home/whfweb/lede/staging_dir/host/bin/install -c -m 644 ./doc/libtool.1 ./doc/libtoolize.1 '/home/whfweb/lede/staging_dir/host/share/man/man1'
make[7]: Leaving directory '/home/whfweb/lede/build_dir/host/libtool-2.4.2'
make[6]: Leaving directory '/home/whfweb/lede/build_dir/host/libtool-2.4.2'
make[5]: Leaving directory '/home/whfweb/lede/build_dir/host/libtool-2.4.2'
make[4]: Leaving directory '/home/whfweb/lede/build_dir/host/libtool-2.4.2'
/home/whfweb/lede/staging_dir/host/bin/sed -i -e 's,\(hardcode_into_libs\)=yes,\1=no,g' /home/whfweb/lede/staging_dir/host/bin/libtool
cp -fpR /home/whfweb/lede/staging_dir/host/bin/libtool /home/whfweb/lede/staging_dir/host/bin/libtool-ucxx
/home/whfweb/lede/staging_dir/host/bin/sed -i -e 's,-lstdc++,-luClibc++,g' /home/whfweb/lede/staging_dir/host/bin/libtool-ucxx
mkdir -p /home/whfweb/lede/staging_dir/host/stamp
touch /home/whfweb/lede/build_dir/host/libtool-2.4.2/.built
touch /home/whfweb/lede/staging_dir/host/stamp/.libtool_installed
make[3]: Leaving directory '/home/whfweb/lede/tools/libtool'
time: tools/libtool/compile#51.70#42.54#372.56
make[3]: Entering directory '/home/whfweb/lede/tools/flex'
. /home/whfweb/lede/include/shell.sh; gzip -dc /home/whfweb/lede/dl/flex-2.6.4.tar.gz | tar -C /home/whfweb/lede/build_dir/host/flex-2.6.4/.. -xf -
bash: line 1: /usr/bin/gzip: cannot execute binary file: Exec format error
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[3]: *** [Makefile:37: /home/whfweb/lede/build_dir/host/flex-2.6.4/.prepared9123281e67f8bdce8a3d63fba422bf50_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory '/home/whfweb/lede/tools/flex'
time: tools/flex/compile#0.00#0.10#1.53
    ERROR: tools/flex failed to build.
make[2]: *** [tools/Makefile:162: tools/flex/compile] Error 1
make[2]: Leaving directory '/home/whfweb/lede'
make[1]: *** [tools/Makefile:158: /home/whfweb/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyynynyyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/whfweb/lede'
make: *** [/home/whfweb/lede/include/toplevel.mk:230: world] Error 2

ecrasy, 2022-07-25:
解决办法是重新make menuconfig

WYC-2020, 2022-07-25:
删除flex-2.6.4.tar.gz 这个压缩包重新下载

closed 2022-07-25 by aiamadeus