coolsnowwolf/lede #9832: eth0网口无法使用，插线灯不亮

xflan, 2022-07-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：升级后，eth0网口无法使用，插线灯不亮 退回老版本后正常

### (2) 路由器型号和固件版本

A：
型号 openwrt-ramips-mt7621-ubnt_edgerouter-x-sfp
固件版本 commit d43bee457ce990bf2dd14ee627079637f6f4b460

WYC-2020, 2022-07-25:
回退老版本 也不说是那个老版本

xflan, 2022-07-25:
老版本 OpenWrt R22.4.1 / LuCI Master (git-22.095.24346-72703c8)

xflan, 2022-07-25:
> 

OpenWrt R22.4.1 / LuCI Master (git-22.095.24346-72703c8)

xflan, 2022-07-25:
> 回退老版本 也不说是那个老版本

老版本 OpenWrt R22.4.1 / LuCI Master (git-22.095.24346-72703c8)
内核版本 5.4.188

XDYas, 2022-07-30:
Ubiquiti EdgeRouter X 也有同样的问题，整个7月编译的都是一样，包括7月30日编译的版本，退回Commit f6f103b1cb4bfb09ff8d690618b0fe9132e07f96 版本正常使用。
日志如下：
Sat Jul 16 00:37:49 2022 kern.notice kernel: [    0.000000] Linux version 5.4.203 (xdyas@debian) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4667-14e109d0b)) #0 SMP Fri Jul 15 02:05:30 2022
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] MIPS: machine is Ubiquiti EdgeRouter X
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Sat Jul 16 00:37:49 2022 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sat Jul 16 00:37:49 2022 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sat Jul 16 00:37:49 2022 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] Zone ranges:
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000]   HighMem  empty
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] Movable zone start for each node
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] Early memory node ranges
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Sat Jul 16 00:37:49 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Sat Jul 16 00:37:49 2022 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Sat Jul 16 00:37:49 2022 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sat Jul 16 00:37:49 2022 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s27152 r8192 d22000 u57344
Sat Jul 16 00:37:49 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s27152 r8192 d22000 u57344 alloc=14*4096
Sat Jul 16 00:37:49 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
Sat Jul 16 00:37:49 2022 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] Writing ErrCtl register=00045e00
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] Readback ErrCtl register=00045e00
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] Memory: 250088K/262144K available (6165K kernel code, 273K rwdata, 1296K rodata, 1276K init, 235K bss, 12056K reserved, 0K cma-reserved, 0K highmem)
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] NR_IRQS: 256
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.015564] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.087890] pid_max: default: 32768 minimum: 301
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.097224] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.111630] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.130031] rcu: Hierarchical SRCU implementation.
Sat Jul 16 00:37:49 2022 kern.info kernel: [    0.140281] smp: Bringing up secondary CPUs ...
Sat Jul 16 00:37:49 2022 kern.warn kernel: [    6.775818] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sat Jul 16 00:37:49 2022 kern.warn kernel: [    6.775828] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[... cut ...]
Sat Jul 16 00:39:59 2022 daemon.notice netifd: wan (5761): sh: out of range
Sat Jul 16 00:39:59 2022 daemon.info pppd[5796]: Plugin pppoe.so loaded.
Sat Jul 16 00:39:59 2022 daemon.info pppd[5796]: PPPoE plugin from pppd 2.4.9
Sat Jul 16 00:39:59 2022 daemon.notice pppd[5796]: pppd 2.4.9 started by root, uid 0
Sat Jul 16 00:40:00 2022 daemon.warn odhcpd[2220]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Sat Jul 16 00:40:14 2022 daemon.warn pppd[5796]: Timeout waiting for PADO packets

lwb1978, 2022-11-26:
我今天也遇到了，新编译的er-x，内核版本5.4.224也是eth0网口不亮灯，没想到这问题7月至今都未修复，郁闷。

closed 2026-09-05 by coolsnowwolf