coolsnowwolf/lede #9833: amd 200ge组的openwrt 想docker一个jellyfin  vega3集成的显卡驱动怎么编译

achinese888, 2022-07-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：amd 200ge组的openwrt 想docker一个jellyfin  vega3集成的显卡驱动怎么编译 make kernel_menuconfig勾选了amdgpu但是编译出来没有dir目录，没有驱动  启动有报错

### (2) 路由器型号和固件版本

A：x86 amd200ge 集成vega3

### (3) 详细日志

A：报错是这一段，好像没有正确加载
[    2.762066] [drm] amdgpu kernel modesetting enabled.
[    2.767253] [drm] initializing kernel modesetting (RAVEN 0x1002:0x15DD 0x1565:0x170B 0xCB).
[    2.775610] amdgpu 0000:09:00.0: amdgpu: Trusted Memory Zone (TMZ) feature enabled
[    2.783189] [drm] register mmio base: 0xFCC00000
[    2.787810] [drm] register mmio size: 524288
[    2.792110] [drm] add ip block number 0 <soc15_common>
[    2.797258] [drm] add ip block number 1 <gmc_v9_0>
[    2.802055] [drm] add ip block number 2 <vega10_ih>
[    2.806937] [drm] add ip block number 3 <psp>
[    2.811298] [drm] add ip block number 4 <gfx_v9_0>
[    2.816098] [drm] add ip block number 5 <sdma_v4_0>
[    2.820979] [drm] add ip block number 6 <powerplay>
[    2.825863] [drm] add ip block number 7 <dm>
[    2.830140] [drm] add ip block number 8 <vcn_v1_0>
[    2.834983] amdgpu 0000:09:00.0: Direct firmware load for amdgpu/raven_gpu_info.bin failed with error -2
[    2.844469] amdgpu 0000:09:00.0: Falling back to sysfs fallback for: amdgpu/raven_gpu_info.bin
[    3.637958] tsc: Refined TSC clocksource calibration: 3193.998 MHz
[    3.644159] clocksource: tsc: mask: 0xffffffffffffffff max_cycles: 0x2e0a244aeba, max_idle_ns: 440795290469 ns
[    3.654192] clocksource: Switched to clocksource tsc
[   63.473984] amdgpu 0000:09:00.0: amdgpu: Failed to load gpu_info firmware "amdgpu/raven_gpu_info.bin"
[   63.483217] amdgpu 0000:09:00.0: amdgpu: Fatal error during GPU init
[   63.489577] amdgpu 0000:09:00.0: amdgpu: amdgpu: finishing device.
[   63.495763] x86/PAT: swapper/0:1 freeing invalid memtype [mem 0x00000000-0xffffffffffffffff]
[   63.504233] amdgpu: probe of 0000:09:00.0 failed with error -110

Hyy2001X, 2022-07-25:
i915 也有相同的 `failed with error -2` 报错，内核 5.18 全新编译

已经在 [config-5.18](https://github.com/coolsnowwolf/lede/blob/master/target/linux/x86/config-5.18) 中追加了
> CONFIG_DRM=y
CONFIG_DRM_I915=y

以下是日志
`root@Op:~# dmesg | grep i915`
**[    0.583082] i915 0000:00:02.0: Direct firmware load for i915/icl_dmc_ver1_09.bin failed with error -2**
**[    0.584256] i915 0000:00:02.0: Falling back to sysfs fallback for: i915/icl_dmc_ver1_09.bin**
[    0.587622] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    0.667837] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    0.833905] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    1.003974] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
**[   64.705141] i915 0000:00:02.0: [drm] Failed to load DMC firmware i915/icl_dmc_ver1_09.bin. Disabling runtime power management.**
[   64.706338] i915 0000:00:02.0: [drm] DMC firmware homepage: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915

`root@Op:~# ls /lib/firmware/i915/` 下有完整的 igpu 驱动
`root@Op:~# ls /dev/dri` >  `card0       renderD128`

achinese888, 2022-07-26:
> i915 也有相同的 `failed with error -2` 报错，内核 5.18 全新编译
>
> 已经在 [config-5.18](https://github.com/coolsnowwolf/lede/blob/master/target/linux/x86/config-5.18) 中追加了
>
>
> 以下是日志 `root@Op:~# dmesg | grep i915` **[ [...]

也就是新版本问题解决了吗，我make kernel——menuconfig选了没用，没有你说的这个目录 /dev/dri

achinese888, 2022-07-29:
发现个问题，我这个核显是radeon，我勾选了这个后不报错但是还是没有dev/dri这个目录，难搞

achinese888, 2022-07-29:
> 发现个问题，我这个核显是radeon，我勾选了这个后不报错但是还是没有dev/dri这个目录，难搞

整个日志就这一句有关
[drm] radeon kernel modesetting enabled.
有没有大佬来帮帮忙，国外openwrt论坛看不懂

closed 2022-08-01 by achinese888