coolsnowwolf/lede #9838: 编译大家没出问题吗？

Jejz168, 2022-07-26:
ERROR: package/feeds/packages/ruby [host] failed to build.

太安静了。都没这个报错吗？😂😂

wx1wxc, 2022-07-26:
> make[5]: Entering directory '/workdir/openwrt/build_dir/hostpkg/ruby-3.1.2'
make[6]: Entering directory '/workdir/openwrt/build_dir/hostpkg/ruby-3.1.2/ext/date'
make[6]: Leaving directory '/workdir/openwrt/build_dir/hostpkg/ruby-3.1.2/ext/date'
make[6]: Entering directory '/workdir/openwrt/build_dir/hostpkg/ruby-3.1.2/ext/digest'
make[6]: Leaving directory '/workdir/openwrt/build_dir/hostpkg/ruby-3.1.2/ext/digest'
make[6]: Entering directory '/workdir/openwrt/build_dir/hostpkg/ruby-3.1.2/ext/digest/sha2'
make[6]: Leaving directory '/workdir/openwrt/build_dir/hostpkg/ruby-3.1.2/ext/digest/sha2'
make[6]: Entering directory '/workdir/openwrt/build_dir/hostpkg/ruby-3.1.2/ext/monitor'
make[6]: Leaving directory '/workdir/openwrt/build_dir/hostpkg/ruby-3.1.2/ext/monitor'
make[6]: Entering directory '/workdir/openwrt/build_dir/hostpkg/ruby-3.1.2/ext/openssl'
compiling ossl_pkey.c
ossl_pkey.c: In function 'ossl_pkey_export_traditional':
ossl_pkey.c:681:62: error: dereferencing pointer to incomplete type 'EVP_PKEY' {aka 'struct evp_pkey_st'}
  681 |  EVP_PKEY_asn1_get0_info(NULL, NULL, NULL, NULL, &aname, pkey->ameth);
      |                                                              ^~
make[6]: *** [Makefile:323: ossl_pkey.o] Error 1
make[6]: Leaving directory '/workdir/openwrt/build_dir/hostpkg/ruby-3.1.2/ext/openssl'
make[5]: *** [exts.mk:115: ext/openssl/all] Error 2
make[4]: *** [uncommon.mk:305: build-ext] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/hostpkg/ruby-3.1.2'
make[3]: *** [Makefile:1156: /workdir/openwrt/build_dir/hostpkg/ruby-3.1.2/.built] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/hostpkg/ruby-3.1.2'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/ruby'
time: package/feeds/packages/ruby/host-compile#1.15#0.26#1.37
    ERROR: package/feeds/packages/ruby [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/ruby/host/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

云编译 一样有报错

Jejz168, 2022-07-26:
> 云编译 一样有报错

那我就放心了。真安静。我以为都不报错

zgc, 2022-07-26:
同样的问题

zhangguanzhang, 2022-07-26:
同样的问题，我还以为大家都没问题 🤣

Jejz168, 2022-07-26:
> 同样的问题，我还以为大家都没问题 🤣

哈哈哈。我以为就我一个。都没人说话。太安静了

hcym, 2022-07-26:
master分支的问题，用2203没事，大雕不幸就是master分支的

ecrasy, 2022-07-26:
昨天下午两点编译的X86和树莓派4b
都没问题成功的
L大的packages除了cjdns全部编译打包
额外加了passwall和Helloworld

zhangguanzhang, 2022-07-26:
> 昨天下午两点编译的X86和树莓派4b 都没问题成功的 L大的packages除了cjdns全部编译打包 额外加了passwall和Helloworld

大佬，我的编译步骤和你的差不多，不过我开了很多app编译，我把整个目录都缓存了，然后第二次action在 `make package/compile` 还是需要编译很久怎么破

hcym, 2022-07-26:
ruby，编译clash的依赖，我是master分支遇过，立马取消clash及相关依赖，稍有遗憾

Jejz168, 2022-07-26:
> ruby，编译clash的依赖，我是master分支遇过，立马取消clash及相关依赖，稍有遗憾

是的。好像就是clash的问题

Jejz168, 2022-07-26:
> ruby，编译clash的依赖，我是master分支遇过，立马取消clash及相关依赖，稍有遗憾

已更新了。快冲。开始编译。没clash。没灵魂

Marcus0605, 2022-07-26:
害！拼命重装ubuntu，找云编译，都报错，耗了我两天的时间。。。我一直以为是我自己的问题！！

hcym, 2022-07-26:
2103和2203都编译了，master分支没有，就是和大雕库一样没有clash而已，其他应有尽有😙🤣😂

Jejz168, 2022-07-26:
> 害！拼命重装ubuntu，找云编译，都报错，耗了我两天的时间。。。我一直以为是我自己的问题！！ 那现在兄弟们都编译成功了吗？

哈哈哈。我看都没人说话。原来大家都这样。我以为就我一人是这样

WYC-2020, 2022-07-26:
就没出过问题

Jejz168, 2022-07-26:
> 就没出过问题

你昨天。今天没编译。就不要说没出问题了

Jejz168, 2022-07-26:
> 2103和2203都编译了，master分支没有，就是和大雕库一样没有clash而已，其他应有尽有😙🤣😂

可以了。能编译过了。已解决

WYC-2020, 2022-07-27:
> > 就没出过问题
> 
> 你昨天。今天没编译。就不要说没出问题了

我说你咋不信呢，说没问题你还不信，这玩意最近几天都没更新，谁告诉你 我昨天没有编译的，你怕是自己又去修改了这个库的默认选择吧，如果只是编译被openclash依赖 是完全没问题的 我今天还试了一下，一样没有打你们那个补丁，完全没问题，并且ossl_pkey.c一样编译出来，不知道你们是怎么遇到的
[image]
[image]

WYC-2020, 2022-07-27:
> > 就没出过问题
> 
> 你昨天。今天没编译。就不要说没出问题了

刚有试了下吧ruby 下面所有的库都选择，编译 也没遇到你们说的这个，你可以吧你的.config 发给我 我这边编译看下，反正我是打死都遇不到，哈哈

Jejz168, 2022-07-27:
唉。已经修复了。你现在编译还能有什么问题哦

WYC-2020, 2022-07-27:
> 唉。已经修复了。你现在编译还能有什么问题哦

你真是看文字不认真，我说了我没有那个patch，就是你们所谓的修复那个patch

hcym, 2022-07-27:
OK了，😙😅🤣🤗