coolsnowwolf/lede #9839: libpcre.so.1: cannot open shared object file: No such file or directory

Shaco-Ma, 2022-07-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
最新代码,使用默认menuconfig,编译X86出错

### (1) 具体问题

A：install: error while loading shared libraries: libpcre.so.1: cannot open shared object file: No such file or directory

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include  -I../include -D_GNU_SOURCE  -I/home/shaco/My_data/lede/staging_dir/target-x86_64_musl/usr/include   -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro  -L../src  selabel_lookup.c  -lselinux -lfts -o selabel_lookup
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include  -I../include -D_GNU_SOURCE  -I/home/shaco/My_data/lede/staging_dir/target-x86_64_musl/usr/include   -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro  -L../src  selabel_lookup_best_match.c  -lselinux -lfts -o selabel_lookup_best_match
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include  -I../include -D_GNU_SOURCE  -I/home/shaco/My_data/lede/staging_dir/target-x86_64_musl/usr/include   -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro  -L../src  selabel_partial_match.c  -lselinux -lfts -o selabel_partial_match
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include  -I../include -D_GNU_SOURCE  -I/home/shaco/My_data/lede/staging_dir/target-x86_64_musl/usr/include   -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro  -L../src  selinux_check_access.c  -lselinux -lfts -o selinux_check_access
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include  -I../include -D_GNU_SOURCE  -I/home/shaco/My_data/lede/staging_dir/target-x86_64_musl/usr/include   -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro  -L../src  selinux_check_securetty_context.c  -lselinux -lfts -o selinux_check_securetty_context
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include  -I../include -D_GNU_SOURCE  -I/home/shaco/My_data/lede/staging_dir/target-x86_64_musl/usr/include   -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro  -L../src  selinuxenabled.c  -lselinux -lfts -o selinuxenabled
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include  -I../include -D_GNU_SOURCE  -I/home/shaco/My_data/lede/staging_dir/target-x86_64_musl/usr/include   -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro  -L../src  selinuxexeccon.c  -lselinux -lfts -o selinuxexeccon
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include  -I../include -D_GNU_SOURCE  -I/home/shaco/My_data/lede/staging_dir/target-x86_64_musl/usr/include   -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro  -L../src  setenforce.c  -lselinux -lfts -o setenforce
x86_64-openwrt-linux-musl-gcc -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/shaco/My_data/lede/build_dir/target-x86_64_musl/libselinux-3.2=libselinux-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include  -I../include -D_GNU_SOURCE  -I/home/shaco/My_data/lede/staging_dir/target-x86_64_musl/usr/include   -L/home/shaco/My_data/lede/staging_dir/toolchain-x86_64_gcc-8. [... cut ...]
make[3]: Leaving directory '/home/shaco/My_data/lede/package/libs/libselinux'
time: package/libs/libselinux/compile#11.14#2.14#13.74
    ERROR: package/libs/libselinux failed to build.
make[2]: *** [package/Makefile:116: package/libs/libselinux/compile] Error 1
make[2]: Leaving directory '/home/shaco/My_data/lede'
make[1]: *** [package/Makefile:110: /home/shaco/My_data/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/shaco/My_data/lede'
make: *** [/home/shaco/My_data/lede/include/toplevel.mk:230：world] 错误 2

Shaco-Ma, 2022-07-27:
你们都没出现这种问题吗?

WYC-2020, 2022-07-27:
> 你们都没出现这种问题吗?

clean 在compile 这个我看了makefile 是有依赖这个库的，没问题的，我自己也试过没问题

Shaco-Ma, 2022-07-27:
> 

但是这个库我看了代码目录里面也有,但是找不到啊,你compile没有问题吗?

closed 2022-08-15 by aiamadeus

naplesblue, 2023-05-30:
我遇到相同问题，已经解决。在另一个issue中做了回答。因为当初搜索的时候，也搜到过这里，所以，也贴一下，供后面的人参考。

https://github.com/coolsnowwolf/lede/issues/10762#issue-1531597009