coolsnowwolf/lede #9840: 最近升级了luci-theme-argon后怎么每次刷固件变成默认的bootstrap了？

GreatMichaelLee, 2022-07-27:
之前都是每次刷完还是argon主题，最近好像仓库动了一下argon theme, 之后每次刷完就变成默认的bootstrap了，要自己手工在luci里改回配置为argon，请问哪里能persistent为argon?

Jejz168, 2022-07-27:
可以阿。再要么不编译bootstrap

GreatMichaelLee, 2022-07-27:
> 可以阿。再要么不编译bootstrap

那个是default, make menuconfig里不能选

Jejz168, 2022-07-27:
> > 可以阿。再要么不编译bootstrap
> 
> 那个是default, make menuconfig里不能选

sed -i '/set luci.main.mediaurlbase=\/luci-static\/bootstrap/d' feeds/luci/themes/luci-theme-bootstrap/root/etc/uci-defaults/30_luci-theme-bootstrap
sed -i 's/luci-theme-bootstrap/luci-theme-argon/g' ./feeds/luci/collections/luci/Makefile
sed -i 's/luci-theme-bootstrap/luci-theme-argon/g' feeds/luci/collections/luci-nginx/Makefile

Jejz168, 2022-07-27:
再试试。多看看别人的。都是经验

lovehackintosh, 2022-07-27:
在make menuconfig前执行:
sed -i 's/luci-theme-bootstrap/luci-theme-argon/g' feeds/luci/collections/luci/Makefile
然后正常编译

zhangguanzhang, 2022-07-27:
```
    # 天灵的非18.06 分支，openwrt 的 21.02 22.03 必须 luci-theme-argon 这种 21 分支的主题
    # argonne 是 18.06 分支使用
    if echo $repo_branch | grep -Pq '18\.0';then
        sed -ri 's/luci-theme-\S+/luci-theme-argonne/g' feeds/luci/collections/luci/Makefile
        sed -i 's/argon=y/argonne=y/' .config
        find -type d -regex '.*/luci-[a-z]+-argon' -exec rm -rf {} \;
    else
        # 
        sed -ri 's/luci-theme-\S+/luci-theme-argon/g' feeds/luci/collections/luci/Makefile
        sed -i 's/argonne=y/argon=y/' .config
        # 这个不兼容 openwrt 
        find -type d -name 'luci-*-argonne*' -exec rm -rf {} \;
    fi
```

qyh214, 2022-07-28:
sed -i '/uci commit luci/i\uci set luci.main.mediaurlbase="/luci-static/argon"' `find package -type f -path '*/default-settings/files/zzz-default-settings'

GreatMichaelLee, 2022-07-29:
> ```
>     # 天灵的非18.06 分支，openwrt 的 21.02 22.03 必须 luci-theme-argon 这种 21 分支的主题
>     # argonne 是 18.06 分支使用
>     if echo $repo_branch | grep -Pq '18\.0';then
>         sed -ri [...]

看不懂，啥意思，怎么应用，能否推到仓库？

Jejz168, 2022-07-29:
> > > ```
> > >     # 天灵的非18.06 分支，openwrt 的 21.02 22.03 必须 luci-theme-argon 这种 21 分支的主题
> > >     # argonne 是 18.06 分支使用
> > >     if echo $repo_branch | grep -Pq '18\.0';then
> > >         sed -ri 's/luci-theme-\S+/luci-theme-argonne/g' feeds/luci/collections/luci/Makefile
> > >         sed -i 's/argon=y/argonne=y/' .config
> > >         find -type d -regex '.*/luci-[a-z]+-argon' -exec rm -rf {} \;
> > >     else
> > >         # 
> > >         sed -ri 's/luci-theme-\S+/luci-theme-argon/g' feeds/luci/collections/luci/Makefile
> > >         sed -i 's/argonne=y/argon=y/' .config
> > >         # 这个不兼容 openwrt 
> > >         find -type d -name 'luci-*-argonne*' -exec rm -rf {} \;
> > >     fi
> > > ```
> > 
> > 
> > 看不懂，啥意思，怎么应用，能否推到仓库？
> 
> 
> 如果你是自己编译就修改这个luci的makefile，如果你是用的别人编译的固件去问问他

说复杂了。让他们完全看不懂了😂

GreatMichaelLee, 2022-07-30:
按以上各位说的方法，然而并没有卵用，编译完（clean编译的）刷出来还是默认bootstrap.

Jejz168, 2022-07-30:
> 按以上各位说的方法，然而并没有卵用，编译完（clean编译的）刷出来还是默认bootstrap.

更简单方法。手动改config文件也不行？把bootstrap取消掉。

zhangguanzhang, 2022-07-30:
> 按以上各位说的方法，然而并没有卵用，编译完（clean编译的）刷出来还是默认bootstrap.
```
grep luci-theme feeds/luci/collections/luci/Makefile
```
查看下主题

GreatMichaelLee, 2022-07-30:
> > 按以上各位说的方法，然而并没有卵用，编译完（clean编译的）刷出来还是默认bootstrap.
> 
> ```
> grep luci-theme feeds/luci/collections/luci/Makefile
> ```
> 
> 查看下主题

builder@Build-Server:~$ grep luci-theme feeds/luci/collections/luci/Makefile
        +uhttpd +uhttpd-mod-ubus +luci-mod-admin-full +luci-theme-argon \

harleywong, 2022-08-02:
我也有这问题，不知道为什么，每次更新固件，主题都从argon变成Material。

qyh214, 2022-08-02:
sed -i '/uci commit luci/i\uci set luci.main.mediaurlbase="/luci-static/argon"' `find package -type f -path '*/default-settings/files/zzz-default-settings'`

GreatMichaelLee, 2022-08-06:
> sed -i '/uci commit luci/i\uci set luci.main.mediaurlbase="/luci-static/argon"' `find package -type f -path '*/default-settings/files/zzz-default-settings'`

感谢，可以了。

closed 2022-08-15 by aiamadeus