coolsnowwolf/lede #9842: 编译时qbittorrent报错

MikuMaxZ, 2022-07-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：qbittorrent编译时候报错

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：Collected errors:
 * resolve_conffiles: Existing conffile /home/maxz/Desktop/lede/build_dir/target-x86_64_musl/root-x86/etc/config/socat is different from the conffile in the new package. The new conffile will be placed at /home/maxz/Desktop/lede/build_dir/target-x86_64_musl/root-x86/etc/config/socat-opkg.
 * check_data_file_clashes: Package qBittorrent-static wants to install file /home/maxz/Desktop/lede/build_dir/target-x86_64_musl/root-x86/usr/bin/qbittorrent-nox
	But that file is already provided by package  * qbittorrent
 * opkg_install_cmd: Cannot install package qBittorrent-static.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/maxz/Desktop/lede'
make[1]: *** [package/Makefile:111: /home/maxz/Desktop/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/maxz/Desktop/lede'
make: *** [/home/maxz/Desktop/lede/include/toplevel.mk:230：world] 错误 2

MikuMaxZ, 2022-07-27:
socat这个我已经通过更新解决了

ricksuzade-maker, 2022-07-27:
使用动态编译即可

zhangguanzhang, 2022-07-27:
`qBittorrent-static` 和 `qbittorrent_dynamic` 只能选一个

MikuMaxZ, 2022-07-28:
现在有两个qbittorrent一个luci-app-qbittorren一个luci-app-qbittorren-simple，这两个都是选的静态编译，而且选了一个还是报错

xiangfeidexiaohuo, 2022-07-28:
看报错 说是想安装qBittorrent-static，但是已经安装了qBittorrent。多选了吧，luci-app-qbittorren和luci-app-qbittorren-simple 二选一，然后qBittorrent-static 和 qbittorrent_dynamic二选一。不会选，直接改config啊，保证都只一个就成。

MikuMaxZ, 2022-07-29:
>

closed 2022-07-29 by MikuMaxZ