coolsnowwolf/lede #9847: 更新最新源码后编译出错

mitsukileung, 2022-07-28:
k2p 

Applying /workdir/openwrt/target/linux/ramips/patches-5.4/0300-mtd-rawnand-add-driver-support-for-MT7621-nand-flash.patch using plaintext: 
patching file drivers/mtd/nand/raw/Kconfig
patching file drivers/mtd/nand/raw/Makefile
The next patch would create the file drivers/mtd/nand/raw/mt7621_nand.c,
which already exists!  Applying it anyway.
patching file drivers/mtd/nand/raw/mt7621_nand.c
Hunk #1 FAILED at 1 (different line endings).
1 out of 1 hunk FAILED -- saving rejects to file drivers/mtd/nand/raw/mt7621_nand.c.rej
Patch failed!  Please fix /workdir/openwrt/target/linux/ramips/patches-5.4/0300-mtd-rawnand-add-driver-support-for-MT7621-nand-flash.patch!
make[4]: *** [Makefile:33: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.203/.prepared_df8caaba439c622fdf918d4074a8ce7c] Error 1
make[4]: Leaving directory '/workdir/openwrt/target/linux/ramips'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#12.63#6.77#15.60
    ERROR: target/linux failed to build.
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[1]: *** [target/Makefile:18: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

lovehackintosh, 2022-07-28:
不好意思,我昨天提交的commit忘了修改5.4内核的patch,我会开个PR修复
你去把/workdir/openwrt/target/linux/ramips/patches-5.4/0300-mtd-rawnand-add-driver-support-for-MT7621-nand-flash.patch 47行以后的内容全部删掉，然后重新make即可

mitsukileung, 2022-07-28:
> 

好的，我去试试，谢谢

closed 2022-07-28 by aiamadeus

pipihu, 2022-07-28:
```
make[5]: Leaving directory '/home/op/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.203'
make -C /home/op/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.203 KCFLAGS="-fmacro-prefix-map=/home/op/lede/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/op/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Jul 28 02:05:40 2022" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/op/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.203 CC="mipsel-openwrt-linux-musl-gcc" vmlinux vmlinuz modules
make[5]: Entering directory '/home/op/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.203'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  CC      drivers/mtd/nand/raw/mt7621_nand.o
drivers/mtd/nand/raw/mt7621_nand.c:690:24: warning: 'struct nand_interface_config' declared inside parameter list will not be visible outside of this definition or declaration
           const struct nand_interface_config *conf)
                        ^~~~~~~~~~~~~~~~~~~~~
drivers/mtd/nand/raw/mt7621_nand.c: In function 'mt7621_nfc_setup_interface':
drivers/mtd/nand/raw/mt7621_nand.c:699:33: error: passing argument 1 of 'nand_get_sdr_timings' from incompatible pointer type [-Werror=incompatible-pointer-types]
  timings = nand_get_sdr_timings(conf);
                                 ^~~~
In file included from drivers/mtd/nand/raw/mt7621_nand.c:19:
./include/linux/mtd/rawnand.h:509:56: note: expected 'const struct nand_data_interface *' but argument is of type 'const struct nand_interface_config *'
 nand_get_sdr_timings(const struct nand_data_interface *conf)
                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
drivers/mtd/nand/raw/mt7621_nand.c: In function 'mt7621_nfc_ecc_init':
drivers/mtd/nand/raw/mt7621_nand.c:838:15: error: 'struct nand_ecc_ctrl' has no member named 'engine_type'
  if (nand->ecc.engine_type != NAND_ECC_ENGINE_TYPE_ON_HOST) {
               ^
drivers/mtd/nand/raw/mt7621_nand.c:838:31: error: 'NAND_ECC_ENGINE_TYPE_ON_HOST' undeclared (first use in this function); did you mean 'ECC_ENGINE_TIMEOUT'?
  if (nand->ecc.engine_type != NAND_ECC_ENGINE_TYPE_ON_HOST) {
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                               ECC_ENGINE_TIMEOUT
drivers/mtd/nand/raw/mt7621_nand.c:838:31: note: each undeclared identifier is reported only once for each function it appears in
drivers/mtd/nand/raw/mt7621_nand.c: At top level:
drivers/mtd/nand/raw/mt7621_nand.c:927:3: error: 'const struct nand_controller_ops' has no member named 'setup_interface'; did you mean 'setup_data_interface'?
  .setup_interface = mt7621_nfc_setup_interface,
   ^~~~~~~~~~~~~~~
   setup_data_interface
drivers/mtd/nand/raw/mt7621_nand.c:927:21: error: initialization of 'int (*)(struct nand_chip *, int,  const struct nand_data_interface *)' from incompatible pointer type 'int (*)(struct nand_chip *, int,  const struct nand_interface_config *)' [-Werror=incompatible-pointer-types]
  .setup_interface = mt7621_nfc_setup_interface,
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/mtd/nand/raw/mt7621_nand.c:927:21: note: (near initialization for 'mt7621_nfc_controller_ops.setup_data_interface')
drivers/mtd/nand/raw/mt7621_nand.c: In function 'mt7621_nfc_init_chip':
drivers/mtd/nand/raw/mt7621_nand.c:1232:19: error: 'NAND_USES_DMA' undeclared (first use in this function); did you mean 'NAND_OP_CMD'?
  nand->options |= NAND_USES_DMA | NAND_NO_SUBPAGE_WRITE | NAND_SKIP_BBTSCAN;
                   ^~~~~~~~~~~~~
                   NAND_OP_CMD
drivers/mtd/nand/raw/mt7621_nand.c:1236:11: error: 'struct nand_ecc_ctrl' has no member named 'engine_type'
  nand->ecc.engine_type = NAND_ECC_ENGINE_TYPE_ON_HOST;
           ^
drivers/mtd/nand/raw/mt7621_nand.c:1236:26: error: 'NAND_ECC_ENGINE_TYPE_ON_HOST' undeclared (first use in this function); did you mean 'ECC_ENGINE_TIMEOUT'?
  nand->ecc.engine_type = NAND_ECC_ENGINE_TYPE_ON_HOST;
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                          ECC_ENGINE_TIMEOUT
cc1: some warnings being treated as errors
make[9]: *** [scripts/Makefile.build:262: drivers/mtd/nand/raw/mt7621_nand.o] Error 1
make[8]: *** [scripts/Makefile.build:497: drivers/mtd/nand/raw] Error 2
make[7]: *** [scripts/Makefile.build:497: drivers/mtd/nand] Error 2
make[6]: *** [scripts/Makefile.build:497: drivers/mtd] Error 2
make[5]: *** [Makefile:1734: drivers] Error 2
make[5]: Leaving directory '/home/op/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.203'
make[4]: *** [Makefile:25: /home/op/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.203/.modules] Error 2
make[4]: Leaving directory '/home/op/lede/target/linux/ramips'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/op/lede/target/linux'
time: target/linux/compile#2.89#2.32#4.81
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/home/op/lede'
make[1]: *** [target/Makefile:18: /home/op/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/op/lede'
make: *** [/home/op/lede/include/toplevel.mk:230: world] Error 2
op@debian-openwrt:~/lede$ 

```
刚刚重新拉取源码编译，问题依旧。