coolsnowwolf/lede #9854: 建议：能否将vnstat的流量记录数据库db文件保存在重启不丢失的分区内，现在是一重启，数据就没有了

modelsun, 2022-07-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：建议：能否将VNStat的数据库db文件保存在重启不丢失的分区内，现在是一重启，数据就没有了

### (2) 路由器型号和固件版本

A：hiwifi HC5962  mt7621

### (3) 详细日志

A：

shiytian, 2022-07-29:
更改 db 位置即可
 vim /etc/vnstat.conf 

# location of the database directory
#DatabaseDir "/var/lib/vnstat"
DatabaseDir "/root/db/vnstat"

shiytian, 2022-07-29:
另外，建议添加 vnstat2，当前 vnstati 为 1.x 版本，无法自定义graph scale, 导致图表太小无法自定义调整。

modelsun, 2022-07-29:
> 更改 db 位置即可 vim /etc/vnstat.conf
> 
> # location of the database directory
> #DatabaseDir "/var/lib/vnstat" DatabaseDir "/root/db/vnstat"

modelsun, 2022-08-07:
> 更改 db 位置即可 vim /etc/vnstat.conf
> 
> # location of the database directory
> #DatabaseDir "/var/lib/vnstat" DatabaseDir "/root/db/vnstat"

这是编译后的吧，编译前怎么改？

Plutonium141, 2022-08-15:
luci-app-vnstat/luasrc/model/cbi/vnstat.lua
这个文件的第16行你可以在编译前改一下看看

modelsun, 2022-08-15:
> luci-app-vnstat/luasrc/model/cbi/vnstat.lua 这个文件的第16行你可以在编译前改一下看看

用的 > luci-app-vnstat2，改了 luasrc/model/cbi/vnstat.lua 中的 dbdir ，不管用，还需要同步改vnstat.conf中的路径才行（编译后改的），但编译前不知道怎么改 【捂脸】

closed 2024-05-28 by aiamadeus