coolsnowwolf/lede #9862: 6in4隧道无法正常重启

lzy0702, 2022-07-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：在wan6配置了He.net的动态6in4隧道之后，虚拟机启动时能正确连接上隧道，但如果需要重启隧道，点击页面上“连接”按钮后无法正常重启隧道。系统内置的配置隧道的sh脚本会持续运行且无法杀掉。必须重新启动虚拟机才能正常连接上隧道。防火墙规则是默认的。

### (2) 路由器型号和固件版本

A：x86_64 于 Win Server 2022的Hyper-V下，无SR-IOV. 源码于北京时间2022-07-29 中午12时许签出。内核版本5.15.57

### (3) 详细日志

A：系统内配置隧道的sh脚本不在系统日志和内核日志中输出任何信息

lzy0702, 2022-07-30:
点击wan6的“连接”按钮后的内核日志（点击前系统已启动约21分钟）
`[  785.270232] hv_balloon: Balloon request will be partially fulfilled. Balloon floor reached.
[ 1085.262403] hv_balloon: Balloon request will be partially fulfilled. Balloon floor reached.`
系统日志（时间为美国东部时间（UTC-4））
`Fri Jul 29 21:44:27 2022 daemon.info dhcpcd[4190]: eth0: REPLY6 received from fe80::6664:9bff:feb0:7fc5
Fri Jul 29 21:44:27 2022 daemon.info dhcpcd[4190]: eth0: will expire
Fri Jul 29 21:44:27 2022 daemon.err dhcpcd[4190]: eth0: DHCPv6 lease expired
Fri Jul 29 21:44:28 2022 daemon.info dhcpcd[4190]: eth0: REPLY6 received from fe80::6664:9bff:feb0:7fc5
Fri Jul 29 21:44:28 2022 daemon.info dhcpcd[4190]: eth0: will expire
Fri Jul 29 21:44:28 2022 daemon.err dhcpcd[4190]: eth0: DHCPv6 lease expired
Fri Jul 29 21:44:30 2022 daemon.info dhcpcd[4190]: eth0: REPLY6 received from fe80::6664:9bff:feb0:7fc5
Fri Jul 29 21:44:30 2022 daemon.info dhcpcd[4190]: eth0: will expire
Fri Jul 29 21:44:30 2022 daemon.err dhcpcd[4190]: eth0: DHCPv6 lease expired
Fri Jul 29 21:44:31 2022 daemon.info dhcpcd[4190]: eth0: REPLY6 received from fe80::6664:9bff:feb0:7fc5
Fri Jul 29 21:44:31 2022 daemon.info dhcpcd[4190]: eth0: will expire
Fri Jul 29 21:44:31 2022 daemon.err dhcpcd[4190]: eth0: DHCPv6 lease expired`

系统进程里出现了几个杀不死的6in4.sh进程，敏感信息用星号掩去，改mtu无效果
`{6in4.sh} /bin/sh ./6in4.sh 6in4 setup wan6 {"proto":"6in4","peeraddr":"***.***.***.***","ip6addr":"**********::2/64","ip6prefix":["********::/64"],"tunnelid":"******","username":"*******","password":"*******","mtu":1480,"mtu":1480}`

coolsnowwolf, 2026-09-05:
6in4 重启失败只有现象描述，没有 `/etc/config/network`、6in4 脚本输出、进程状态或 debug 日志，无法确认 netifd/6in4 的具体缺陷；请按当前版本提供最小复现，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf