coolsnowwolf/lede #9867: Intel82576双口网卡其中一个网口报错igb 0000:01:00.0 eth0: Reset adapter

interweaveOS, 2022-07-30:
## 1.问题
内核报错 igb 0000:01:00.0 eth0: Reset adapter

## 2. 详细叙述

### (1) 具体问题

只要在82576网卡的eth0网口上连接网线到电脑或者路由器，就会不断重置eth0网口，网卡上的灯也亮一会就灭，灭了一会又亮，导致无法访问互联网

### (2) 路由器型号和固件版本

HP T620 PLUS | OpenWrt 21.02.2 2022071510 内核版本  5.4.179

### (3) 详细日志

[ 2859.074340] ------------[ cut here ]------------
[ 2859.074353] NETDEV WATCHDOG: eth0 (igb): transmit queue 0 timed out
[ 2859.074416] WARNING: CPU: 0 PID: 0 at dev_watchdog+0x21c/0x220
[ 2859.074418] Modules linked in: xt_FULLCONENAT rtl8192cu rtl8192c_common rtl_usb qcserial pppoe ppp_async option mt76x0u mt76x0_common iwlmvm iwldvm ipw cdc_mbim ath10k_pci ath10k_core ath usb_wwan usb_serial_simple ti_usb_3410_5052 sr9700 smsc95xx sierra_net sierra rtlwifi rtl8812au rndis_host qmi_wwan pppox ppp_mppe ppp_generic plusb pl2303 oti6858 mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt7663u mt7663_usb_sdio_common mt7615_common mt76_usb mt76_connac_lib mt76 mos7840 mos7720 mct_u232 mcs7830 mac80211 keyspan kalmia iwlwifi ipt_REJECT io_edgeport huawei_cdc_ncm garmin_gps ftdi_sio dm9601 cypress_m8 cp210x ch341 cfg80211 cdc_subset cdc_ncm cdc_ether belkin_sa ax88179_178a asix ark3116 aqc111 amdgpu zstd xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_bpf
[ 2859.074539] xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xr_usb_serial_common visor via_velocity via_rhine usbserial usbnet usblp usbatm ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_fsm ts_bm slhc sky2 skge sis900 sis190 sfc_falcon sfc rtl8150 radeon r8169 r8152 r8125 r6040 pegasus pcnet32 ntfs3 niu nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount ne2k_pci mmc_spi mlx5_core mlx4_en mlx4_core mdio_gpio mdio_bitbang macvlan lzo kvm kaweth irqbypass iptable_raw
[ 2859.074649] iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables igc iavf i40e hso hid_cp2112 gpu_sched forcedeth ezusb exfat ethoc et131x e1000e e100 dmx_usb crc7 crc_ccitt compat cm109 chaoskey cdc_wdm cdc_acm br_netfilter bnx2x bnx2 be2net auxiliary atl2 atl1e atl1c atl1 asn1_decoder alx 8390 8139too 8139cp fuse act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact configs sg evdev drivetemp i2c_dev ledtrig_usbport spi_ks8995 siit xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink st rtl8366s rtl8366rb rtl8306 ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip17xx b53_mdio b53_common nf_log_ipv6
[ 2859.074764] nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3 nfsd nfs msdos bonding 3c59x ixgbevf ixgbe igbvf igb e1000 ifb dummy mdio ip_tunnel hfcpci hfcmulti rpcsec_gss_krb5 auth_rpcgss oid_registry dm_raid veth tun nbd mISDN_dsp l1oip mISDN_core xfrm_user xfrm_ipcomp af_key xfrm_algo udf crc_itu_t lockd sunrpc grace minix hfsplus hfs cramfs configfs cifs autofs4 rxrpc 9p dns_resolver dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax br2684 atm aoe multipath fscache 9pnet_virtio 9pnet raid456 async_raid6_recov async_pq async_xor async_memcpy async_tx raid10 raid1 raid0 linear md_mod nls_utf8 nls_cp936 natsemi vxlan udp_tunnel ip6_udp_tunnel sha512_ssse3 sha512_generic sha256_ssse3 sha256_generic libsha256 sha1_ssse3 sha1_generic seqiv jitterentropy_rng drbg pcbc md5 md4 hmac ghash_generic gcm fcrypt echainiv des_generic libdes deflate cts ctr cmac ccm cbc authenc arc4 crypto_acompress nls_iso8859_1 nls_cp437 vfat fat uas
[ 2859.074905] xhci_plat_hcd dwc3 dwc2 rtl8367b swconfig rtl8366_smi sata_via sata_sil24 sata_sil pata_pdc202xx_old sata_nv sata_mv pata_artop fsl_mph_dr_of ehci_platform ehci_fsl mvsas xfs reiserfs jfs btrfs zstd_decompress zstd_compress xxhash xor raid6_pq lzo_decompress lzo_compress libcrc32c button_hotplug libsas tg3 b44 ssb ptp realtek broadcom bcm_phy_lib bcm84881 pps_core mii libphy
[ 2859.074963] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.179 #0
[ 2859.074966] Hardware name: Hewlett-Packard HP t620 PLUS Quad Core TC/213D, BIOS L40 v02.10 07/03/2015
[ 2859.074974] RIP: 0010:dev_watchdog+0x21c/0x220
[ 2859.074981] Code: 00 49 63 4d e0 eb 92 4c 89 e7 c6 05 1e 51 ae 00 01 e8 98 34 fc ff 89 d9 4c 89 e6 48 c7 c7 a8 22 09 82 48 89 c2 e8 cc 07 12 00 <0f> 0b eb c0 55 48 89 e5 41 57 49 89 d7 41 56 4d 89 c6 41 55 49 89
[ 2859.074985] RSP: 0018:ffffc90000003e90 EFLAGS: 00010286
[ 2859.074989] RAX: 0000000000000037 RBX: 0000000000000000 RCX: 0000000000000006
[ 2859.074993] RDX: 0000000000000007 RSI: 0000000000000002 RDI: ffff88813b41c4d0
[ 2859.074996] RBP: ffffc90000003ec0 R08: 0000000000000001 R09: 00000000000003d0
[ 2859.074999] R10: 0000000000000004 R11: 0000000000000000 R12: ffff888133b50000
[ 2859.075002] R13: ffff888133b50480 R14: 0000000000000008 R15: 0000000000000000
[ 2859.075006] FS: 0000000000000000(0000) GS:ffff88813b400000(0000) knlGS:0000000000000000
[ 2859.075010] CS: 0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 2859.075013] CR2: 00007f8e8de8e57c CR3: 00000001305c6000 CR4: 00000000000406f0
[ 2859.075016] Call Trace:
[ 2859.075022]
[ 2859.075031] ? dev_reset_queue.constprop.48+0x80/0x80
[ 2859.075040] call_timer_fn.isra.40+0x16/0x80
[ 2859.075047] run_timer_softirq+0x164/0x180
[ 2859.075054] ? lapic_next_event+0x1b/0x20
[ 2859.075061] ? clockevents_program_event+0x8b/0xf0
[ 2859.075071] __do_softirq+0xc9/0x20e
[ 2859.075078] irq_exit+0x6f/0x80
[ 2859.075084] smp_apic_timer_interrupt+0x66/0xa0
[ 2859.075091] apic_timer_interrupt+0xf/0x20
[ 2859.075094]
[ 2859.075102] RIP: 0010:cpuidle_enter_state+0x137/0x2c0
[ 2859.075107] Code: 89 c6 e8 3c 1e 9d ff 45 84 ff 74 17 9c 58 0f 1f 44 00 00 f6 c4 02 0f 85 67 01 00 00 31 ff e8 50 08 a1 ff fb 66 0f 1f 44 00 00 <45> 85 ed 0f 88 c5 00 00 00 49 63 cd 4c 2b 75 d0 48 8d 04 49 48 c1
[ 2859.075110] RSP: 0018:ffffffff82203e08 EFLAGS: 00000246 ORIG_RAX: ffffffffffffff13
[ 2859.075116] RAX: ffff88813b4249c0 RBX: ffffffff82256940 RCX: 000000000000001f
[ 2859.075119] RDX: 00000299ae1b8845 RSI: 00000000401eba76 RDI: 0000000000000000
[ 2859.075122] RBP: ffffffff82203e38 R08: 0000000000000002 R09: 0000000000024240
[ 2859.075124] R10: 0000053971ae4dd6 R11: 0000000000000000 R12: ffff88813ab8e000
[ 2859.075127] R13: 0000000000000002 R14: 00000299ae1b8845 R15: 0000000000000000
[ 2859.075135] ? cpuidle_enter_state+0x114/0x2c0
[ 2859.075141] cpuidle_enter+0x2f/0x40
[ 2859.075148] call_cpuidle+0x1e/0x40
[ 2859.075153] do_idle+0xf4/0x1c0
[ 2859.075158] cpu_startup_entry+0x18/0x20
[ 2859.075167] rest_init+0xa4/0xa6
[ 2859.075172] arch_call_rest_init+0x9/0xc
[ 2859.075176] start_kernel+0x568/0x576
[ 2859.075181] x86_64_start_reservations+0x29/0x2b
[ 2859.075186] x86_64_start_kernel+0x71/0x74
[ 2859.075191] secondary_startup_64+0xa4/0xb0
[ 2859.075197] ---[ end trace 4d4164ffec44a293 ]---
[ 2859.075296] igb 0000:01:00.0 eth0: Reset adapter
[ 2859.439670] br-lan: port 1(eth0) entered disabled state
[ 2861.355002] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX/TX
[ 2861.355363] br-lan: port 1(eth0) entered blocking state
[ 2861.355369] br-lan: port 1(eth0) entered forwarding state
[ 2865.385033] br-lan: received packet on eth1 with own address as source address (addr:00:1b:21:23:19:33, vlan:0)
[ 2867.010425] igb 0000:01:00.0 eth0: Reset adapter
[ 2867.375714] br-lan: port 1(eth0) entered disabled state
[ 2869.262976] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX/TX
[ 2869.263330] br-lan: port 1(eth0) entered blocking state
[ 2869.263337] br-lan: port 1(eth0) entered forwarding state
[ 2870.391953] br-lan: received packet on eth1 with own address as source address (addr:00:1b:21:23:19:33, vlan:0)
[ 2873.222501] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Down
[ 2873.223156] br-lan: port 1(eth0) entered disabled state
[ 2873.223520] igb 0000:01:00.0 eth0: Reset adapter

coolsnowwolf, 2022-08-01:
因为环路了

interweaveOS, 2022-08-01:
可是就那一个网口这样，而且我是直连电脑，不存在环路这一说

coolsnowwolf, 2026-09-05:
Intel 82576 在 OpenWrt 21.02.2/Linux 5.4.179 上的单端口 igb reset，缺少硬件 EEPROM、PCIe 链路和更换端口对照；更像硬件/链路或旧内核驱动问题，无法在当前 LEDE 代码中针对修复，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf