coolsnowwolf/lede #9871: 新固件重启后无线被禁用

axlexan, 2022-07-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：大半年没更新过固件了，便拉取了最新的代码后编译
启动后发现无线处于禁用状态，经过测试后发现，使用之前的备份配置，一切正常

经过对比发现/etc/config/network文件有所不同
[image]

[image]

虽不是什么大问题，我觉得还是有必要上报下，不知道是我的问题还是代码的问题

### (2) 路由器型号和固件版本

A：Redmi AC2100

### (3) 详细日志
Sat Jul 30 10:18:08 2022 kern.notice kernel: [    0.000000] Linux version 5.4.203 (clrlinux@xubuntu-pc) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4755-d8e34160b)) #0 SMP Sat Jul 30 02:17:58 2022
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] MIPS: machine is Xiaomi Redmi Router AC2100
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Sat Jul 30 10:18:08 2022 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sat Jul 30 10:18:08 2022 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sat Jul 30 10:18:08 2022 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] Zone ranges:
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000]   HighMem  empty
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] Movable zone start for each node
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] Early memory node ranges
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Sat Jul 30 10:18:08 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Sat Jul 30 10:18:08 2022 kern.debug kernel: [    0.000000]   Normal zone: 288 pages used for memmap
Sat Jul 30 10:18:08 2022 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sat Jul 30 10:18:08 2022 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s26576 r8192 d22576 u57344
Sat Jul 30 10:18:08 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s26576 r8192 d22576 u57344 alloc=14*4096
Sat Jul 30 10:18:08 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Sat Jul 30 10:18:08 2022 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] Writing ErrCtl register=00042140
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] Readback ErrCtl register=00042140
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] Memory: 120972K/131072K available (6031K kernel code, 268K rwdata, 728K rodata, 1276K init, 230K bss, 10100K reserved, 0K cma-reserved, 0K highmem)
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] NR_IRQS: 256
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.000008] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.007876] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.074045] pid_max: default: 32768 minimum: 301
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.078833] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jul 30 10:18:08 2022 kern.info kernel: [    0.086034] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[... 369 lines cut ...]
Sat Jul 30 10:19:48 2022 daemon.info pppd[3793]: syncppp not active
Sat Jul 30 10:19:48 2022 daemon.info pppd[3793]: Remote message: Authentication success,Welcome!
Sat Jul 30 10:19:48 2022 daemon.notice pppd[3793]: PAP authentication succeeded
Sat Jul 30 10:19:48 2022 daemon.notice pppd[3793]: peer from calling number 10:51:72:1B:24:AB authorized
Sat Jul 30 10:19:48 2022 daemon.notice pppd[3793]: local  LL address fe80::b447:36ea:fc03:bceb
Sat Jul 30 10:19:48 2022 daemon.notice pppd[3793]: remote LL address fe80::1251:72ff:fe1b:24ab
Sat Jul 30 10:19:48 2022 daemon.notice pppd[3793]: local  IP address 100.65.172.221
Sat Jul 30 10:19:48 2022 daemon.notice pppd[3793]: remote IP address 100.65.128.1
Sat Jul 30 10:19:48 2022 daemon.notice pppd[3793]: primary   DNS address 202.102.213.68
Sat Jul 30 10:19:48 2022 daemon.notice pppd[3793]: secondary DNS address 61.132.163.68
Sat Jul 30 10:19:48 2022 daemon.notice netifd: Network device 'pppoe-wan' link is up
Sat Jul 30 10:19:48 2022 daemon.notice netifd: Interface 'wan' is now up
Sat Jul 30 10:19:48 2022 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Sat Jul 30 10:19:48 2022 daemon.notice netifd: Interface 'wan_6' is enabled
Sat Jul 30 10:19:48 2022 daemon.notice netifd: Interface 'wan_6' has link connectivity
Sat Jul 30 10:19:48 2022 daemon.notice netifd: Interface 'wan_6' is setting up now
Sat Jul 30 10:19:48 2022 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sat Jul 30 10:19:48 2022 daemon.err odhcp6c[4464]: Failed to send SOLICIT message to ff02::1:2 (Operation not permitted)
Sat Jul 30 10:19:49 2022 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Sat Jul 30 10:19:49 2022 daemon.err odhcp6c[4464]: Failed to send SOLICIT message to ff02::1:2 (Operation not permitted)
Sat Jul 30 10:19:54 2022 daemon.notice netifd: Interface 'wan_6' is now up
Sat Jul 30 10:19:54 2022 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Sun Jul 31 10:55:08 2022 authpriv.info dropbear[4888]: Child connection from 192.168.1.146:58726
Sun Jul 31 10:55:12 2022 authpriv.notice dropbear[4888]: Password auth succeeded for 'root' from 192.168.1.146:58726
Sun Jul 31 10:55:46 2022 cron.err crond[2090]: time disparity of 1475 minutes detected

EnnawYang, 2022-07-31:
麻烦搜索下以往issue，另源码发生较大变化的情况下建议不保留配置升级

axlexan, 2022-08-01:
> 麻烦搜索下以往issue，另源码发生较大变化的情况下建议不保留配置升级

刷新固件时，执行了不保存原配置（后面测试时也多次执行了重置），后才发现这个问题的

使用旧的配置则没问题

coolsnowwolf, 2026-09-05:
报告来自 2022 年 MT7621/Redmi AC2100 旧固件，现象由未保存配置后的无线接口/网络配置差异引起，使用备份配置即可恢复；没有提供当前源码中的稳定复现或驱动错误。关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf