coolsnowwolf/lede #9873: TurboACC疑似编译不了了

whatawhiz, 2022-07-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：前段时间编译了个新版的固件用着。突然想看看turboacc的设置打开没，就找了半天没找到。我还以为是编译忘了勾。准备重新编译一个，结果pull了最新的代码，打开menuconfig之后发现，turboacc的主选项没了，子选项还在。不知道是不是大家都这样。如果只是我有问题我就好清除了重新下代码了。

### (2) 路由器型号和固件版本

A：OpenWrt R22.7.7 / LuCI Master (git-22.196.12585-0f5cc90)

### (3) 详细日志

A：
[image]

yanxin152133, 2022-07-31:
我昨天刚编译一次，没啥问题啊

yanxin152133, 2022-07-31:
[image]

whatawhiz, 2022-07-31:
> [image]

你menuconfig里有turboacc的选项吗？
有的话应该就是我这文件哪里有问题了。等会全清了重新试下

yanxin152133, 2022-07-31:
> > [image]
> 
> 你menuconfig里有turboacc的选项吗？ 有的话应该就是我这文件哪里有问题了。等会全清了重新试下

[image]
跟你这个一样，我记得我因为网络原因编译好几次其中一次我还专门把这下边的两个都勾选了

whatawhiz, 2022-07-31:
> 

总算找到问题出在哪了。。。刚才把Include Flow Offload选上之后，主选项才出来。。估计是之前有commit改了makefile的依赖。。

yanxin152133, 2022-07-31:
又一个熬夜编译的

closed 2022-08-01 by aiamadeus