coolsnowwolf/lede #9877: 为什么最近一周编译就没能成功过？

Travel2Here, 2022-08-01:
按之前的配置没动过只是更新编译，失败2次；
随后怀疑自己的问题，清空目录重来一遍，失败；
再来一把重新配置，失败；
最后一把，自定义源都删了保留默认，继续失败。。。
而且默认内核随着pull，版本不断变 一会儿5.15，一会18，一会10.。。是啥情况呀？

ecrasy, 2022-08-01:
代码更新了，之前的配置不能复用了
建议每次make menuconfig之后
使用scripts/diffconfig.sh > result.config保存差异部分
减少版本更替带来的冲突
每次编译之前make defconfig会将差异保存的config恢复至完整

ecrasy, 2022-08-01:
我的actions编译，没啥问题，除了cjdns需要修正Makefile，L大的所有package都编译通过。

Travel2Here, 2022-08-01:
> 我的actions编译，没啥问题，除了cjdns需要修正Makefile，L大的所有package都编译通过。

感谢哥们的耐心回答，你是没添加第三方feed编译的？

Shaco-Ma, 2022-08-01:
> 我的actions编译，没啥问题，除了cjdns需要修正Makefile，L大的所有package都编译通过。

大佬,为什么我本地编译全新代码什么都没修改编译不过呢,总是提示
install: error while loading shared libraries: libpcre.so.1: cannot open shared object file: No such file or directory

Travel2Here, 2022-08-01:
刚又编译了一次成功了，看来跟不同时间获取到的版本有很大关系。碰运气。（本次和之前一样，只是在原版的基础上增加了kenzo8的软件包库）
[image]

hzjnet, 2022-08-01:
> 我的actions编译，没啥问题，除了cjdns需要修正Makefile，L大的所有package都编译通过。

请教这个如何修正，能否把修正后的发出来共享？ 我只有这个编译不正常了，谢谢

ecrasy, 2022-08-01:
在我的github actions仓库里
自己去拿

hzjnet, 2022-08-02:
> 在我的github actions仓库里 自己去拿

谢谢，终于看见一个方法了！

closed 2022-10-26 by aiamadeus