coolsnowwolf/lede #9880: 如何彻底解决与lede插件冲突的问题

gbaoye, 2022-08-01:
困扰我很久的问题

ecrasy, 2022-08-01:
1. make menuconfig
   编译L大源码纯净版固件
2. 引入你的packages feeds
   update & install
3. make menuconfig
  选择你要编译的插件为m
4. 使用命令单独编译插件
  make package/feeds/packages/luci-app-插件名/compile -j$(nproc)
  编译生成目录bin/packages/$DEVICE_PLAFORM
5. 将插件ipk包拷贝发送至路由使用opkg命令安装
6. 最后一个最有效的办法：让你的插件开发者遵守开发规范

gbaoye, 2022-08-01:
> 1. make menuconfig
>    编译L大源码纯净版固件
> 2. 引入你的packages feeds
>    update & install
> 3. make menuconfig
>    选择你要编译的插件为m
> 4. 使用命令单独编译插件
>    make package/feeds/packages/luci-app-插件名/compile -j$(nproc)
>    编译生成目录bin/packages/$DEVICE_PLAFORM
> 5. 将插件ipk包拷贝发送至路由使用opkg命令安装
> 6. 最后一个最有效的办法：让你的插件开发者遵守开发规范

我用的[AutoBuild](https://github.com/gbaoye/Newifi3-OpenWrt--AutoBuild)

xiangfeidexiaohuo, 2022-08-01:
有一个办法，自己不编译，用别人现成的，嘿嘿。

zhangguanzhang, 2022-08-02:
https://github.com/zhangguanzhang/Actions-OpenWrt/tree/test
我的项目，固件默认slim，不安装插件，然后有本地源，用啥就安装啥 😘

closed 2022-10-26 by aiamadeus